esp-rs/rust-build #17: Add cargo and rust fmt to the toolchain build

grbd, 2021-11-11:
Hi,
I've been experimenting with my own build of the toolchain for a while now
https://github.com/esp-rs/rust

One of the things I've discovered is a unique bug associated with windows when using a linked / custom toolchain such as esp
Typically some tools such as "cargo make" or "rust-analyser" under visual studio code
tend to run cargo multiple times at the same time (similar to running it multithreaded)

If cargo isn't installed as part of the custom toolchain, then it tends to try and use the cargo native to the host.
But because of the way it does this, it tends to fail under windows when you try to run it more than once at the same time.

  * https://github.com/rust-lang/rustup/issues/2889

The best workaround I've found is to just install cargo into the custom toolchain
```
# Fixes VSCode Issues with rust-analyser / cargo-make
x.py build --stage 2 cargo
```

I also tend to add rustfmt for reformatting code
```
# Add rustfmt for reformatting code
x.py build --stage 2 rustfmt
```

Would there be any issues in adding these to the workflows script?

grbd, 2021-11-18:
Just to add I've recently been doing a rather big re-write of the github workflows
I've gotten everything split into composite actions to make it easier to handle / read so that the top level workflow isn't massive.
I've also found a workaround for the path issue under windows, and fixed some build related issues as well.
and split the build into multiple jobs to avoid disk space issues.

Currently doing some tweaking for ubuntu then I'll test out macos since it should have the same unix based paths.

georgik, 2021-11-24:
rustfmt is being built as part of the toolchain. Let me double check.

georgik, 2021-11-24:
@grbd Which toolchain do you use as default?
Former instructions were written in that way that toolchain esp is set as default.
We changed the instructions to leave original default toolchain installed on the system and add option `+esp` to cargo command when building.
The build command looks like this: `cargo +esp build`
This command should result in using tools which are not part of custom toolchain directly from default toolchain.

grbd, 2021-11-24:
I'm using it from a windows platform at the moment.
But it doesn't work if you try and run cargo multiple times at the same time while using a custom toolchain.
(which some dev tools such as rust-analyser, or cargo make do)

I think it's a wider issue with rust / rustup in the way they make cargo visible while using the custom toolchain
I think it's also only a windows issue.

georgik, 2021-12-02:
@grbd I'm running build with esp-rs/rust branch esp-1.57.0.0 - Ubuntu 18 rustc fails on this error:
```
Copying stage2 rustc from stage1 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
thread 'main' panicked at 'fs::read(stamp) failed with No such file or directory (os error 2) ("/home/runner/work/rust-build/rust-build/rust/build/x86_64-unknown-linux-gnu/stage2-rustc/x86_64-unknown-linux-gnu/release/.librustc.stamp")', src/bootstrap/lib.rs:1296:24
```
https://github.com/esp-rs/rust-build/runs/4394875515?check_suite_focus=true#step:8:7911
Any idea what might be wrong?

grbd, 2021-12-02:
Hmm,
not sure at this point although I think a few things have been updated
The failure seems to be at
```
Copying stage2 rustc from stage1 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
```
I think it's searching for the following but can't find it
```
/home/runner/work/rust-build/rust-build/rust/build/x86_64-unknown-linux-gnu/stage2-rustc/x86_64-unknown-linux-gnu/release/.librustc.stamp
```
I'll try pulling in the changes to a fork then run it on github workflows

grbd, 2021-12-02:
I'm running some builds now to see if I can figure out where it is
my suspicion is that it's something ether in the esp rust src itself or something related to the runner

I did spot that you've changed .github/actions/util/checkout-esp-rust/action.yaml
from rust_version to release_version
Originally the way I had it setup was:

  * rust_version - the tag / branch checked out from esp-rs/rust such as 1.56.0
  * release_version - the name given to the final archive such as 1.56.0.1 (to indicate changes via rust-build)

georgik, 2021-12-02:
@grbd Thank you very much for testing the build.

Here is explanation of action.yaml change: I've changed one occurrence of rust_version to release_version, because @MabezDev marked new branch as esp-1.57.0.0. In 1.56 it was marked as esp-1.56.0. Having build number at the end matching the release_version is better, because it's easier to recognize changes if multiple build of same Rust toolchain occurs.

Meanwhile I'm building all the versions locally, just to have working reference until we resolve mystery of GitHub runners. Rust builds are pretty complex and we encountered several strange issues like this before.

grbd, 2021-12-02:
Ok that makes sense.
I'm testing the original build first (just to check github hasn't changed anything
then I'm going to slowly add the changes in from 1.57 to see where it fails

georgik, 2021-12-02:
Hm, the issue seems to be related to .stamp on GitHub Runners. 
I can recall some other project which was pushed to repo with timestamp "in-the-future" and GitHub Runner clocks were in different time zone which looks like "past" to cmake, and cmake was stuck in infinite loop of refresh. The solution was to touch all files before running cmake.

grbd, 2021-12-03:
So I updated my own workflows to be identical to these but using 1.56.0 and that seemed to work fine
then I switched just the version from 1.56.0 to 1.57.0.0 and thats when it failed (on github's native runner)
So the difference I suspect is somewhere within https://github.com/esp-rs/rust

I might try and give this a quick go just to see if it makes any difference
https://github.com/szenius/set-timezone

georgik, 2021-12-03:
@MabezDev @jessebraham  Any idea which change in Rust 1.57.0 bs 1.56.0 might caused build behavior described above as mentioned by @grbd ?

grbd, 2021-12-03:
after a bit more testing it looks like it fails during the build of compiler/rustc
``` bash
# Works ok here
python3 src/bootstrap/configure.py  --experimental-targets=Xtensa --enable-extended --dist-compression-formats='xz'
# Fails during the build of compiler/rustc
python3 x.py build --exclude src/doc --stage 2 compiler/rustc
```

The error shows up at
```
Assembling stage2 compiler (x86_64-unknown-linux-gnu)
Uplifting stage1 std (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu)
Copying stage2 std from stage1 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
Uplifting stage1 rustc (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu)
Copying stage2 rustc from stage1 (x86_64-unknown-linux-gnu -> x86_64-unknown-linux-gnu / x86_64-unknown-linux-gnu)
Assembling stage3 compiler (x86_64-unknown-linux-gnu)
thread 'main' panicked at 'fs::read(stamp) failed with No such file or directory (os error 2) ("/home/runner/work/rust-build/rust-build/rust/build/x86_64-unknown-linux-gnu/stage2-rustc/x86_64-unknown-linux-gnu/release/.librustc.stamp")', src/bootstrap/lib.rs:1296:24
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Build completed unsuccessfully in 2:01:59
Error: Process completed with exit code 1.
```

Maybe you get different results if you try building everything at once.
The workflows just build compiler/rustc as part of a job to get around disk space issues.

I think the separate build of the tools and library/std works ok
So maybe it's some form of build dependency related issue, might try building library/std first before hand to see what happens.

grbd, 2021-12-04:
With RUST_BACKTRACE=1 enabled
```
thread 'main' panicked at 'fs::read(stamp) failed with No such file or directory (os error 2) ("/home/runner/work/rust-build/rust-build/rust/build/x86_64-unknown-linux-gnu/stage2-rustc/x86_64-unknown-linux-gnu/release/.librustc.stamp")', src/bootstrap/lib.rs:1296:24
stack backtrace:
   0: rust_begin_unwind
             at /rustc/59eed8a2aac0230a8b53e89d4e99d55912ba6b35/library/std/src/panicking.rs:517:5
   1: core::panicking::panic_fmt
             at /rustc/59eed8a2aac0230a8b53e89d4e99d55912ba6b35/library/core/src/panicking.rs:101:14
   2: bootstrap::Build::read_stamp_file
             at ./src/bootstrap/lib.rs:1296:24
   3: <bootstrap::compile::Assemble as bootstrap::builder::Step>::run
             at ./src/bootstrap/compile.rs:1107:27
   4: bootstrap::builder::Builder::ensure
             at ./src/bootstrap/builder.rs:1574:23
   5: bootstrap::builder::Builder::compiler
             at ./src/bootstrap/builder.rs:627:9
   6: <bootstrap::compile::Assemble as bootstrap::builder::Step>::make_run
             at ./src/bootstrap/compile.rs:1034:30
   7: bootstrap::builder::StepDescription::maybe_run
             at ./src/bootstrap/builder.rs:175:13
   8: bootstrap::builder::StepDescription::run
             at ./src/bootstrap/builder.rs:231:21
   9: bootstrap::builder::Builder::run_step_descriptions
             at ./src/bootstrap/builder.rs:619:9
  10: bootstrap::builder::Builder::execute_cli
             at ./src/bootstrap/builder.rs:599:9
  11: bootstrap::Build::build
             at ./src/bootstrap/lib.rs:624:13
  12: bootstrap::main
             at ./src/bootstrap/bin/main.rs:33:5
  13: core::ops::function::FnOnce::call_once
             at /rustc/59eed8a2aac0230a8b53e89d4e99d55912ba6b35/library/core/src/ops/function.rs:227:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

```

georgik, 2021-12-06:
@grbd Thank you for the analysis.

I've tried to switch the order of rustc and std, but it does not help.
https://github.com/esp-rs/rust-build/runs/4429880343?check_suite_focus=true#step:9:84

grbd, 2021-12-06:
Yep I tried the same thing myself without much luck

georgik, 2021-12-07:
Components were added to artifacts. Thank you @grbd for the contribution. Code is on main.

GitHub runner limitation for 1.57 will be addressed as separate issue.

closed 2021-12-07 by georgik