esp-rs/rust-build #20: riscv32imc-unknown-none-elf is not recognized by +esp toolchain

georgik, 2021-11-29:
```
cd esp32c3-hal/examples/empty
cargo +esp-1.56.0.1 build
error[E0463]: can't find crate for `core`
  |
  = note: the `riscv32imc-unknown-none-elf` target may not be installed
```
While classical night works:
```
cargo build
Finished dev [unoptimized + debuginfo] target(s) in 10.94s
```

I would expect, that our toolchain is able to process the same build.
Adding the target to our toolchain does not work:
```
rustup target add riscv32imc-unknown-none-elf --toolchain +esp-1.56.0.1
error: toolchain '+esp-1.56.0.1' does not support components: +esp-1.56.0.1 is a custom toolchain
```

It seems that we need to turn on component support.

MabezDev, 2021-11-29:
This is because we don't build the `riscv32imc-unknown-none-elf` targets core when we build our compiler fork. We also cannot add the target from another toolchain (such as nightly), which seems to be a limitation of `rustup`.

georgik, 2021-12-07:
non-blocking issue, moving to v1.58

georgik, 2022-09-22:
No longer valid. Closing

closed 2022-09-22 by georgik