esp-rs/rust-build #23: SIGABRT: Failed to compile `clap v2.34.0`

thorlucas, 2021-12-02:
Cloning and compiling on `macOS Monterey 12.0.1` fails when building `clap v2.34.0`:

```
error: could not compile `clap`

Caused by:
  process didn't exit successfully: `rustc --crate-name clap --edition=2018 /Users/thorcorreia/.cargo/registry/src/github.com-1ecc6299db9ec823/clap-2.34.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C split-debuginfo=unpacked -C debuginfo=2 --cfg 'feature="ansi_term"' --cfg 'feature="atty"' --cfg 'feature="color"' --cfg 'feature="default"' --cfg 'feature="strsim"' --cfg 'feature="suggestions"' --cfg 'feature="vec_map"' -C metadata=ac26a8e48a625ff4 -C extra-filename=-ac26a8e48a625ff4 --out-dir /Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps -L dependency=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps --extern ansi_term=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libansi_term-7bb5c8085badc50a.rmeta --extern atty=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libatty-02f4983f38003a7d.rmeta --extern bitflags=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libbitflags-45b4e505e2497911.rmeta --extern strsim=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libstrsim-2b74c9109d8bb938.rmeta --extern textwrap=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libtextwrap-9684b827efe0b904.rmeta --extern unicode_width=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libunicode_width-d208e9921c947bc5.rmeta --extern vec_map=/Users/thorcorreia/Dev/rust-esp32-std-demo/target/debug/deps/libvec_map-63037abaf65b04ec.rmeta --cap-lints allow` (signal: 6, SIGABRT: process abort signal)
```

georgik, 2021-12-02:
Thank you for reporting the problem. It seems to be the same problem as #15 
Today we're preparing 1.57 build which should resolve the problem.

georgik, 2021-12-02:
@thorlucas The issue is still present in 1.57.0.0.
Recommended workaround: use Podman/Docker image https://hub.docker.com/r/espressif/idf-rust on Monterey

Any hint what could be the root-cause is appreciated.

georgik, 2021-12-03:
The problem might be solved by `--set rust.jemalloc`: https://github.com/rust-lang/rust/blob/acbe4443cc4c9695c0b74a7b64b60333c990a400/src/ci/github-actions/ci.yml#L456
Testing the new build.

georgik, 2021-12-03:
@thorlucas We found the reason of SIGABRT. The build of Rust for macOS x86_64 must contain option `--set rust.jemalloc`.
The new build is ready. Please test following script https://github.com/esp-rs/rust-build/tree/build/1.57: 
```
./test-rust-toolchain.sh -t 1.57.0.1 --installation-mode reinstall
```
This will deploy toolchain with name esp-1.57.0.1. You can call your builds with `cargo +esp-1.57.0.1 ..`.

georgik, 2021-12-07:
Solved in version 1.57.0.1.

closed 2021-12-07 by georgik