esp-rs/rust-build #24: Add install-rust-toolchain.sh to release artifacts

rgcottrell, 2021-12-03:
I think it would be helpful to add the install-rust-toolchain.sh installer script to the release artifacts. This would make it easier to automate the process of installing the binary toolchains by simply downloading and then running the installer script. This would allow individual users to install the toolchain without needing to check out the source repository first.

MabezDev, 2021-12-03:
@georgik this seems doable, we should just be able to append to the tar archive that `x.py dist` gives us? I don't know if that would mess up any signing or verification (if we do any at the moment) though.

georgik, 2021-12-03:
That's great idea. 👍

georgik, 2021-12-07:
Added .sh and .ps1.

closed 2021-12-07 by georgik