esp-rs/rust-build #26: `install-rust-toolchain.sh` bundled with tag 1.57.0.2 installs version 1.57.0.0

bugadani, 2021-12-08:


georgik, 2021-12-15:
Thanks for reporting the problem. The relevant version of the script is also part of Releases.
Also it's possible to explicitly specify the version as a parameter to install script using `--toolchain-version` parameter.
The shifted tag is confusing. We should consider how to handle the situation for 1.58. 

BTW: @bugadani  Which CI do you use? Are you considering also Podman/Docker image idf-rust or idf-rust-examples? In case of GitHub there is also GitHub action which is able to use these images.

bugadani, 2021-12-15:
I'm using GitHub actions, but a custom script because either there was no action yet when I started it, or I couldn't discover it very well. I'll take a look on what's available now.

PR #32 (2022-02-02): Build/1.58
Update build scripts for 1.58

Closes #26
closed 2022-02-02 by georgik