esp-rs/rust-build #28: Mention in docs that rustup installed through snap has different paths set and creates issues

lenalebt, 2021-12-21:
Hey!
I'm new to rust and was on to trying to replace some of my esp32 coding on arduino IDE with rust. I tried following this guide, but ran into trouble that I now resolved by installing rustup not via snap, but manually. Apparently the snap version of rustup stores its data somewhere else and therefore the default paths do not match well with the defaults set here.

Maybe it's enough for the next person to stumble upon this issue. I did not work out the snap paths but installed it manually instead, making it work instantly.

Cheers and keep up the good work :).

georgik, 2021-12-22:
@lenalebt Thank you for reporting the problem.

The recommended installation method is to use https://github.com/esp-rs/rust-build/blob/main/install-rust-toolchain.sh

We also recommend to use: https://github.com/esp-rs/rust-build/blob/main/test-rust-toolchain.sh
The test script performs whole installation and builds an example of rust-std-demo application

In case of ESP32C3 you can use Rust compiler directly from rust-lang, because it has RISC-V architecture which is part of standard toolchain.

Regarding snaps we will add it to test scenario.

lenalebt, 2021-12-22:
Yeah, so I used the https://github.com/esp-rs/rust-build/blob/main/install-rust-toolchain.sh - but rustup was already installed on my system beforehand, using a snap package, from some time ago (another rust project). And the script fails to install the esp toolchain into the "correct" directory then, leaving me with "rustup toolchain list" working, just not showing the esp toolchain... I did not understand what happened at first, this is why I thought it might make sense to mention that in the docs.

georgik, 2022-03-07:
@SergioGasquez Please, could you take a look and Snap installation?

SergioGasquez, 2022-03-07:
Just tried running `./install-rust-toolchain.sh` having rust installed via snap, here is the error that I am getting:
```
* installing /home/sergio/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu
already installed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--       0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     100   140  100   140    0     0    410      0 --:--:-- --:--:-- --:--:--   409
100   653  100   653    0     0   1071      0 --:--:-- --:--:-- --:--:--  1071
Warning: Failed to create the file /home/sergio/.cargo//bin/cargo-espflash: No
Warning: such file or directory
  0 9783k    0   877    0     0    955      0  2:54:50 --:--:--  2:54:50   955
curl: (23) Failure writing output to destination
```
If I remove Rust installation from snap, the error disappears

georgik, 2022-05-20:
We would recommend different approach: Use Dev Containers instead of Snaps.
It provides same base functionality and adds more conveniet way to develop. It also works across different operating systems.
Example of project with Dev Container with Wokwi integration is here: https://github.com/georgik/rustzx-esp32#build-and-run-wokwi-simulation
Other alternative is to use GitPod, which provides the same functionality like Dev Containers, but it cloud: https://github.com/georgik/rustzx-esp32#build-using-gitpod

Closing the issue since there is different solution which provides the same functionality. Feel free to reopen the issue if none of those methods mentioned above is working for you.

closed 2022-05-20 by georgik