esp-rs/rust-build #29: Cargo binary for macOS assumes OpenSSL is installed via Homebrew

stevecheckoway, 2021-12-25:
When following the instructions to install the toolchain on macOS, the `cargo` that gets installed links to OpenSSL via the `openssl-sys` crate and has library paths that assume OpenSSL was installed via Homebrew.

This leads to the following error if OpenSSL is installed by, for example, MacPorts.
```
$ cargo +esp
dyld: Library not loaded: /usr/local/opt/openssl@1.1/lib/libssl.1.1.dylib
  Referenced from: /Users/steve/.rustup/toolchains/esp/bin/cargo
  Reason: image not found
Abort trap: 6
```

This can be fixed using `install_name_tool(1)`.
```
$ install_name_tool -change /usr/local/opt/openssl@1.1/lib/libssl.1.1.dylib /opt/local/lib/libssl.3.dylib \
                    -change /usr/local/opt/openssl@1.1/lib/libcrypto.1.1.dylib /opt/local/lib/libcrypto.3.dylib \
                    cargo
```

This allows `cargo` to run, but the libraries have different compatibility versions so I'm not sure any OpenSSL functions would actually work correctly or not.

georgik, 2022-06-29:
The problem was detected also with 1.62.0.0. Suggested solution works.
Here is small tweak to acquire `cargo` command from active PATH:
```
install_name_tool -change /usr/local/opt/openssl@1.1/lib/libssl.1.1.dylib /opt/local/lib/libssl.3.dylib \
 -change /usr/local/opt/openssl@1.1/lib/libcrypto.1.1.dylib /opt/local/lib/libcrypto.3.dylib \
 `which cargo`
```

georgik, 2022-07-22:
Solved in version 1.62.0.0

closed 2022-07-22 by georgik

stevecheckoway, 2022-07-27:
Hi @georgik,

This issue has not been fixed.

Steps to reproduce:

1. Clone the `rust-build` repository.
2. Run the install script
   ```
   [DEP:~/esp/rust-build] (main) steve% ./install-rust-toolchain.sh
   Processing configuration:
   --build-target          = esp32,esp32s2,esp32s3
   --cargo-home            = /Users/steve/.cargo
   --clear-cache           = YES
   --esp-idf-version       =
   --export-file           =
   --extra-crates          = ldproxy cargo-espflash
   --installation-mode     = install
   --llvm-version          = esp-14.0.0-20220415
   --minified-esp-idf      = NO
   --minified-llvm         = YES
   --nightly-version       = nightly
   --rustup-home           = /Users/steve/.rustup
   --system-packages       = openssl@3
   --toolchain-version     = 1.62.1.0
   --toolchain-destination = /Users/steve/.rustup/toolchains/esp
   /Users/steve/.cargo/bin/rustup
   nightly-aarch64-apple-darwin
   Processing toolchain for aarch64-apple-darwin - operation: install
   Installing system packages: openssl@3
   Warning: Unable to find command brew. Skipping installation of system package.
   ** downloading: https://github.com/esp-rs/rust-build/releases/download/v1.62.1.0/rust-1.62.1.0-aarch64-apple-darwin.tar.xz
     % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                    Dload  Upload   Total   Spent    Left  Speed
     0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
   100 75.7M  100 75.7M    0     0  11.2M      0  0:00:06  0:00:06 --:--:-- 13.8M
   install: creating uninstall script at /Users/steve/.rustup/toolchains/esp/lib/rustlib/uninstall.sh
   install: installing component 'rustc'
   install: installing component 'rust-std-aarch64-apple-darwin'
   install: installing component 'cargo'
   install: installing component 'rustfmt-preview'
   install: installing component 'clippy-preview'
   
       rust installed.
   
     % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                    Dload  Upload   Total   Spent    Left  Speed
     0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
   100 2324k  100 2324k    0     0  3094k      0 --:--:-- --:--:-- --:--:-- 8444k
   install: creating uninstall script at /Users/steve/.rustup/toolchains/esp/lib/rustlib/uninstall.sh
   install: installing component 'rust-src'
   
       rust src installed.
   
   * installing /Users/steve/.espressif/tools/xtensa-esp32s3-elf-gcc/8_4_0-esp-2021r2-patch3-aarch64-apple-darwin
   already installed
   * installing /Users/steve/.espressif/tools/xtensa-esp32s2-elf-gcc/8_4_0-esp-2021r2-patch3-aarch64-apple-darwin
   already installed
   * installing /Users/steve/.espressif/tools/xtensa-esp32-elf-gcc/8_4_0-esp-2021r2-patch3-aarch64-apple-darwin
   already installed
   * installing /Users/steve/.espressif/tools/xtensa-esp32-elf-clang/esp-14.0.0-20220415-aarch64-apple-darwin
   already installed
   Removing cached dist files:
    - rust-1.62.1.0-aarch64-apple-darwin
    - rust-1.62.1.0-aarch64-apple-darwin.tar.xz
    - rust-src-1.62.1.0
    - rust-src-1.62.1.0.tar.xz
    - xtensa-esp32-elf-llvm14_0_0-esp-14.0.0-20220415-aarch64-apple-darwin.tar.xz
    - *-elf-gcc*.tar.gz
   Add following command to /Users/steve/.zshrc
   export LIBCLANG_PATH="/Users/steve/.espressif/tools/xtensa-esp32-elf-clang/esp-14.0.0-20220415-aarch64-apple-darwin/lib/"
   export PATH="/Users/steve/.espressif/tools/xtensa-esp32-elf-gcc/8_4_0-esp-2021r2-patch3-aarch64-apple-darwin/bin/:/Users/steve/.espressif/tools/xtensa-esp32s2-elf-gcc/8_4_0-esp-2021r2-patch3-aarch64-apple-darwin/bin/:/Users/steve/.espressif/tools/xtensa-esp32s3-elf-gcc/8_4_0-esp-2021r2-patch3-aarch64-apple-darwin/bin/:$PATH"
   ```
3. Examine the newly installed `cargo` to see which libraries it links against.
   ```
   [DEP:~/esp/rust-build] (main) steve% otool -L ~/.rustup/toolchains/esp/bin/cargo
   /Users/steve/.rustup/toolchains/esp/bin/cargo:
   	/usr/lib/libiconv.2.dylib (compatibility version 7.0.0, current version 7.0.0)
   	/System/Library/Frameworks/Security.framework/Versions/A/Security (compatibility version 1.0.0, current version 60157.60.19)
   	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1856.105.0)
   	/opt/homebrew/opt/openssl@3/lib/libssl.3.dylib (compatibility version 3.0.0, current version 3.0.0)
   	/opt/homebrew/opt/openssl@3/lib/libcrypto.3.dylib (compatibility version 3.0.0, current version 3.0.0)
   	/System/Library/Frameworks/SystemConfiguration.framework/Versions/A/SystemConfiguration (compatibility version 1.0.0, current version 1163.60.3)
   	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
   	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
   	/usr/lib/libresolv.9.dylib (compatibility version 1.0.0, current version 1.0.0)
   ```

So rather than assuming that the openssl libraries were in `/usr/local/opt` as it did when I filed this issue, it now assumes that they are in `/opt/homebrew/opt`. We can see this if we try running `cargo +esp --version`:

```
[DEP:~/esp/rust-build] (main) steve% cargo +esp --version
dyld[10805]: Library not loaded: /opt/homebrew/opt/openssl@3/lib/libssl.3.dylib
  Referenced from: /Users/steve/.rustup/toolchains/esp/bin/cargo
  Reason: tried: '/opt/homebrew/opt/openssl@3/lib/libssl.3.dylib' (no such file), '/Users/steve/.rustup/toolchains/esp/lib/libssl.3.dylib' (no such file), '/Users/steve/lib/libssl.3.dylib' (no such file), '/usr/local/lib/libssl.3.dylib' (no such file), '/usr/lib/libssl.3.dylib' (no such file)
zsh: abort      cargo +esp --version
```

The underlying issue appears not to have been solved.

stevecheckoway, 2022-07-27:
I just noticed that the path differences are due to architecture. On an Apple M1, it's looking in `/opt/homebrew/opt`. On x86-64, it's looking in `/usr/local/opt`. But in either case, the pre-built artifacts are linking against a library not provided by macOS.

thynus, 2025-02-21:
Issue not fixed:

cargo-generate --version
dyld[4468]: Library not loaded: '/opt/local/libexec/openssl3/lib/libssl.3.dylib'
  Referenced from: '/Users/baxter/.cargo/bin/cargo-generate'
  Reason: tried: '/opt/local/libexec/openssl3/lib/libssl.3.dylib' (no such file), '/usr/local/lib/libssl.3.dylib' (no such file), '/usr/lib/libssl.3.dylib' (no such file)
zsh: abort      cargo-generate --version

rustc 1.84.1 (e71f9a9a9 2025-01-27)
cargo 1.84.1 (66221abde 2024-11-19)

PR #128 (2022-08-04): Build without linking to homebrew libs
The `openssl-sys` crate will either link against a Homebrew-provided
OpenSSL, or it will build OpenSSL itself.

Using the `cargo-native-static` build option will force `openssl-sys` to
build OpenSSL itself. This produces binaries that do not link against
any non-system-provided libraries. This is how official builds of
`rust-lang/rust` are performed. See
https://github.com/rust-lang/rust/blob/da5b546d2e563747b16a16dae83bacf49aa0bf3b/src/ci/run.sh#L54

Fixes: #29