esp-rs/rust-build #30: Unable to build `rust-esp32-std-demo` using Docker and cargo

akselbor, 2022-01-02:
Hello!

I seem unable to build `rust-esp32-std-demo` using the Docker image `espressif/idf-rust-examples:latest` (image ID `562dd6cc6e54`) using cargo. The issue seems to be related to PlatformIO. Builds using `idf.py build` work fine.

The host machine is an Intel-Mac running macOS Big Sur v11.2.3.

Suggestions?

Relevant logs:
```text
docker run -it espressif/idf-rust-examples
Detecting the Python interpreter
Checking "python" ...
Python 3.8.10
"python" has been detected
Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool xtensa-esp32-elf found in PATH: esp-2021r1-8.4.0.
Using Python interpreter in /opt/esp/python_env/idf5.0_py3.8_env/bin/python
Checking if Python packages are up to date...
Python requirements from /opt/esp/idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin
  /opt/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin
  /opt/esp/tools/cmake/3.20.3/bin
  /opt/esp/tools/openocd-esp32/v0.10.0-esp32-20211111/openocd-esp32/bin
  /opt/esp/python_env/idf5.0_py3.8_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

==============================================================================
=           Docker image with ESP-IDF, Rust compiler and examples            =
=              https://github.com/espressif/rust-esp32-example               =
==============================================================================

Available examples:

* "cargo-first" approach
  `` ``` ``sh
    cd /opt/rust-esp32-std-demo
    cargo +esp build --release
    espflash /dev/ttyUSB0 target/xtensa-esp32-espidf/release/rust-esp32-std-demo
    cargo pio espidf monitor -e release /dev/ttyUSB0
  `` ``` ``

* "idf.py-first" approach - integration via CMake files
  `` ``` ``sh
    cd /opt/rust-esp32-example
    idf.py build
  `` ``` ``
root@918e12cae58c:/opt# cd rust-esp32-std-demo
root@918e12cae58c:/opt/rust-esp32-std-demo# export set RUST_BACKTRACE=1 && cargo +esp build --release
warning: unused config key `unstable.configurable-env` in `/opt/rust-esp32-std-demo/.cargo/config.toml`
warning: unused config key `unstable.extra-link-arg` in `/opt/rust-esp32-std-demo/.cargo/config.toml`
   Compiling compiler_builtins v0.1.49
   Compiling core v0.0.0 (/opt/rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.103
   Compiling cc v1.0.69
   Compiling std v0.0.0 (/opt/rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
   Compiling libc v0.2.109
[... 266 lines cut ...]
  Found compatible PlatformIO Core 5.2.4 -> /root/.platformio/penv/bin/platformio

  --- stderr
  Error: Compatible PlatformIO Core not found.
  Reason: PlatformIO Core was installed using another platform `Linux-5.11.0-1021-azure-x86_64-with-glibc2.29`. Your current platform: Linux-5.10.47-linuxkit-x86_64-with-glibc2.29
  Error: invalid type: null, expected a string at line 1 column 9390

  Stack backtrace:
     0: anyhow::error::<impl core::convert::From<E> for anyhow::Error>::from
     1: <core::result::Result<T,F> as core::ops::try_trait::FromResidual<core::result::Result<core::convert::Infallible,E>>>::from_residual
     2: embuild::pio::Pio::json
     3: embuild::pio::Pio::frameworks
     4: embuild::pio::Resolver::resolve_platform_all
     5: embuild::pio::Resolver::resolve
     6: build_script_build::build_driver::build
     7: build_script_build::main
     8: core::ops::function::FnOnce::call_once
     9: std::sys_common::backtrace::__rust_begin_short_backtrace
    10: std::rt::lang_start::{{closure}}
    11: std::rt::lang_start_internal
    12: std::rt::lang_start
    13: main
    14: __libc_start_main
    15: _start
```

ivmarkov, 2022-01-18:
This is fixed in the meantime.

georgik, 2022-02-08:
Thank you @ivmarkov . Closing. Feel free to reopen if issue is still present.

closed 2022-02-08 by georgik

ivmarkov, 2022-02-08:
@georgik what I meant is that the issue is fixed in esp-idf-sys since some time, but I have no idea if we have updated the docker images since then, sorry for the confusion.

georgik, 2022-02-09:
@akselbor The image was updated with 1.58.0.0. Please pull new latest.

Note: I've noticed that `--features "native"` is not mentioned in the motd file.
Nevertheless, CI is passing with the latest version.