esp-rs/rust-build #31: Add a flag to the installation script to disable installing `cargo-espflash`

jessebraham, 2022-01-10:
I'm using the installation script from this repository for the [xtensa-toolchain](https://github.com/esp-rs/xtensa-toolchain) action, which does not need `cargo-espflash` installed. This only increases the run time of the action, which I'm trying to minimize.

It would be nice to have a flag which would allow me to disable the installation of `cargo-espflash`. I am able to do so with `ldproxy` already, for example.

jessebraham, 2022-01-11:
Perhaps even just making it one of the default values for `--extra-crates` would suffice, as that's how I'm disabling `ldproxy` presently.

georgik, 2022-02-24:
@jessebraham cargo-espflash moved to default value. If you set the value without cargo-espflash it won't install it. Version 1.59.0.0
Please test the build.

closed 2022-02-24 by georgik