esp-rs/rust-build #33: Request build for windows-gnu

materoy, 2022-01-15:
Current toolchain : stable-x86_64-pc-windows-gnu

The only available binary for windows is for msvc toolchain

georgik, 2022-02-02:
@materoy Thank you for the request. Please, could you explain the value of having also pc-windows-gnu?
MSVC toolchain performs well and is using native Windows libraries.

Also feel free to add GitHub workflow PR to the project which builds the desired toolchain. 

Note: The current build for Windows even for MSVC is little bit more challenging, because GitHub runners are too small to handle it and the build must be done locally.

materoy, 2022-02-02:
MSVC toolchain depends on visual studio installation which is not the case for minimalists, who would not like to spend thousands of megabytes on idling applications while there is an option which does't require it.
 
But yeah sure  I'll add x86_64-pc-windows-gnu workflow, and make a pr. I'll close the issue when the pr is merged 
Thank you

georgik, 2022-02-03:
Thank you @materoy for the explanation. 

It's true that having MSVC toolchain requires dependencies which could be deployed by Visual Studio Installer:
- MSVCv142 - VS2019 C++ x86/64 build tools
- Windows 10 SDK
These two dependencies requires a user to install several GB of extra software.

I was not aware of the gnu toolchain. If it can make the deployment smaller and reliable for Windows it's valuable alternative indeed and it could help us to remove one extra installation step fow Windows users: https://github.com/esp-rs/rust-build/#prerequisites-1

Let me add this topic also to Rust ESP32 Community meeting: https://github.com/esp-rs/rust/discussions/105

Please, if you know the configuration for building GNU based Windows toolchain let me know. Here is the configuration for building Windows toolchain with MSVC:
```
python3 src/bootstrap/configure.py --experimental-targets=Xtensa --enable-extended --tools=clippy,cargo,rustfmt --dist-compression-formats='xz' 
python3 x.py dist --stage 2
```

georgik, 2022-02-08:
Attempt to build the toolchain failed. I welcome any idea how to resolve the problem.

New installation of MSYS2.
Commands:
```
pacman -Su
pacman -S --needed base-devel mingw-w64-x86_64-toolchain
pacman -S git ninja python3
python3 src/bootstrap/configure.py --experimental-targets=Xtensa --enable-extended --tools=clippy,cargo,rustfmt --dist-compression-formats='xz' --host 'x86_64-pc-windows-gnu'
python3 x.py dist --stage 2
cd c:
git clone --recursive git@github.com:esp-rs/rust.git g
cd g
```

Result:
```
   Compiling serde v1.0.125
error: linker `x86_64-w64-mingw32-gcc` not found
  |
  = note: The system cannot find the file specified. (os error 2)

error: could not compile `winapi-x86_64-pc-windows-gnu` due to previous error
warning: build failed, waiting for other jobs to finish...
error: build failed
failed to run: /c/g/build/x86_64-pc-windows-gnu/stage0/bin/cargo build --manifest-path /c/g/src/bootstrap/Cargo.toml
Build completed unsuccessfully in 0:00:02
```

georgik, 2022-02-09:
Update: 
- Added compiler to PATH

Problem: 
- dist does not produce dist

```
pacman -Su
pacman -S --needed base-devel mingw-w64-x86_64-toolchain
pacman -S git ninja python3 cmake
export PATH=$PATH:/mingw64/bin
python3 x.py dist --stage 2
cd c:
git clone --recursive git@github.com:esp-rs/rust.git g
cd g
python3 src/bootstrap/configure.py --experimental-targets=Xtensa --enable-extended --tools=clippy,cargo,rustfmt --dist-compression-formats='xz' --host 'x86_64-pc-windows-gnu'
python3 x.py dist --stage 2
```

Result - build seems to end after the first step without dist directory
```
Finished dev [unoptimized] target(s) in 46.66s
Build completed successfully in 0:01:48
```

georgik, 2022-02-09:
From build Rust build docs:
```
 If you've already got git, python,
# or CMake installed and in PATH you can remove them from this list. Note
# that it is important that you do **not** use the 'python2', 'cmake' and 'ninja'
# packages from the 'msys2' subsystem. The build has historically been known
# to fail with these packages.
```
https://github.com/esp-rs/rust#mingw

After this correction the build seems to continue.

georgik, 2022-02-09:
Build succeeded. 

Build time: 2:33
Artifact size: 183 MB

Build artifact uploaded to releases: 
- https://github.com/esp-rs/rust-build/releases/tag/v1.58.0.0
- https://github.com/esp-rs/rust-build/releases/download/v1.58.0.0/rust-1.58.0.0-x86_64-pc-windows-gnu.tar.xz

georgik, 2022-02-09:
Toolchain and installation scripts are available at branch:
https://github.com/esp-rs/rust-build/tree/feature/windows-gnu-toolchain

Installation:
```
.\Install-RustToolchain.ps1 -HostTriple x86_64-pc-windows-gnu
```

Installation + testing build:
```
.\Test-RustToolchain.ps1 -HostTriple x86_64-pc-windows-gnu
```

Build of an project failed with error:
```
cargo build
error: process didn't exit successfully: `rustc -vV` (exit code: 0xc0000139, STATUS_ENTRYPOINT_NOT_FOUND)
```

georgik, 2022-04-26:
Additional research: It's not possible to build toolchain when mingw is deployed by `choco`
```
thread 'main' panicked at 'Could not find 'libiconv.a' in ["c:/programdata/chocolatey/lib/mingw/tools/install/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/", "c:/programdata/chocolatey/lib/mingw/tools/install/mingw64/bin/../lib/gcc/", "c:/programdata/chocolatey/lib/mingw/tools/install/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/lib/x86_64-w64-mingw32/11.2.0/", "c:/programdata/chocolatey/lib/mingw/tools/install/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/lib/../lib/", "c:/programdata/chocolatey/lib/mingw/tools/install/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../x86_64-w64-mingw32/11.2.0/", "c:/programdata/chocolatey/lib/mingw/tools/install/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../lib/", "/mingw/lib/x86_64-w64-mingw32/11.2.0/", "/mingw/lib/../lib/", "c:/programdata/chocolatey/lib/mingw/tools/install/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/lib/", "c:/programdata/chocolatey/lib/mingw/tools/install/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../", "/mingw/lib/"]', src\bootstrap\dist.rs:127:13
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Build completed unsuccessfully in 0:01:17
```

georgik, 2022-07-19:
Added experimental package to 1.62.1.0 pre-release.
The dist build of the package failed with:
```
Dist extended stage1 (x86_64-pc-windows-gnu)
Dist rust-1.62.1-dev-x86_64-pc-windows-gnu
        finished in 901.635 seconds
thread 'main' panicked at 'could not read dir "C:\\r\\build\\tmp\\tarball\\rust\\x86_64-pc-windows-gnu\\rust-analysis-1.62.1-dev-x86_64-pc-windows-gnu\\rust-analysis-x86_64-pc-windows-gnu": Os { code: 3, kind: NotFound, message: "The system cannot find the path specified." }', src\bootstrap\lib.rs:1547:25
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Build completed unsuccessfully in 4:04:54
```
We will perform tests on this artifact. The setup required a clean VM machine and tools installed via chocolatey.
Branch with changes: https://github.com/esp-rs/rust-build/tree/feature/windows-gnu-toolchain

georgik, 2022-07-21:
The build error fixed by:
```
choco install msys2 -y
cd C:\tools\msys64\
msys2_shell.cmd -mingw64
```
Then in MSY2 shell:
```
pacman -Sy pacman-mirrors
pacman -s mingw-w64-x86_64-python             mingw-w64-x86_64-cmake             mingw-w64-x86_64-gcc             mingw-w64-x86_64-ninja
```
Then back in PowerShell:
```
$env:PATH+=";C:\tools\msys64\mingw64\bin"
cargo +esp build --release
```
Build failed:
```
Some errors have detailed explanations: E0405, E0463.
For more information about an error, try `rustc --explain E0405`.
error[E0463]: can't find crate for `core`
   --> ....\.cargo\registry\src\github.com-1ecc6299db9ec823\nb-1.0.0\src\lib.rs:189:5
    |
189 | use core::fmt;
    |     ^^^^ can't find crate
```
This indicates problem with manual deployment of core.

georgik, 2022-07-21:
No, it's not possible to use  just self contained gcc. 
Path correction:
```
$env:PATH+=";........\rust-1.62.1-dev-x86_64-pc-windows-gnu\rust-mingw\lib\rustlib\x86_64-pc-windows-gnu\bin\self-contained"
``
Failed with:
```
\\esp\\lib\\rustlib\\x86_64-pc-windows-gnu\\lib\\rsend.o"
  = note: ld: cannot find crt2.o: No such file or directory
          ld: cannot find crtbegin.o: No such file or directory

error: could not compile `proc-macro2` due to previous error
```

georgik, 2022-07-21:
Using cargo from dist package, instead of from rustup:
```
$env:PATH="C:\......\Desktop\rust-1.62.1-dev-x86_64-pc-windows-gnu\cargo\bin;$env:PATH"
```

Failed on missing git when cloning ESP-IDF. Fixed by:
```
choco install git
$env:PATH+=";C:\Program Files\Git\bin"
```

georgik, 2022-07-21:
Failed on missing pip, because it was not using environment deployed by idf-installer
```
  C:\tools\msys64\mingw64\bin\python.exe: No module named pip
  Error: Could not install esp-idf
```
Fixed by:
```
Invoke-WebRequest https://bootstrap.pypa.io/get-pip.py -OutFile get-pip.py
python get-pip.py
```

Failed with:
```
     self._execute_child(args, executable, preexec_fn, close_fds,
    File "C:\tools\msys64\mingw64\lib\python3.10\subprocess.py", line 1438, in _execute_child
      hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
  FileNotFoundError: [WinError 2] The system cannot find the file specified
  Error: Could not install esp-idf

  Caused by:
      command '"python" "C:\\g\\.embuild\\espressif\\esp-idf\\release-v4.4\\tools\\idf_tools.py" "--idf-path" "C:\\g\\.embuild\\espressif\\esp-idf\\release-v4.4" "--quiet" "--non-interactive" "install-python-env"' exited with non-zero status code 1
```

MSYS Python 3.10. does not seem to be compatible with ESP-IDF.

georgik, 2022-07-21:
Install ESP-IDF using Windows installer to avoid collision with MSYS tools and in activate ESP-IDF PowerShell start build with
```
$env:PATH="C:\....\Desktop\rust-1.62.1-dev-x86_64-pc-windows-gnu\cargo\bin;$env:PATH"
$env:PATH+=";C:\tools\msys64\mingw64\bin"
cargo build
```
Failed on missing LLVM.
```
 thread 'main' panicked at 'Unable to find libclang: "couldn't find any valid shared libraries matching: ['clang.dll', 'libclang.dll'], set the `LIBCLANG_PATH` environment variable to a path where one of these files can be found (i
```
Added LLVM from https://github.com/esp-rs/rust-build/releases/tag/llvm-project-14.0-minified
```
$env:PATH+=";C:\Espressif\tools\xtensa-esp32-elf-clang\bin"
$env:LIBCLANG_PATH="C:\Espressif\tools\xtensa-esp32-elf-clang\bin\libclang.dll"
```
Failed on missing ldproxy
```
cargo install ldproxy
```
Result: success

georgik, 2022-07-21:
@materoy Please, can you test the following instructions on how to get the toolchain working with GNU based toolchain?
https://github.com/esp-rs/rust-build/tree/feature/windows-gnu-toolchain#windows-x86_64-gnu
The updated installer is on the `feature/windows-gnu-toolchain` branch.

georgik, 2022-07-22:
Thank you for requesting the feature. It was merged and released with 1.62.1.0.
Please follow README instructions for the deployment.

closed 2022-07-22 by georgik

materoy, 2022-07-22:
Thank you @georgik, for your work.

PR #123 (2022-07-22): Feature/windows gnu toolchain
Add support for Windows Rust GNU toolchain

Closes #33