esp-rs/rust-build #38: Check cargo-espflash version during isntallation

georgik, 2022-02-16:
Check whether cargo-espflash has new version.
cargo install might fail on following error which results in keeping previous version e.g. v1.1.0
```
cargo install cargo-espflash
    Updating crates.io index
  Downloaded cargo-espflash v1.2.0
  Downloaded 1 crate (23.7 KB) in 0.35s
error: binary `cargo-espflash.exe` already exists in destination
Add --force to overwrite
```

S3j5b0, 2022-02-24:
Hi. I am having this issue, it looks like this:

I'm very unsure about what to do. 
I have tried removing the package like so:

```
$ cargo uninstall cargo-espflash
error: package ID specification `cargo-espflash` did not match any packages
```
But then it can't find it.

MabezDev, 2022-02-25:
Seems to work for me?
```
cargo uninstall cargo-espflash
    Removing /home/mabez/.cargo/bin/cargo-espflash
```

Possible a bug in a specific cargo verson? I'm running cargo 1.60.

S3j5b0, 2022-02-25:
hmm, I'm running `1.58.1` 
I'm gonna try updating and reevaluate

SergioGasquez, 2022-03-01:
Just ran into the same issue with `1.59.0`

georgik, 2022-03-01:
Thank you for confirming the problem. The installer should be more resilient and it should handle this situation.

jessebraham, 2022-03-01:
Just as one more data point, I ran into this today as well. While building the following `Dockerfile`:

```Dockerfile
FROM espressif/idf

ENV LC_ALL=C.UTF-8
ENV LANG=C.UTF-8
ENV RUSTUP_HOME=/opt/rustup
ENV CARGO_HOME=/opt/cargo
ENV PATH=/opt/cargo/bin:/opt/rustup/bin:/opt/esp/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/bin/:$PATH

RUN curl -LO https://raw.githubusercontent.com/esp-rs/rust-build/main/install-rust-toolchain.sh \
 && chmod +x ./install-rust-toolchain.sh \
 && ./install-rust-toolchain.sh \
 && rm -f ./install-rust-toolchain.sh
```

Results in:

```text
#5 318.3  Downloading crates ...
#5 318.7   Downloaded cargo-espflash v1.3.0
#5 318.8 error: binary `cargo-espflash` already exists in destination
#5 318.8 Add --force to overwrite

...

#5 433.5     Finished release [optimized] target(s) in 4m 56s
#5 433.5   Installing /opt/cargo/bin/ldproxy
#5 433.5    Installed package `ldproxy v0.3.0` (executable `ldproxy`)
#5 433.6      Summary Successfully installed ldproxy! Failed to install cargo-espflash (see error(s) above).
#5 433.6 error: some crates failed to install
------
Dockerfile:9
--------------------
   8 |     
   9 | >>> RUN curl -LO https://raw.githubusercontent.com/esp-rs/rust-build/main/install-rust-toolchain.sh \
  10 | >>>  && chmod +x ./install-rust-toolchain.sh \
  11 | >>>  && ./install-rust-toolchain.sh \
  12 | >>>  && rm -f ./install-rust-toolchain.sh
  13 |     
--------------------
error: failed to solve: process "/bin/sh -c curl -LO https://raw.githubusercontent.com/esp-rs/rust-build/main/install-rust-toolchain.sh  && chmod +x ./install-rust-toolchain.sh  && ./install-rust-toolchain.sh  && rm -f ./install-rust-toolchain.sh" did not complete successfully: exit code: 101
FATA[0559] unrecognized image format 
```

georgik, 2022-03-07:
Workaround - skip cargo-espflash: `./install-rust-toolchain.sh --extra-crates "ldproxy"`

georgik, 2022-03-07:
@MabezDev found the reason.

When `cargo-espflash` is downloaded as binary artifact from GitHub releases and placed to `.cargo/bin` then cargo does not track it and subsequent installation fails.

Here is how to simulate the problem:
```
cargo uninstall cargo-espflash
touch ~/.cargo/bin/cargo-espflash
cargo install cargo-espflash
    Updating crates.io index
error: binary `cargo-espflash` already exists in destination
Add --force to overwrite
```
Exit code 101

PR #45 (2022-03-09): ci: install cargo-espflash from binary
Closes #38
closed 2022-03-09 by georgik