esp-rs/rust-build #44: Error compiling

DavidCarl, 2022-03-03:
Hello.

I am currently trying to use the library [x25519-dalek](https://crates.io/crates/x25519-dalek) for a project that needs to run on a ESP32. This means that we have no operating system and no STD. But the library should be in the category "No standard library" which should be fine for me in this case.

I am using this build chain, however it decides to give me an error.

But when attempting to build it I recieve the following error:

```
error: linking with `ldproxy` failed: exit status: 1
...
  = note: Running ldproxy
          Error: Linker /home/carl/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/bin/xtensa-esp32-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /home/carl/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/carl/Documents/git/github/esp-rs/esp32-build/target/xtensa-esp32-espidf/debug/deps/libclear_on_drop-7f6bb154f7984c2b.rlib: error adding symbols: file format not recognized
          collect2: error: ld returned 1 exit status
```

[full log here](https://gist.github.com/DavidCarl/232c59a5911c8623569c0390470944c4)

And the project I am running and failing the build is [this](https://github.com/DavidCarl/esp32-test) - The following [line](https://github.com/DavidCarl/esp32-test/blob/main/src/main.rs#L55) is what triggers this error 

Hope this is something you can help me with.

igrr, 2022-03-03:
Could you try running
* `file /home/carl/Documents/git/github/esp-rs/esp32-build/target/xtensa-esp32-espidf/debug/deps/libclear_on_drop-7f6bb154f7984c2b.rlib`
* `/home/carl/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/bin/xtensa-esp32-elf-objdump -h /home/carl/Documents/git/github/esp-rs/esp32-build/target/xtensa-esp32-espidf/debug/deps/libclear_on_drop-7f6bb154f7984c2b.rlib`

to see what's wrong with the static library the linker is complaining about in the error message?

MabezDev, 2022-03-03:
Sometimes I've seen this error when a crate includes assembly that is not meant for the current target, but a quick look at https://github.com/dalek-cryptography/x25519-dalek seems its pure Rust.

DavidCarl, 2022-03-03:
The first command

```
file /home/carl/Documents/git/github/esp-rs/esp32-build/target/xtensa-esp32-espidf/debug/deps/libclear_on_drop-7f6bb154f7984c2b.rlib
/home/carl/Documents/git/github/esp-rs/esp32-build/target/xtensa-esp32-espidf/debug/deps/libclear_on_drop-7f6bb154f7984c2b.rlib: current ar archive
```

The second command

```
/home/carl/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/bin/xtensa-esp32-elf-objdump -h /home/carl/Documents/git/github/esp-rs/esp32-build/target/xtensa-esp32-espidf/debug/deps/libclear_on_drop-7f6bb154f7984c2b.rlib
In archive /home/carl/Documents/git/github/esp-rs/esp32-build/target/xtensa-esp32-espidf/debug/deps/libclear_on_drop-7f6bb154f7984c2b.rlib:

clear_on_drop-7f6bb154f7984c2b.clear_on_drop.61694cba-cgu.0.rcgu.o:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00000000  00000000  00000000  00000034  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .literal._ZN13clear_on_drop21clear_stack_on_return11clear_stack17h3ca4c7fc9541b1efE 00000010  00000000  00000000  00000034  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  2 .text._ZN13clear_on_drop21clear_stack_on_return11clear_stack17h3ca4c7fc9541b1efE 0000002d  00000000  00000000  00000044  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  3 .debug_abbrev 000000e5  00000000  00000000  00000071  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  4 .debug_info   000001bb  00000000  00000000  00000156  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
  5 .debug_aranges 00000020  00000000  00000000  00000311  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
  6 .debug_str    0000028f  00000000  00000000  00000331  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  7 .debug_pubnames 000000fb  00000000  00000000  000005c0  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
  8 .debug_pubtypes 00000064  00000000  00000000  000006bb  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
  9 .note.GNU-stack 00000000  00000000  00000000  0000071f  2**0
                  CONTENTS, READONLY
 10 .debug_frame  00000028  00000000  00000000  00000720  2**2
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
 11 .debug_line   000000da  00000000  00000000  00000748  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
/home/carl/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/bin/xtensa-esp32-elf-objdump: hide.o: file format not recognized
/home/carl/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/bin/xtensa-esp32-elf-objdump: lib.rmeta: file format not recognized
```

I hope this makes more sense for you than for me :)

S3j5b0, 2022-03-07:
Hi, does someone know about this? I have the same issue trying to compile the same resource?

igrr, 2022-03-07:
This seems to be coming from https://docs.rs/clear_on_drop/latest/clear_on_drop/ which has this bit not compatible with Xtensa: https://github.com/cesarb/clear_on_drop/blob/master/src/hide.c#L6-L11