esp-rs/rust-build #46: Stack overflow, despite not doing anything crazy?

S3j5b0, 2022-03-12:
Hi, me and my partner have been trying to run our project on a esp32 wroom.

So far, w're trying to run [this](https://github.com/DavidCarl/esp32-test/blob/main/src/main.rs) project. There is a certain line, where one of our dependencies use serde_cbor to serialize an object, the function looks like [this](https://github.com/DavidCarl/oscore/blob/master/src/edhoc/api.rs#L81):

```

    pub fn generate_message_1(
        self,
        r#type: isize,
        suites: isize,
    ) -> Result<(Vec<u8>, PartyI<Msg2Receiver>), EarlyError> {
        // Encode the necessary informati'on into the first message
        let msg_1 = Message1 {
            r#type,
            suite: suites,
            x_i: self.0.x_i.as_bytes().to_vec(), // sending PK as vector
            c_i: self.0.c_i,
        };
        // Get CBOR sequence for message
        let msg_1_seq = util::serialize_message_1(&msg_1)?;
        // Copy for returning
        let msg_1_bytes = msg_1_seq.clone();

        Ok((
            msg_1_bytes,
            PartyI(Msg2Receiver {
                i_ecdh_ephemeralsecret: self.0.secret,
                stat_priv: self.0.static_secret,
                stat_pub: self.0.static_public,
      //          auth: self.0.auth,
                kid: self.0.kid,
                msg_1_seq,
            }),
        ))
    }
}
```

And the serialization function that is lastly called looks like this:
```

fn serialize(object: impl Serialize, offset: usize) -> Vec<u8> {
    // Serialize to byte vector
    let mut v = serde_cbor::to_vec(&object).expect("error during serialization");
    // Return everything starting from the offset
    v.drain(offset..).collect()
}
```

I might not be super helpfull to include the code, but I thought I would do it anyway.

Anyways, this runs with no issues on our own machines, but as soon as we try to flash it onto our ESP32 wroom, this happens:

```
***ERROR*** A stack overflow in task main has been detected.
Backtrace:0x40081a36:0x3ffb63d00x40084d21:0x3ffb63f0 0x40087476:0x3ffb6410 0x40086309:0x3ffb6490 0x40084e20:0x3ffb64c0 0x40084dd2:0x3ffb64f0 0x400e9735:0x00000004  |<-CORRUPTED
ELF file SHA256: 0000000000000000
```

We should have 520KB disposable, and in my mind, what we're doing here should definitely not overflow that.

Is there any configurations that one can do to access more ram on the device? or what is this likely to be caused by? it seems very unlikely to me that what were doing now overfills 520 kb

MabezDev, 2022-03-13:
You have total of 520kb of RAM, but the default thread stack will be much lower. You can increase the main task stack via https://github.com/ivmarkov/rust-esp32-std-demo/blob/b5c88c11be381e6a19d85694954666f319fa827f/sdkconfig.defaults#L2 and you can increase the stack size of created threads with the `ThreadBuilder` API of the standard library.

closed 2022-03-13 by MabezDev

S3j5b0, 2022-03-21:
@MabezDev  Hi again I now programmed a bit more on this thing. My code is [here](https://gist.github.com/S3j5b0/eaaaf43fa3e1b477ab50b7fac10b63d6), it's using tcp to do a little handhsake
Upping the stack size to 9000 fixed the issue, but Now I added tcp to the mix, so i'ts not just doing the handshake locally.

Now it overflows on the very last line, and however much I increase the stack, nothing helps
Why is this?isnt the stack overflow just an issue of stack space?