esp-rs/rust-build #50: set release-channel=stable for builds

MabezDev, 2022-03-22:
We should set `--release-channel=stable` when configuring our compiler builds. I.e `./configure --experimental-targets=Xtensa --release-channel=stable`

Before:
```rust
rustc +esp-dev --version --verbose
rustc 1.59.0 (04e60933c 2022-03-14)
binary: rustc
commit-hash: unknown
commit-date: unknown
host: x86_64-unknown-linux-gnu
release: 1.59.0
LLVM version: 13.0.0
```

After:

```rust
rustc +esp-dev --version --verbose
rustc 1.59.0 (04e60933c 2022-03-14)
binary: rustc
commit-hash: 04e60933cbe54af30d339efd67da6b43016433ad
commit-date: 2022-03-14
host: x86_64-unknown-linux-gnu
release: 1.59.0
LLVM version: 13.0.0
```

MabezDev, 2022-03-24:
Updating this to mention that we should set the channel to `nightly`. Even though we target stable with our releases, we need unstable features for most of our ecosystem hence we need to allow that with the `nightly` release channel.

georgik, 2022-04-07:
Name of all artifacts in dist directory changed to nightly and it does not contain version anymore.
E.g. rustc-nightly-x86_64-pc-windows-msvc.tar.gz
It's necessary to update also upload scripts to map the name

georgik, 2022-04-08:
Solved in https://github.com/esp-rs/rust-build/releases/tag/v1.60.0.0

closed 2022-04-08 by georgik