esp-rs/rust-build #51: Update of toolchain

georgik, 2022-04-04:
Scenario: User deployed toolchain 1.56.0.0 and later on user wants upgrade to 1.59.0.0
The installation script does not display any warning an quits.

Workaround:
- run installation with parameter `-i`: `./install-rust-toolchain.sh -i reinstall`
- or delete ~/.rustup/toolchains/esp

Expected behavior: Installer allows updated of the toolchain in the similar way like rustup

stappersg, 2022-07-16:
> Expected behavior: Installer allows updated of the toolchain in the similar way like rustup

Thing I would like to know: Where is _merge with upstream `rustup`_ being tracked?

georgik, 2022-08-22:
@stappersg There is an open request for "custom channels" for rustup. That would solve the problem: https://github.com/rust-lang/rustup/issues/2696

georgik, 2023-10-19:
It does not seems lilely to fix this limitation. Closing the issue.

closed 2023-10-19 by georgik