esp-rs/rust-build #54: Incomplete installation instructions and script issue

turgu1, 2022-04-08:
Hello,

First, thanks a lot for the effort to get Rust working for Espressif devices. These are suggestions to have the installation process a bit smoother.

1) Installation instructions

From the installation instruction, it is not clear that the ESP-IDF framework needs to be installed and which version is supported. Please add this information in the Prerequisites in the README.

2) End of script update to .zshrc

I'm using Ubuntu 20.4. Bash is the standard shell. As there is no .zshrc in the user's main folder, the updates on PATH and other env variables are not done. It would be handy to have the .bashrc and .profile automatically updated in this context. Or say so in some output text to help newbies like me do it by hand.

Thanks again!
Guy

georgik, 2022-04-12:
Thank you for reporting the problem @turgu1 

Regarding the first problem: It's not necessary to install ESP-IDF. Cargo with Embuild will pull ESP-IDF into .embuild directory. The version of ESP-IDF could be controlled by environment variable `ESP_IDF_VERSION` (like here https://github.com/esp-rs/rust-build/blob/main/test-rust-toolchain.sh#L19)

Regarding shell configuration. Good point, the script could be able to detect preferred shell and offer instructions to store values in `.bashrc` in case of bash.

@SergioGasquez Please, could you review the installation instructions so that it makes more sense for Linux users. It might be also worthy to mention DevContainers so that a user does not need to install the toolchain at all.

turgu1, 2022-04-12:
Guys, thanks!!!

georgik, 2022-04-19:
@turgu1 Please review current version of README. Thank you.

turgu1, 2022-04-20:
Hello georgik,

Thanks for the changes to the README.
I have tried to use the procedure on a fresh Ubuntu Linux VM. Here are some results and  other changes that would enhance the README file:

1. The shell needs to be restarted after the update to the shell startup script with the exported variables. Maybe a good idea to say so in the README.

2. Replace the installation command:

```
./install-rust-toolchain.sh
```

with:

```
$ git clone https://github.com/esp-rs/rust-build.git
$ cd rust-build
$ ./install-rust-toolchain.sh
```

3. I have cloned the rust-esp32-std-demo app. and tried to use the `cargo espflash` command. Does not work if access to the USB device is not allowed using the following command followed by a reboot:

```
$ sudo usermod -a -G dialout <username>
```

4. Starting again the `cargo espflash` command, after returning to the proper folder. I've got the following error:

```
turgu1@ubuntu:~/Dev/rust-esp32-std-demo$ cargo espflash --target xtensa-esp32-espidf /dev/ttyUSB0
Serial port: /dev/ttyUSB0
Connecting...

Unable to connect, retrying with extra delay...

error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --target xtensa-esp32-espidf --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (exit status: 1)
  --- stderr
  error: Error loading target specification: Could not find specification for target "xtensa-esp32-espidf". Run `rustc --print target-list` for a list of built-in targets
```

5. I then used the following command:

```
$ rustup default esp
```

The cargo espflash did worked. 

Hope this will help to update the README file.
Cheers!
Guy

jhaand, 2022-05-06:
The `rustup default esp` should be added to the instructions.

georgik, 2022-05-24:
@SergioGasquez Please review suggested solution with setting toolchain `esp` as default (PR #74)

IMO we should not force a user to change default toolchain to esp. Esp toolchain is custom toolchain. Some tooling like CLion might have problem when overriding the toolchain using `rustup default`

There are two other solutions which might be closer to Rust way:
- use `+esp` in the command line when building using `cargo` which will add custom toolchain on top of default one: `cargo +esp build`
- specify the toolchain name in `rust-toolchain.toml`

More details: https://rust-lang.github.io/rustup/overrides.html

SergioGasquez, 2022-05-24:
Hello! 

Yes, using `esp` as the default toolchain solves the "issue" but I agree with @georgik's opinion about enforcing users to use `esp` as default toolchain. 

I created #75 to improve documentation on toolchain overriding.

PR #56 (2022-04-19): Feature/improve documentation
- Improve Documentation:
   - Installation 
   - Add building steps for cargo-espflash
- Improve install script:
   - Add `--help` argument to print information about options
   - Add logic to detect which shell is the user using

Closes #54
closed 2022-05-27 by georgik