esp-rs/rust-build #59: If `$CARGO_HOME` is set, installation script tries to source potentially non-existent file

svenstaro, 2022-04-23:
The script currently does this:
```sh
        if [ ! -z "${CARGO_HOME}" ]; then
            source ${CARGO_HOME}/env
        else
```
So it assumes that `${CARGO_HOME}/env` exists just because `${CARGO_HOME}` is set. This crashes my current environment and I have to create a `env` file just for this script not to choke. It'd be nice if the script also checked for actual existence of that file before blindly sourcing it.

closed 2022-04-28 by georgik

mariansam, 2022-10-29:
Hi, in my environment this issues still persists. The [`source_cargo`](https://github.com/esp-rs/rust-build/blob/main/install-rust-toolchain.sh#L160) function tries to source the `env` file either from default `~/.cargo/env` or from special `$CARGO_HOME/env`. However if `$CARGO_HOME` is set, BUT the `env` file doesn't exist, it resets `$CARGO_HOME` to the default directory (`~/.cargo`) for some reason (see [line 169](https://github.com/esp-rs/rust-build/blob/main/install-rust-toolchain.sh#L169), I think that line should be removed).

SergioGasquez, 2022-10-31:
Hi @mariansam! Can you try reproducing the issue with [`espup`](https://github.com/esp-rs/espup)?

PR #60 (2022-04-28): Check for existence of ${CARGO_HOME}/env
This fixes #59.
commit de5ffe6e (2022-04-28): Check for existence of ${CARGO_HOME}/env (#60)

This fixes #59.