esp-rs/rust-build #61: Add CC and AR variables of compiler name to generated export script

svenstaro, 2022-04-24:
The variable export script currently has this content more or less:
```shell
export PATH="/home/svenstaro/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/bin/:$PATH"
export LIBCLANG_PATH="/home/svenstaro/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/lib/"
export PIP_USER=no
```
This will currently make `build.rs`-driven cross-compilation impossible since the `CC` and `AR` are unset and so a crate such as `cc` will still use the host's regular `cc` and `ar` commands.

Consider this simple `build.rs`:
```rust
fn main() -> anyhow::Result<()> {
    println!("cargo:rerun-if-changed=c/hello.c");
    println!("cargo:rerun-if-changed=target/export-esp-rust.sh");
    cc::Build::new()
        .file("c/hello.c")
        .compile("hello");

    embuild::build::CfgArgs::output_propagated("ESP_IDF")?;
    embuild::build::LinkArgs::output_propagated("ESP_IDF")
}
```
This won't compile properly as it stands right now because of the aforementioned reason. I suggest expanding the exported variables to include all of the cross compilation toolchain tools.

In fact, I wonder why this is not being done and whether I'm missing something here.

Consider this:
```shell
export CC="xtensa-esp32-elf-gcc"
export AR="xtensa-esp32-elf-ar"
export PATH="/home/svenstaro/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/bin/:$PATH"
export LIBCLANG_PATH="/home/svenstaro/.espressif/tools/xtensa-esp32-elf-clang/esp-13.0.0-20211203-x86_64-unknown-linux-gnu/lib/"
export PIP_USER=no
```
Would allow you to compile and link most simple C programs directly and without further hassle. What do you think?

georgik, 2022-11-10:
I would like to add also g++ there.

Something line this for RISC-V target:
```
export CC=riscv32-esp-elf-gcc
export CXX=riscv32-esp-elf-g++
```

This might help with building dependencies like fltk-rs.

@SergioGasquez Might be useful to consider this in context of espup.

jasta, 2023-01-24:
I'm hitting this as well and created this sample repo to test whether I was going crazy: https://github.com/jasta/esp-cc-broken.  This prevents me from building lvgl-rs and I'd suspect virtually any other crate that uses the cc (or maybe even bindgen?) crate.

jasta, 2023-01-24:
The solution is definitely not to add CC/CXX though as this is likely to break quite a few other things, but rather we need to figure out how this is supposed to be done nominally so we can build crates that use the cc/bindgen crates without special awareness of embuild.  I'd wager the problem exists somewhere in the space of all the random big lists of target triples you see scattered about various tools.  For example: https://github.com/rust-lang/cc-rs/blob/main/src/lib.rs#L2883

Right now I can't quite figure out what this is all _supposed_ to be.  Rust's story around cross compilation is so new and changing rapidly that it's very hard to tell what is the right thing to do...