esp-rs/rust-build #68: Ubuntu 22.04 is using openSSL 3.0 as default

klenkiven, 2022-05-17:


georgik, 2022-05-20:
@klenkiven thank you for reporting the issue. We will check options for the deployment.
@SergioGasquez could you verify it with Dev Container by switching base image to Ubuntu 22.04?

SergioGasquez, 2022-05-20:
I was able to reproduce the error using the following Dockerfile:

```dockerfile
ARG VARIANT=22.04
FROM ubuntu:${VARIANT}
ENV DEBIAN_FRONTEND=noninteractive
ENV LC_ALL=C.UTF-8
ENV LANG=C.UTF-8
# Arguments
ARG CONTAINER_USER=esp
ARG CONTAINER_GROUP=esp
ARG NIGHTLY_TOOLCHAIN_VERSION=nightly
ARG XTENSA_TOOLCHAIN_VERSION=1.61.0.0
ARG ESP_IDF_VERSION=release/v4.4
ARG ESP_BOARD=esp32
ARG INSTALL_RUST_TOOLCHAIN=install-rust-toolchain.sh
# Install dependencies
RUN apt-get update \
    && apt-get install -y git curl gcc clang ninja-build libudev-dev unzip xz-utils \
    python3 python3-pip libusb-1.0-0 libssl-dev pkg-config libtinfo5  libpython2.7 \
    && apt-get clean -y && rm -rf /var/lib/apt/lists/* /tmp/library-scripts
# Set user
RUN adduser --disabled-password --gecos "" ${CONTAINER_USER}
USER ${CONTAINER_USER}
WORKDIR /home/${CONTAINER_USER}
# Install rust toolchain(s) with extra crates
ENV PATH=${PATH}:/home/${CONTAINER_USER}/.cargo/bin:/home/${CONTAINER_USER}/opt/bin
# Use LLVM installer
ADD --chown=${CONTAINER_USER}:${CONTAINER_GROUP} \
    https://github.com/esp-rs/rust-build/releases/download/v${XTENSA_TOOLCHAIN_VERSION}/${INSTALL_RUST_TOOLCHAIN} \
    /home/${CONTAINER_USER}/${INSTALL_RUST_TOOLCHAIN}
RUN chmod a+x ${INSTALL_RUST_TOOLCHAIN} \
    && ./${INSTALL_RUST_TOOLCHAIN} \
    --extra-crates "ldproxy cargo-espflash" \
    --clear-cache "YES" \
    --build-target ${ESP_BOARD} \
    --nightly-version ${NIGHTLY_TOOLCHAIN_VERSION} \
    --esp-idf-version ${ESP_IDF_VERSION} \
    --minified-esp-idf "YES" \
    --export-file /home/${CONTAINER_USER}/export-esp.sh
# Activate ESP-IDF and Xtensa Rust toolchain environment
RUN echo "source /home/${CONTAINER_USER}/export-esp.sh > /dev/null 2>&1" >> ~/.bashrc
```

And then try to compile [`esp-idf-template`](https://github.com/esp-rs/esp-idf-template) project:
``` bash
$ cargo install cargo-generate && cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template cargo --name rust-project --define mcu=esp32 --define toolchain=esp --define espidfver=v4.4 --define std=true && cd rust-project && cargo +esp build
```

Threadzless, 2022-06-03:
One solution to this issue is to install openssl v1.1 alongside v3.

```bash
wget https://www.openssl.org/source/openssl-1.1.1o.tar.gz
tar xf openssl-1.1.1o.tar.gz
cd openssl-1.1.1o
./config
make
sudo make install
echo 'export LD_LIBRARY_PATH="/usr/local/lib:/usr/local/lib64"' >> ~/.bashrc
source ~/.bashrc
```

Of course, making the tool chain work with either version of openssl would be more elegant, but I imagine that will take some time.

georgik, 2022-07-22:
@Threadzless Thank you for suggesting the workaround. 
If Ubuntu guys don't fix the linking issue till 1.63, we will add the correction to the installation script.

georgik, 2022-08-22:
@Threadzless we applied solution of similar problem from mac #128 suggested by @stevecheckoway . It seems that it solved the problem also for Linux.

Added flag to build `--enable-cargo-native-static`: https://github.com/esp-rs/rust-build/pull/135/commits/90baaec74437d6fb8cb2b4c4209e3c2177dd96ab

Please, test 1.63.0.1 release: https://github.com/esp-rs/rust-build/releases/tag/v1.63.0.1

closed 2022-08-22 by georgik