esp-rs/rust-build #70: Support for ESP-IDF v5.0 CI

georgik, 2022-05-17:


georgik, 2022-07-22:
Blocked by https://github.com/esp-rs/esp-idf-sys/issues/99

brianredbeard, 2022-09-15:
Upstream blocker is currently closed.

georgik, 2022-09-23:
5.1 with activated ESP-IDF fails. Created issue #150 

Test with 5.0:
```
git clone --branch v5.0-beta1 --recursive --depth 1 --shallow-submodules git@github.com:espressif/esp-idf.git esp-idf-v5.0-beta1
cd esp-idf-v5.0-beta1
./install.sh
source ./export.sh
cd ../esp-idf-rs-test
cargo espflash --release
```

Build with activated environment failed:
```

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.0",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/Users/rust/projects/esp-idf-v5.0-beta1",
          ),
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using activated esp-idf v5.0.0 environment at '/Users/rust/projects/esp-idf-v5.0-beta1'
  ERROR: /Users/rust/projects/esp-idf-rs-test/.embuild/espressif/python_env/idf5.0_py3.9_env/bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it
  CMake Error at /Users/rust/projects/esp-idf-v5.0-beta1/tools/cmake/build.cmake:346 (message):
    Some Python dependencies must be installed.  Check above message for
    details.
  Call Stack (most recent call first):
    /Users/rust/projects/esp-idf-v5.0-beta1/tools/cmake/build.cmake:484 (__build_check_python)
    CMakeLists.txt:14 (idf_build_process)

  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /Users/rust/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.48/src/lib.rs:975:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

georgik, 2022-09-23:
Without activated environment there is a failure:
```
  Using esp-idf v5.1.0 at '/Users/georgik/projects/esp-idf-rs-test/.embuild/espressif/esp-idf/master'
  ERROR: tool cmake does not have versions compatible with platform macos-arm64
  Error: Could not install esp-idf
```

Both tests were done on M1.

@n3xed Please, could you check why is embuild expecting python in .embuild directory when the build was started in activated ESP-IDF environment. The problem mentioned in post above.

The error with cmake looks strange, because cmake is registered there also for macos-arm64 `tools.json`: https://github.com/espressif/esp-idf/blob/master/tools/tools.json#L642

@SergioGasquez Please, perform similar test on Linux.

SergioGasquez, 2022-09-23:
Here are my test results on Linux:
Generate command: 
```sh
cargo generate --git https://github.com/esp-rs/esp-idf-template cargo --name test50 --vcs none --silent -d mcu=esp32 -d std=true -d espidfver=mainline -d devcontainer=false
```
Then I edited `ESP_IDF_VERSION` to `v5.0-beta1` in `.cargo/config.toml` file
- Using activated environment with esp-idf `v5.0-beta1` branch:
```sh
 --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.0",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/home/sergio/.espressif/frameworks/esp-idf-v5.0-beta1",
          ),
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using activated esp-idf v5.0.0 environment at '/home/sergio/.espressif/frameworks/esp-idf-v5.0-beta1'
  ERROR: /home/sergio/Documents/Espressif/tests/test50/.embuild/espressif/python_env/idf5.0_py3.10_env/bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it
  CMake Error at /home/sergio/.espressif/frameworks/esp-idf-v5.0-beta1/tools/cmake/build.cmake:346 (message):
    Some Python dependencies must be installed.  Check above message for
    details.
  Call Stack (most recent call first):
    /home/sergio/.espressif/frameworks/esp-idf-v5.0-beta1/tools/cmake/build.cmake:484 (__build_check_python)
    CMakeLists.txt:14 (idf_build_process)

  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.48/src/lib.rs:975:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
Same result as @georgik had on MacM1.

N3xed, 2022-09-24:
> ```
>   Using activated esp-idf v5.0.0 environment at '/Users/rust/projects/esp-idf-v5.0-beta1'
>   ERROR: /Users/rust/projects/esp-idf-rs-test/.embuild/espressif/python_env/idf5.0_py3.9_env/bin/python [...]

This problem should be fixed by commit esp-rs/esp-idf-sys@e85717dfc06566c0288613c1961b9804e02dd5c5 (see commit message for why it happened).

On tag `v5.0-beta1` there is still the problem with the duplicate `i2s_channel_t` definition which I've first mentioned [here](https://github.com/esp-rs/esp-idf-sys/issues/99#issuecomment-1209830817).
 This was already fixed in the `release/v5.0` branch by commit https://github.com/espressif/esp-idf/commit/0f0baa5bb4a1ac474956d491ffbb40338a5dea37. So I'd suggest using that branch for now.

<details>
<summary>Dupplicate i2s_channel_t error</summary>

```
error[E0428]: the name `i2s_channel_t` is defined multiple times
     --> C:\Users\n3xed\dev\esp-idf-test\target\riscv32imc-esp-espidf\debug\build\esp-idf-sys-9c6a2e480ecf6322\out\bindings.rs:68418:1
      |
68342 | pub struct i2s_channel_t {
      | ------------------------ previous definition of the type `i2s_channel_t` here
...
68418 | pub type i2s_channel_t = c_types::c_int;
      | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ `i2s_channel_t` redefined here
      |
      = note: `i2s_channel_t` must be defined only once in the type namespace of this module

error[E0117]: only traits defined in the current crate can be implemented for primitive types
     --> C:\Users\n3xed\dev\esp-idf-test\target\riscv32imc-esp-espidf\debug\build\esp-idf-sys-9c6a2e480ecf6322\out\bindings.rs:68341:17
      |
68341 | #[derive(Debug, Copy, Clone)]
      |                 ^^^^
      |                 |
      |                 impl doesn't use only types from inside the current crate
      |                 `i32` is not defined in the current crate
      |
      = note: define and implement a trait or new type instead
      = note: this error originates in the derive macro `Copy` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0117]: only traits defined in the current crate can be implemented for primitive types
     --> C:\Users\n3xed\dev\esp-idf-test\target\riscv32imc-esp-espidf\debug\build\esp-idf-sys-9c6a2e480ecf6322\out\bindings.rs:68341:23
      |
68341 | #[derive(Debug, Copy, Clone)]
      |                       ^^^^^
      |                       |
      |                       impl doesn't use only types from inside the current crate
      |                       `i32` is not defined in the current crate
      |
      = note: define and implement a trait or new type instead
      = note: this error originates in the derive macro `Clone` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0117]: only traits defined in the current crate can be implemented for primitive types
     --> C:\Users\n3xed\dev\esp-idf-test\target\riscv32imc-esp-espidf\debug\build\esp-idf-sys-9c6a2e480ecf6322\out\bindings.rs:68341:10
      |
68341 | #[derive(Debug, Copy, Clone)]
      |          ^^^^^
      |          |
      |          impl doesn't use only types from inside the current crate
      |          `i32` is not defined in the current crate
      |
      = note: define and implement a trait or new type instead
      = note: this error originates in the derive macro `Debug` (in Nightly builds, run with -Z macro-backtrace for more info)
```

</details>

> The error with cmake looks strange, because cmake is registered there also for macos-arm64 `tools.json`: https://github.com/espressif/esp-idf/blob/master/tools/tools.json#L642

That is indeed strange. Maybe an old master clone? I can't really test this since I don't have an M1 Mac.

> ```shell
> error[E0308]: mismatched types
>   --> /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.9/src/lib.rs:31:62
>    |
> 31 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
>    |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`
> 
> error[E0308]: mismatched types
>   --> /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.9/src/lib.rs:33:51
>    |
> 33 | const ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t;
>    |                                                   ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`
> ```

This error is because of the `time_t` size change from `v4` to `v5` (see [this doc comment](https://github.com/esp-rs/esp-idf-sys/blob/e73e5ee869ff3e994626f1f7e1d7c17679dfb25f/src/lib.rs#L19:L27)), not sure if this came up in a meeting already. Maybe @ivmarkov can say more about this.

georgik, 2022-09-26:
Thank you @N3xed for explanation. It seems that esp-idf-template contains outdated version of embuild.
@SergioGasquez Please, could you update it.

SergioGasquez, 2022-09-29:
Hi! A few updates, `esp-idf-template` dependencies are updated in https://github.com/esp-rs/esp-idf-template/pull/58. Here its what I am doing and encountering:
- Using esp-idf-template with updated dependencies
- Using `release/v5.0` branch
- Added `rustflags = "--cfg espidf_time64"` to the `build` section of `config.toml`
I am having the following error, both when using an activated environment (with esp-idf-sys `master`) and with no activated environment: 
```sh
error[E0308]: mismatched types
  --> /home/esp/.cargo/git/checkouts/esp-idf-sys-d4c2e811b42d99d6/e73e5ee/src/lib.rs:31:62
   |
31 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`
```
Which is a subset of the error we were seeing before.

ivmarkov, 2022-09-29:
That's because the libc crate version containing the necessary change is not released yet. I thought somebody would anyway publish a new version in a matter of days (used to be the case before). But that didn't happen. I guess Ill have to PR a new version release myself.

SergioGasquez, 2022-11-07:
## Activated env
```
  Error: failed to generate bindings in file '/Users/sergio/Documents/Espressif/tests/dev50/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-cf0cb59a10fbadc3/out/bindings.rs'

  Caused by:
      clang diagnosed error: /Users/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.11/src/include/esp-idf/bindings.h:83:10: fatal error: 'esp_vfs_console.h' file not found
```
## No activated env
```
  Error: failed to generate bindings in file '/Users/sergio/Documents/Espressif/tests/dev50/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-cf0cb59a10fbadc3/out/bindings.rs'

  Caused by:
      clang diagnosed error: /Users/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.11/src/include/esp-idf/bindings.h:83:10: fatal error: 'esp_vfs_console.h' file not found
```

ivmarkov, 2022-11-07:
> ## Activated env
> ```
>   Error: failed to generate bindings in file [...]

In the meantime, both the libc crate changes, as well as using a newer libc are in latest Rust nightlies, so what remains is @MabezDev to include an updated libc version in the 1.65 ESP Rust compiler release. 

The above error report is something new however, and has nothing to do with `libc`. @SergioGasquez : are the errors from above with latest ESP IDF `master`, or with some other branch?

SergioGasquez, 2022-11-07:
> The above error report is something new however, and has nothing to do with `libc`. @SergioGasquez : are the errors from above with latest ESP IDF `master`, or with some other branch?

I was using [v5.0-dev](https://github.com/espressif/esp-idf/tree/v5.0-dev)

MabezDev, 2022-11-07:
> In the meantime, both the libc crate changes, as well as using a newer libc are in latest Rust nightlies, so what remains is @MabezDev to include an updated libc version in the 1.65 ESP Rust compiler release.

This is done in 1.65.0.1, is this what you're using @SergioGasquez?

SergioGasquez, 2022-11-07:
> This is done in 1.65.0.1, is this what you're using @SergioGasquez?

I was using `1.65.0.0`, I know it was expected to fail due to `libc` version but wanted to verify that it fails, log it into the issue, and then check that with `1.65.0.1` the issue is solved.

ivmarkov, 2022-11-07:
> > The above error report is something new however, and has nothing to do with `libc`. @SergioGasquez : are the errors from above with latest ESP IDF `master`, or with some other branch?
> 
> I was using [v5.0-dev](https://github.com/espressif/esp-idf/tree/v5.0-dev)

Can you try with `master` or `release/v5.0`? I don't think anybody had touched `v5.0-dev` in the last 12 months?

SergioGasquez, 2022-11-07:
Using master, and an activated env (pulled a few days ago):
```
error[E0308]: mismatched types
  --> /Users/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.11/src/lib.rs:36:62
   |
36 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

For more information about this error, try `rustc --explain E0308`.
error: could not compile `esp-idf-sys` due to previous error
```

SergioGasquez, 2022-11-08:
> Is this 1.65.0.0 or 1.65.0.1? 

`1.65.0.0`

> Also, I assume you've put the espidf_time64 flag?

Yes, I am using esp-idf-template which already adds it when using `mainline`

SergioGasquez, 2022-11-08:
Update: Using `1.65.0.1` works fine with `esp-idf-template` `mainline`!

ivmarkov, 2022-11-08:
We still need to do some tests. As in that time-related functions in Rust standard library - when called - do not crash and do not return weird results.

SergioGasquez, 2022-11-09:
Hi! Just found out that `esp-idf-template` does not build for `esp32c3 -  mainline`:
```
error[E0308]: mismatched types
  --> /Users/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.11/src/lib.rs:36:62
   |
36 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

error[E0308]: mismatched types
  --> /Users/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.11/src/lib.rs:38:51
   |
38 | const ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t;
   |                                                   ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

For more information about this error, try `rustc --explain E0308`.
error: could not compile `esp-idf-sys` due to 2 previous errors 
```

closed 2023-03-09 by georgik

Fristi, 2023-05-30:
@georgik 

I'm using a MacBook M1 as well and also getting the error as mentioned before:

```
  Using esp-idf v5.0.2 at '/Users/mark/.espressif/esp-idf/v5.0.2'
  ERROR: tool cmake does not have versions compatible with platform macos-arm64
  Error: Could not install esp-idf

  Caused by:
      command 'IDF_TOOLS_PATH="/Users/mark/.espressif" "/Users/mark/.espressif/python_env/idf5.0_py3.9_env/bin/python" "/Users/mark/.espressif/esp-idf/v5.0.2/tools/idf_tools.py" "--idf-path" "/Users/mark/.espressif/esp-idf/v5.0.2" "--tools-json" "/var/folders/rr/s_760dn15mg3_k0h9j3blgww0000gn/T/.tmpevaCl3" "install" "cmake"' exited with non-zero status code 1
```

I've used the https://github.com/esp-rs/esp-idf-template with `cargo generate`

Using `pyenv` set to Python version 3.9.4

Quite puzzled, been yak shaving for a few hours now. IDF is a beast, if you need more info let me know