esp-rs/rust-build #125: Centralize storage of release number from CI scripts

georgik, 2022-07-22:
Problem: Rust toolchain version number is stored in 23 files. Each release requires an update of all these files.

Find a way to reduce repeated magic numbers in CI scripts.

SergioGasquez, 2022-07-22:
On top of my head, the only option that comes to mind is using [ Actions Secrets](https://docs.github.com/en/actions/security-guides/encrypted-secrets), but it won't show the value if any user would like to know the toolchain version, so I don't really see it as a good option

SergioGasquez, 2022-07-22:
Possible workaround: https://arinco.com.au/blog/github-actions-share-environment-variables-across-workflows/. 
It may also be worth mentioning that for every release we also need to update: 
- [esp-idf-template](https://github.com/esp-rs/esp-idf-template)
  - [Dockerfile](https://github.com/esp-rs/esp-idf-template/blob/master/cargo/.devcontainer/Dockerfile)
  - [.gitpod.Dockerfile](https://github.com/esp-rs/esp-idf-template/blob/master/cargo/.gitpod.Dockerfile#L9)
- [esp-template](https://github.com/esp-rs/esp-template)
  - [Dockerfile](https://github.com/esp-rs/esp-template/blob/main/.devcontainer/Dockerfile)
  - [.gitpod.Dockerfile](https://github.com/esp-rs/esp-template/blob/main/.gitpod.Dockerfile)
- [wokwi-builders](https://github.com/wokwi/wokwi-builders)
  - [esp32-rust/Dockerfile](https://github.com/wokwi/wokwi-builders/blob/main/esp32-rust/Dockerfile#L1)
  - [esp32c3-rust/Dockerfile](https://github.com/wokwi/wokwi-builders/blob/main/esp32c3-rust/Dockerfile#L1)
  - [esp32s2-rust/Dockerfile](https://github.com/wokwi/wokwi-builders/blob/main/esp32s2-rust/Dockerfile#L1)
 
One approach could be to generate `latest` tags. Eg: `esp32_v4.4_latest`,`esp32c3_latest` ... The big problem with this approach is that it can unstabilize those repositories.

SergioGasquez, 2022-07-26:
@jessebraham suggested using: 
```sh
curl --silent https://api.github.com/repos/esp-rs/rust-build/releases | jq -r 'map(select(.prerelease | not)) | first | .tag_name'
```
Which gets you the latest release name which is NOT a pre-release, the `jq` query can be updated as needed.