esp-rs/rust-build #130: Installer breaks when `-n` is empty

har7an, 2022-08-02:
When using the installer to install the toolchain, setting the [`--nightly-version`][1] to an empty value (`-n ""`) causes the `NIGHTLY_VERSION` env var not to be set, in turn causing the installation to fail [at this point][2], because the argument to `install_rust_toolchain` is empty although [it is required][3].

Is this intended behavior?

[1]: https://github.com/esp-rs/rust-build/blob/2f8b079aefb0906abccc565275b3bcad95aa1fba/install-rust-toolchain.sh#L152
[2]: https://github.com/esp-rs/rust-build/blob/2f8b079aefb0906abccc565275b3bcad95aa1fba/install-rust-toolchain.sh#L457
[3]: https://github.com/esp-rs/rust-build/blob/2f8b079aefb0906abccc565275b3bcad95aa1fba/install-rust-toolchain.sh#L163

SergioGasquez, 2022-08-04:
Hello @har7an! I would say this is something that we want to keep and it's intended behavior, the [`--nightly-version`](https://github.com/esp-rs/rust-build/blob/2f8b079aefb0906abccc565275b3bcad95aa1fba/install-rust-toolchain.sh#L152) argument lets you choose a specific version of nightly toolchain, [by default, this value is set to latest nigthly](https://github.com/esp-rs/rust-build/blob/2f8b079aefb0906abccc565275b3bcad95aa1fba/install-rust-toolchain.sh#L21), so the parameter should only be used when wanting to specify a different version, if that is not the case, I would recommend not setting this argument.

har7an, 2022-08-04:
@SergioGasquez I see, thanks. What is the purpose of it, if I may ask? I build OCI containers for private use with the build script that target the esp32, and they run fine after removing the nightly and latest stable toolchains (Not the `esp` one).

And would you mind if I turn this "recommendation" into a hard requirement by enforcing that `NIGHTLY_VERSION` mustn't be empty?

(I build the containers myself because the one offered here is just much too heavy for what I do)

SergioGasquez, 2022-08-04:
>  What is the purpose of it, if I may ask?

There are some cases where the user would like a specific version of `nightly` toolchain instead of `latest`, ie: `nightly-2022-03-10` 

Feel free to use `-n "nightly"`, which will result in the same behaviour as not having the `-n` argument.

> (I build the containers myself because the one offered here is just much too heavy for what I do)

Do you mind elaborating on what do you remove from the container beside the `nightly` toolchain as you already mentioned?

har7an, 2022-08-04:
> Feel free to use `-n "nightly"`, which will result in the same behaviour as not having the `-n` argument.

Okay, that's what I did. Thanks!

> Do you mind elaborating on what do you remove from the container beside the `nightly` toolchain as you already mentioned?

I'm building the container myself entirely. I started doing this before there were individual containers per esp variant without the whole IDF inside. Here is [the Containerfile][1], and here is [the Gitlab registry][2] I'm pushing the containers to.

The containers are pretty barebones: Based off Fedora 36, with openssl, gcc and xz as additional dependencies. Also I'm installing `rust-analyzer` inside so I can easily use it as LSP server with the right toolchain included. I'm sure I could save more space, but the latest container is just shy of 2 GB in size (uncompressed), and I can live with that.

[1]: https://gitlab.com/c8160/embedded-rust/rust-xtensa32/-/blob/main/Containerfile
[2]: https://gitlab.com/c8160/embedded-rust/rust-xtensa32/container_registry/3082493

SergioGasquez, 2022-08-05:
Thanks for sharing the Containerfile! If you don't mind, I'll close this issue!

closed 2022-08-05 by SergioGasquez