esp-rs/rust-build #132: Error during building.

jBernavaPrah, 2022-08-04:
Hi!

I'm pretty new to the rust and I'm trying to build the example that is on the readme, without success.

After a day of tests and failures, I don't know where to go to have this working. 

I'm on Mac Montery (12.4).

The error that I'm getting is: 
```bash
jbernavaprah@MBP-di-Jure rust-esp32-std-demo % cargo espflash --target xtensa-esp32-espidf /dev/tty.usbserial-21220 
Serial port: /dev/tty.usbserial-21220
Connecting...

   Compiling esp-idf-sys v0.31.7
error: failed to run custom build command for `esp-idf-sys v0.31.7`

Caused by:
  process didn't exit successfully: `/Users/jbernavaprah/rust-esp32-std-demo/target/debug/build/esp-idf-sys-3aea8f5e6a6ea28f/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  IDF_PYTHON_ENV_PATH=/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  Current system platform: macos-arm64
  Skipping xtensa-esp32-elf@esp-2021r2-patch3-8.4.0 (already installed)
  Skipping cmake@3.23.1 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
  IDF_PYTHON_ENV_PATH=/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32s2
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  Python requirements from /Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/requirements.txt are satisfied.
  -- Project sdkconfig file /Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out/sdkconfig
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out/gen-sdkconfig.defaults...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32s2...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  -- App "libespidf" version: 1
  -- Adding linker script /Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out/build/esp-idf/esp_system/ld/memory.ld
  -- Adding linker script /Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/ld/esp32/sections.ld.in
  -- Adding linker script /Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.ld
  -- Adding linker script /Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.api.ld
  -- Adding linker script /Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
  -- Adding linker script /Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
  -- Adding linker script /Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
  -- Adding linker script /Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
[... 60 lines cut ...]
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: EspIdfRemote { repo_url: None, git_ref: Branch("release/v4.4") }
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.2 at '/Users/jbernavaprah/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4'
  fatal: No names found, cannot describe anything.
  Built components: esp_ringbuf, efuse, esp_ipc, driver, esp_pm, mbedtls, bootloader, esptool_py, partition_table, app_update, bootloader_support, spi_flash, nvs_flash, pthread, esp_gdbstub, espcoredump, esp_phy, esp_system, esp_rom, hal, vfs, esp_eth, tcpip_adapter, esp_netif, esp_event, wpa_supplicant, esp_wifi, ieee802154, console, openthread, lwip, log, heap, soc, esp_hw_support, xtensa, esp32, esp_common, esp_timer, freertos, newlib, cxx, app_trace, asio, bt, cbor, unity, cmock, coap, nghttp, esp-tls, esp_adc_cal, esp_hid, tcp_transport, esp_http_client, esp_http_server, esp_https_ota, esp_https_server, esp_lcd, protobuf-c, protocomm, mdns, esp_local_ctrl, sdmmc, esp_serial_slave_link, esp_websocket_client, expat, wear_levelling, fatfs, freemodbus, idf_test, jsmn, json, libsodium, mqtt, openssl, perfmon, spiffs, usb, tinyusb, ulp, wifi_provisioning
  error: unknown target triple 'xtensa', please use -triple or -arch
  thread 'main' panicked at 'libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!', /Users/jbernavaprah/.cargo/registry/src/github.com-1ecc6299db9ec823/bindgen-0.60.1/src/ir/context.rs:529:15
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

SergioGasquez, 2022-08-04:
Hello @jBernavaPrah, can you give us more details: What arguments are you using for installation? Are you sourcing the resulting file? Do you have esp-idf installed and activated?

jBernavaPrah, 2022-08-04:
Thanks for the fast response!

So:
1) What arguments are you using for installation?
- If you mean the `install-rust-toolchain.sh` I ran it without any parameter. 
2) Are you sourcing the resulting file?
- I think no, what do you mean by this? 
3) Do you have esp-idf installed and activated?
I think no, I used successfully VS-Code with platform.io to run c++ code on the device.

More context: 
I have an Espressif ESP32-WROOM-32E attached to the using USB on the /dev/tty.usbserial-21220.

jBernavaPrah, 2022-08-04:
Update: after installing esp-idf the error change and now I get:

```bash
jbernavaprah@MBP-di-Jure rust-esp32-std-demo % cargo espflash --target xtensa-esp32-espidf /dev/tty.usbserial-21230                  
Serial port: /dev/tty.usbserial-21230
Connecting...

   Compiling esp-idf-sys v0.31.7
error: failed to run custom build command for `esp-idf-sys v0.31.7`

Caused by:
  process didn't exit successfully: `/Users/jbernavaprah/rust-esp32-std-demo/target/debug/build/esp-idf-sys-3aea8f5e6a6ea28f/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32s2
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/Users/jbernavaprah/esp/esp-idf/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/Users/jbernavaprah/.espressif/python_env/idf4.4_py3.9_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS=  -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  Python requirements from /Users/jbernavaprah/esp/esp-idf/requirements.txt are satisfied.
  -- Project sdkconfig file /Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out/sdkconfig
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out/gen-sdkconfig.defaults...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32s2...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  -- App "libespidf" version: 1
  -- Adding linker script /Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out/build/esp-idf/esp_system/ld/memory.ld
  -- Adding linker script /Users/jbernavaprah/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
  -- Adding linker script /Users/jbernavaprah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
  -- Adding linker script /Users/jbernavaprah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
  -- Adding linker script /Users/jbernavaprah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
  -- Adding linker script /Users/jbernavaprah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
  -- Adding linker script /Users/jbernavaprah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
  -- Adding linker script /Users/jbernavaprah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
  -- Adding linker script /Users/jbernavaprah/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
  -- Adding linker script /Users/jbernavaprah/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
  -- Configuring done
  -- Generating done
  -- Build files have been written to: /Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out/build
  running: "cmake" "--build" "." "--config" "MinSizeRel" "--parallel" "10"
  [1/542] Building C object esp-idf/esp_http_server/CMakeFiles/__idf_esp_http_server.dir/src/httpd_ws.c.obj
  [2/542] Building C object esp-idf/esp_http_client/CMakeFiles/__idf_esp_http_client.dir/lib/http_utils.c.obj
  [3/542] Building C object esp-idf/esp_http_client/CMakeFiles/__idf_esp_http_client.dir/lib/http_header.c.obj
  [4/542] Building C object esp-idf/esp_http_client/CMakeFiles/__idf_esp_http_client.dir/lib/http_auth.c.obj
  [5/542] Building C object esp-idf/esp_http_server/CMakeFiles/__idf_esp_http_server.dir/src/httpd_sess.c.obj
  [6/542] Building C object esp-idf/esp_http_server/CMakeFiles/__idf_esp_http_server.dir/src/httpd_uri.c.obj
  [7/542] Building C object esp-idf/esp_http_server/CMakeFiles/__idf_esp_http_server.dir/src/httpd_txrx.c.obj
[... cut ...]
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using activated esp-idf v4.4.1 environment at '/Users/jbernavaprah/esp/esp-idf'
  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /Users/jbernavaprah/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.48/src/lib.rs:975:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

SergioGasquez, 2022-08-04:
Please, use the following: 

1.` ./install-rust-toolchain.sh --export-file export-esp-rust.sh`
2. Source `export-esp-rust.sh`
3. Try to build the example

jBernavaPrah, 2022-08-04:
Thanks for the matrix room, I will write there next time. 

Using the follow command, I receive now this error.

```bash
jbernavaprah@MBP-di-Jure ~ % ./install-rust-toolchain.sh --export-file export-esp-rust.sh
Processing configuration:
--build-target          = esp32,esp32s2,esp32s3
--cargo-home            = /Users/jbernavaprah/.cargo
--clear-cache           = YES
--esp-idf-version       = 
--export-file           = export-esp-rust.sh
--extra-crates          = ldproxy cargo-espflash
--installation-mode     = install
--llvm-version          = esp-14.0.0-20220415
--minified-esp-idf      = NO
--minified-llvm         = YES
--nightly-version       = nightly
--rustup-home           = /Users/jbernavaprah/.rustup
--system-packages       = openssl@3
--toolchain-version     = 1.62.1.0
--toolchain-destination = /Users/jbernavaprah/.rustup/toolchains/esp
/Users/jbernavaprah/.cargo/bin/rustup
nightly-aarch64-apple-darwin
error: 'rustc' is not installed for the toolchain 'esp'
note: this is a custom toolchain, which cannot use `rustup component add`
help: if you built this toolchain from source, and used `rustup toolchain link`, then you may be able to build the component with `x.py`
rustc version is too low, requires 1.55
calling rustup
info: downloading installer
info: profile set to 'minimal'
info: default host triple is aarch64-apple-darwin
info: skipping toolchain installation

Rust is installed now. Great!

To get started you may need to restart your current shell.
This would reload your PATH environment variable to include
Cargo's bin directory ($HOME/.cargo/bin).

To configure your current shell, run:
source "$HOME/.cargo/env"
Processing toolchain for aarch64-apple-darwin - operation: install
Installing system packages: openssl@3
/opt/homebrew/bin/brew
/opt/homebrew/Cellar/openssl@3/3.0.5/.bottle/etc/ (7 files)
/opt/homebrew/Cellar/openssl@3/3.0.5/bin/c_rehash
/opt/homebrew/Cellar/openssl@3/3.0.5/bin/openssl
/opt/homebrew/Cellar/openssl@3/3.0.5/include/openssl/ (135 files)
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/libcrypto.3.dylib
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/libssl.3.dylib
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/engines-3/ (3 files)
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/ossl-modules/legacy.dylib
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/pkgconfig/ (3 files)
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/ (4 other files)
/opt/homebrew/Cellar/openssl@3/3.0.5/share/doc/ (787 files)
/opt/homebrew/Cellar/openssl@3/3.0.5/share/man/ (5493 files)
Previous installation of toolchain exist in: /Users/jbernavaprah/.rustup/toolchains/esp
Please, remove the directory before new installation.
jbernavaprah@MBP-di-Jure ~ % 

```

And after removing the directory /Users/jbernavaprah/.rustup/toolchains/esp, I have this other error:

```bash
jbernavaprah@MBP-di-Jure ~ % ./install-rust-toolchain.sh --export-file export-esp-rust.sh
Processing configuration:
--build-target          = esp32,esp32s2,esp32s3
--cargo-home            = /Users/jbernavaprah/.cargo
--clear-cache           = YES
--esp-idf-version       = 
--export-file           = export-esp-rust.sh
--extra-crates          = ldproxy cargo-espflash
--installation-mode     = install
--llvm-version          = esp-14.0.0-20220415
--minified-esp-idf      = NO
--minified-llvm         = YES
--nightly-version       = nightly
--rustup-home           = /Users/jbernavaprah/.rustup
--system-packages       = openssl@3
--toolchain-version     = 1.62.1.0
--toolchain-destination = /Users/jbernavaprah/.rustup/toolchains/esp
/Users/jbernavaprah/.cargo/bin/rustup
nightly-aarch64-apple-darwin
error: toolchain 'esp' is not installed
rustc version is too low, requires 1.55
calling rustup
info: downloading installer
info: profile set to 'minimal'
info: default host triple is aarch64-apple-darwin
info: skipping toolchain installation

Rust is installed now. Great!

To get started you may need to restart your current shell.
This would reload your PATH environment variable to include
Cargo's bin directory ($HOME/.cargo/bin).

To configure your current shell, run:
source "$HOME/.cargo/env"
Processing toolchain for aarch64-apple-darwin - operation: install
Installing system packages: openssl@3
/opt/homebrew/bin/brew
/opt/homebrew/Cellar/openssl@3/3.0.5/.bottle/etc/ (7 files)
/opt/homebrew/Cellar/openssl@3/3.0.5/bin/c_rehash
/opt/homebrew/Cellar/openssl@3/3.0.5/bin/openssl
/opt/homebrew/Cellar/openssl@3/3.0.5/include/openssl/ (135 files)
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/libcrypto.3.dylib
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/libssl.3.dylib
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/engines-3/ (3 files)
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/ossl-modules/legacy.dylib
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/pkgconfig/ (3 files)
/opt/homebrew/Cellar/openssl@3/3.0.5/lib/ (4 other files)
/opt/homebrew/Cellar/openssl@3/3.0.5/share/doc/ (787 files)
/opt/homebrew/Cellar/openssl@3/3.0.5/share/man/ (5493 files)
./install-rust-toolchain.sh: line 234: ./rust-1.62.1.0-aarch64-apple-darwin/install.sh: No such file or directory
jbernavaprah@MBP-di-Jure ~ % 

```

Those errors come after installing the esp-idf as you mentioned before.

I tried also removing from the $PATH everything that is related to esp-idf, but I still have the same errors.

jBernavaPrah, 2022-08-04:
So, where I am now:

Installed `esp-idf` v4.4. Following the official guide. 
installed the `./install-rust-toolchain.sh --export-file export-esp-rust.sh`

Both were executed without errors.

When trying `(. ~/export-esp-rust.sh ; cargo build)` It's not building the error is: 
```bash
jbernavaprah@MBP-di-Jure rust-esp32-std-demo % (. ~/export-esp-rust.sh ; cargo build)
   Compiling addr2line v0.16.0
   Compiling esp-idf-sys v0.31.7
   Compiling std v0.0.0 (/Users/jbernavaprah/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
   Compiling proc_macro v0.0.0 (/Users/jbernavaprah/.rustup/toolchains/esp/lib/rustlib/src/rust/library/proc_macro)
error: failed to run custom build command for `esp-idf-sys v0.31.7`

Caused by:
  process didn't exit successfully: `/Users/jbernavaprah/rust-esp32-std-demo/target/debug/build/esp-idf-sys-3aea8f5e6a6ea28f/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32s2
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/Users/jbernavaprah/esp/esp-idf/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/Users/jbernavaprah/.espressif/python_env/idf4.4_py3.9_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  Python requirements from /Users/jbernavaprah/esp/esp-idf/requirements.txt are satisfied.
  -- Project sdkconfig file /Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out/sdkconfig
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out/gen-sdkconfig.defaults...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32s2...
  Loading defaults file /Users/jbernavaprah/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  -- Configuring incomplete, errors occurred!
  See also "/Users/jbernavaprah/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-a821ad9e24a820f7/out/build/CMakeFiles/CMakeOutput.log".

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: Some(
          [
              "sdkconfig.defaults",
              "sdkconfig.defaults.esp32",
              "sdkconfig.defaults.esp32s2",
          ],
      ),
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.0",
[... 30 lines cut ...]
    crosstool-ng version esp-2022r1-RC1 doesn't match supported version
    esp-2021r2-patch3

    Please try to run 'idf.py fullclean' to solve it quickly.

    Check Getting Started documentation if the error continues.

    You can override this error and proceed with build by defining the
    IDF_MAINTAINER environment variable.

  Call Stack (most recent call first):
    /Users/jbernavaprah/esp/esp-idf/components/esp_common/project_include.cmake:7 (crosstool_version_check)
    /Users/jbernavaprah/esp/esp-idf/tools/cmake/build.cmake:325 (include)
    /Users/jbernavaprah/esp/esp-idf/tools/cmake/build.cmake:534 (__build_process_project_includes)
    CMakeLists.txt:14 (idf_build_process)

  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /Users/jbernavaprah/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.48/src/lib.rs:975:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
```

Tried to execute `idf.py fullclean` but nothing changed with the previous command.

SergioGasquez, 2022-08-04:
The esp-idf version you have installed and activated is `v4.4.2` and the [example is expecting release/v4.4](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/.cargo/config.toml#L33). Here are some solutions:
- Go to your esp-idf folder, uninstall it, checkout to `release/v4.4`, install it and source the esp-idf export.sh
- Uninstall esp-idf and use the rust install script to install it for you, so you only have to source one file:
```sh
./install-rust-toolchain.sh \
    --build-target "esp32" \
    --esp-idf-version "release/v4.4" \
    --minified-esp-idf "YES" \
    --export-file export-esp-rust.sh
```
- [Use containers or devcontainers](https://github.com/esp-rs/rust-build#containers-with-rust-esp-environment)

jBernavaPrah, 2022-08-04:
Happy to announce that finally, it get working. 

The problem is... I don't know how. I will do some further testing and let know. But I think the main issue here was the total missing of the esp-idf at the first and after, that I installed the v5, instead of v4.4 of esp-idf.

My 50 cents:

- In the documentation, for a dummy like me and how come from platform.io, is missing the mandataries of install esp-idf. 
More mandatory is to install the esp-idf v4.4, as the v5.0 is not yet compatible with some modules/cargos(?). 
- Also in the documentation I think is better to simplify the installation steps. At the and (for macOS at least) is literally an execution of one file. But used a lot of time to test and understand if I need to use the installation script from the curl example or from the github, in the section MacOS, and what were the differences between those files. In the end, they were the same. 

If you think, I can create a PR for the documentation.

But now, super happy to start this adventure in rust and esp32! :D

SergioGasquez, 2022-08-05:
> The problem is... I don't know how.

Can you check if there is a populated `.embuild` directory on your project folder?

Regarding the PR, we are more than happy to accept any PR! Happy to see that your journey on ESP32s with Rust has just started!

jBernavaPrah, 2022-08-05:
The current state of the .embuild is: 

[image]

SergioGasquez, 2022-08-05:
It looks like you didnt have the esp-idf environment activated. When you run the build command with no esp-idf activated it will download it under the `.embuild` directory and use it. If you use this approach with several projects be aware that I can eat a lot of storage.

I will close the issue, feel free to post any other questions on the matrix channel :)