esp-rs/rust-build #136: Build error related to a certificate error

pixelspark, 2022-08-21:
_Not sure if this is the right place to post this (seems related to the build scripts from this repository and not necessarily specific to the hello world template)_

Trying to install and build the hello world example for Rust on ESP32s3 on macOS (M1) gives me the following error:

````
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
````

I have python 3.10 installed through `brew` (`brew install python@3.10 --force-bottle`). I also ensured the latest certificates are installed (`pip3 install --upgrade certifi`). Full log:

````
% cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template cargo
# chosen settings: esp32s3, with std, esp-idf 4.4, no dev containers, in folder testesp
% cd testesp
% cargo build
   Compiling proc-macro2 v1.0.43
   Compiling compiler_builtins v0.1.79
   Compiling quote v1.0.21
   Compiling unicode-ident v1.0.3
   Compiling syn v1.0.99
   Compiling core v0.0.0 ~/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.132
   Compiling cfg-if v1.0.0
   Compiling memchr v2.5.0
   Compiling serde_derive v1.0.144
   Compiling serde v1.0.144
   Compiling log v0.4.17
   Compiling regex-syntax v0.6.27
   Compiling rustversion v1.0.9
   Compiling autocfg v1.1.0
   Compiling glob v0.3.0
   Compiling bitflags v1.3.2
   Compiling serde_json v1.0.83
   Compiling once_cell v1.13.1
   Compiling heck v0.4.0
   Compiling itoa v1.0.3
   Compiling anyhow v1.0.62
   Compiling ryu v1.0.11
   Compiling libc v0.2.131
   Compiling crossbeam-utils v0.8.11
   Compiling termcolor v1.1.3
   Compiling same-file v1.0.6
   Compiling hashbrown v0.12.3
   Compiling minimal-lexical v0.2.1
   Compiling fnv v1.0.7
   Compiling lazy_static v1.4.0
   Compiling os_str_bytes v6.3.0
   Compiling shlex v1.1.0
   Compiling strsim v0.10.0
   Compiling camino v1.1.1
   Compiling humantime v2.1.0
   Compiling either v1.8.0
   Compiling cc v1.0.73
   Compiling semver v1.0.13
   Compiling textwrap v0.15.0
   Compiling bindgen v0.60.1
   Compiling fastrand v1.8.0
   Compiling remove_dir_all v0.5.3
   Compiling rustc-hash v1.1.0
   Compiling lazycell v1.3.0
   Compiling peeking_take_while v0.1.2
   Compiling std v0.0.0 (~/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/std)
   Compiling zero v0.1.2
   Compiling libloading v0.7.3
   Compiling thread_local v1.1.4
   Compiling indexmap v1.9.1
   Compiling walkdir v2.3.2
   Compiling clang-sys v1.3.3
   Compiling clap_lex v0.2.4
   Compiling xmas-elf v0.8.0
[... 343 lines cut ...]
  WARNING: Git describe was unsuccessful: b''
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.2 at '~/Downloads/testesp/.embuild/espressif/esp-idf/release-v4.4'
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
  WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch3/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to ~/Downloads/testesp/.embuild/espressif/dist/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz.tmp
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
  WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch3/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to ~/Downloads/testesp/.embuild/espressif/dist/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz.tmp
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
  WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch3/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to ~/Downloads/testesp/.embuild/espressif/dist/riscv32-esp-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz.tmp
  ERROR: Failed to download, and retry count has expired
  Traceback (most recent call last):
    File "~/Downloads/testesp/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 1860, in <module>
      main(sys.argv[1:])
    File "~/Downloads/testesp/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 1856, in main
      action_func(args)
    File "~/Downloads/testesp/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 1470, in action_install
      tool_obj.download(tool_version)
    File "~/Downloads/testesp/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 722, in download
      raise DownloadError()
  __main__.DownloadError
  Error: Could not install esp-idf

  Caused by:
      command '"~/Downloads/testesp/.embuild/espressif/python_env/idf4.4_py3.10_env/bin/python" "~/Downloads/testesp/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py" "--idf-path" "~/Downloads/testesp/.embuild/espressif/esp-idf/release-v4.4" "install" "riscv32-esp-elf" "cmake" "ninja"' exited with non-zero status code 1
````

georgik, 2022-08-22:
Thank you @pixelspark for reporting the problem.

The build tools deploys clone of ESP-IDF to .embuild/espressif directory.

Please run: `open /Applications/Python\ X.Y/Install\ Certificates.command` which should update certificates in the system. Replace X.Y by version number available in your system.

pixelspark, 2022-08-22:
Yeah found that solution too, unfortunately that tool is not available for brew-installed Python..?

MabezDev, 2022-08-23:
Perhaps the first option on this comment might help: https://stackoverflow.com/a/62374703

robotoss, 2022-08-24:
@MabezDev 
MacOS Monterey 12.5 (M1 pro)
Your answer did not help
The error is not resolved.

georgik, 2022-08-24:
@robotoss Please, could you try one of following solutions: https://github.com/espressif/esp-idf/issues/4081#issuecomment-883400463

robotoss, 2022-08-24:
@georgik 
`I used python3 -m pip install certifi`
and that didn't help

I don't have DST Root CA X3 in Keychain
ISGR Root X1 is trusted and up-to-date

robotoss, 2022-08-26:
Is there any news? 

I try to reinstall python
And run Certificates.command

```
zoxo@Konstantins-MBP ~ % /Applications/Python\ 3.10/Install\ Certificates.command ; exit;
 -- pip install --upgrade certifi
Requirement already satisfied: certifi in /Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages (2022.6.15)
 -- removing any existing file or link
 -- creating symlink to certifi certificate bundle
 -- setting permissions
 -- update complete

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]
```

```
zoxo@Konstantins-MBP some-test-name % cargo build
   Compiling proc-macro2 v1.0.43
   Compiling compiler_builtins v0.1.73
   Compiling quote v1.0.21
   Compiling unicode-ident v1.0.3
   Compiling syn v1.0.99
   Compiling core v0.0.0 (/Users/zoxo/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.132
   Compiling cfg-if v1.0.0
   Compiling memchr v2.5.0
   Compiling serde_derive v1.0.144
   Compiling serde v1.0.144
   Compiling log v0.4.17
   Compiling rustversion v1.0.9
   Compiling regex-syntax v0.6.27
   Compiling serde_json v1.0.85
   Compiling glob v0.3.0
   Compiling autocfg v1.1.0
   Compiling bitflags v1.3.2
   Compiling anyhow v1.0.62
   Compiling heck v0.4.0
   Compiling once_cell v1.13.1
   Compiling libc v0.2.126
   Compiling ryu v1.0.11
   Compiling itoa v1.0.3
   Compiling crossbeam-utils v0.8.11
   Compiling minimal-lexical v0.2.1
   Compiling os_str_bytes v6.3.0
   Compiling fnv v1.0.7
   Compiling same-file v1.0.6
   Compiling termcolor v1.1.3
   Compiling lazy_static v1.4.0
   Compiling hashbrown v0.12.3
   Compiling shlex v1.1.0
   Compiling cc v1.0.69
   Compiling humantime v2.1.0
   Compiling camino v1.1.1
   Compiling either v1.8.0
   Compiling textwrap v0.15.0
   Compiling bindgen v0.60.1
   Compiling semver v1.0.13
   Compiling strsim v0.10.0
   Compiling cc v1.0.73
   Compiling memchr v2.4.1
   Compiling lazycell v1.3.0
   Compiling fastrand v1.8.0
   Compiling peeking_take_while v0.1.2
[... 408 lines cut ...]
  WARNING: Git describe was unsuccessful: b''
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.2 at '/Users/zoxo/Documents/some-test-name/.embuild/espressif/esp-idf/release-v4.4'
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
  WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch3/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to /Users/zoxo/Documents/some-test-name/.embuild/espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz.tmp
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
  WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch3/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to /Users/zoxo/Documents/some-test-name/.embuild/espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz.tmp
  WARNING: Download failure <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
  WARNING: Failed to download https://github.com/espressif/crosstool-NG/releases/download/esp-2021r2-patch3/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz to /Users/zoxo/Documents/some-test-name/.embuild/espressif/dist/xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch3-macos.tar.gz.tmp
  ERROR: Failed to download, and retry count has expired
  Traceback (most recent call last):
    File "/Users/zoxo/Documents/some-test-name/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 1860, in <module>
      main(sys.argv[1:])
    File "/Users/zoxo/Documents/some-test-name/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 1856, in main
      action_func(args)
    File "/Users/zoxo/Documents/some-test-name/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 1470, in action_install
      tool_obj.download(tool_version)
    File "/Users/zoxo/Documents/some-test-name/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py", line 722, in download
      raise DownloadError()
  __main__.DownloadError
  Error: Could not install esp-idf

  Caused by:
      command '"/Users/zoxo/Documents/some-test-name/.embuild/espressif/python_env/idf4.4_py3.10_env/bin/python" "/Users/zoxo/Documents/some-test-name/.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py" "--idf-path" "/Users/zoxo/Documents/some-test-name/.embuild/espressif/esp-idf/release-v4.4" "install" "xtensa-esp32-elf" "cmake" "ninja" "esp32ulp-elf"' exited with non-zero status code 1
```

pixelspark, 2022-08-27:
A temporary fix that works for me is to download the list of trusted roots from Mozilla [here](https://wiki.mozilla.org/CA/Included_Certificates) (pick [PEM of Root Certificates in Mozilla's Root Store with the Websites (TLS/SSL) Trust Bit Enabled](https://ccadb-public.secure.force.com/mozilla/IncludedRootsPEMTxt?TrustBitsInclude=Websites).

Then:

````
SSL_CERT_FILE=/path/to/IncludedRootsPEM.txt cargo build
````

To test just the offending command:
````
python3 ./.embuild/espressif/esp-idf/release-v4.4/tools/idf_tools.py --idf-path ./.embuild/espressif/esp-idf/release-v4.4 install riscv32-esp-elf cmake ninja
````

robotoss, 2022-08-30:
Greetings, everyone.
I don't know what helped me, but here's my procedure:
1. I completely uninstalled the previously installed python.
2. Cleared the .zshrc of 
```
export LIBCLANG_PATH="/Users/zoxo/.espressif/tools/xtensa-esp32-elf-clang/esp-14.0.0-20220415-aarch64-apple-darwin/lib/"
export PATH="/Users/zoxo/.espressif/tools/xtensa-esp32-elf-gcc/8_4_0-esp-2021r2-patch3-aarch64-apple-darwin/bin/:/Users$
export PATH="$PATH":"$HOME/.pub-cache/bin"
```
3. Delete esp toolchain
4. Instal Python 3.10.6 (https://www.python.org/downloads/)
5. Run Install Certificates.command in Applications/Python3.10
6. Install rust-build by unstruction  https://github.com/esp-rs/rust-build

After that, everything worked.

georgik, 2023-03-09:
There were many updates and fixes, meanwhile. Closing the issue.

closed 2023-03-09 by georgik