esp-rs/rust-build #150: Support for ESP-IDF v5.1 CI

georgik, 2022-09-23:
Build with activated ESP-IDF environment with 5.1 is failing:
```

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/Users/rust/projects/esp-idf-5",
          ),
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using activated esp-idf v5.1.0 environment at '/Users/rust/projects/esp-idf-5'
  ERROR: /Users/rust/projects/esp-idf-rs-test/.embuild/espressif/python_env/idf5.1_py3.9_env/bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it
  CMake Error at /Users/rust/projects/esp-idf-5/tools/cmake/build.cmake:342 (message):
    Some Python dependencies must be installed.  Check above message for
    details.
  Call Stack (most recent call first):
    /Users/rust/projects/esp-idf-5/tools/cmake/build.cmake:480 (__build_check_python)
    CMakeLists.txt:14 (idf_build_process)

  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /Users/rust/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.48/src/lib.rs:975:5
```

SergioGasquez, 2022-09-23:
Here are my test results on Linux:
Generate command: 
```sh
cargo generate --git https://github.com/esp-rs/esp-idf-template cargo --name test50 --vcs none --silent -d mcu=esp32 -d std=true -d espidfver=mainline -d devcontainer=false
```
- Using activated environment with esp-idf `master` branch:
```sh
  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/home/sergio/.espressif/frameworks/master",
          ),
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using activated esp-idf v5.1.0 environment at '/home/sergio/.espressif/frameworks/master'
  ERROR: /home/sergio/Documents/Espressif/tests/test51/.embuild/espressif/python_env/idf5.1_py3.10_env/bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it
  CMake Error at /home/sergio/.espressif/frameworks/master/tools/cmake/build.cmake:360 (message):
    Some Python dependencies must be installed.  Check above message for
    details.
  Call Stack (most recent call first):
    /home/sergio/.espressif/frameworks/master/tools/cmake/build.cmake:498 (__build_check_python)
    CMakeLists.txt:14 (idf_build_process)

  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1

  build script failed, must exit now', /home/sergio/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.48/src/lib.rs:975:5
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
Same result as @georgik had on MacM1.

SergioGasquez, 2022-11-07:
## Activated env
```
error[E0428]: the name `i2s_channel_t` is defined multiple times
     --> /Users/sergio/Documents/Espressif/tests/dev51/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-cf0cb59a10fbadc3/out/bindings.rs:75590:1
      |
75546 | pub struct i2s_channel_t {
      | ------------------------ previous definition of the type `i2s_channel_t` here
...
75590 | pub type i2s_channel_t = c_types::c_uint;
      | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ `i2s_channel_t` redefined here
      |
      = note: `i2s_channel_t` must be defined only once in the type namespace of this module

error[E0117]: only traits defined in the current crate can be implemented for primitive types
     --> /Users/sergio/Documents/Espressif/tests/dev51/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-cf0cb59a10fbadc3/out/bindings.rs:75545:17
      |
75545 | #[derive(Debug, Copy, Clone)]
      |                 ^^^^
      |                 |
      |                 impl doesn't use only types from inside the current crate
      |                 `u32` is not defined in the current crate
      |
      = note: define and implement a trait or new type instead
      = note: this error originates in the derive macro `Copy` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0117]: only traits defined in the current crate can be implemented for primitive types
     --> /Users/sergio/Documents/Espressif/tests/dev51/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-cf0cb59a10fbadc3/out/bindings.rs:75545:23
      |
75545 | #[derive(Debug, Copy, Clone)]
      |                       ^^^^^
      |                       |
      |                       impl doesn't use only types from inside the current crate
      |                       `u32` is not defined in the current crate
      |
      = note: define and implement a trait or new type instead
      = note: this error originates in the derive macro `Clone` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0117]: only traits defined in the current crate can be implemented for primitive types
     --> /Users/sergio/Documents/Espressif/tests/dev51/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-cf0cb59a10fbadc3/out/bindings.rs:75545:10
      |
75545 | #[derive(Debug, Copy, Clone)]
      |          ^^^^^
      |          |
      |          impl doesn't use only types from inside the current crate
      |          `u32` is not defined in the current crate
      |
      = note: define and implement a trait or new type instead
      = note: this error originates in the derive macro `Debug` (in Nightly builds, run with -Z macro-backtrace for more info)
```
## No activated env
```
 Using esp-idf v5.1.0 at '/Users/sergio/Documents/Espressif/tests/dev51/.embuild/espressif/esp-idf/v5.1-dev'
  ERROR: tool cmake does not have versions compatible with platform macos-arm64
  Error: Could not install esp-idf

  Caused by:
      command '"/Users/sergio/Documents/Espressif/tests/dev51/.embuild/espressif/python_env/idf5.1_py3.10_env/bin/python" "/Users/sergio/Documents/Espressif/tests/dev51/.embuild/espressif/esp-idf/v5.1-dev/tools/idf_tools.py" "--idf-path" "/Users/sergio/Documents/Espressif/tests/dev51/.embuild/espressif/esp-idf/v5.1-dev" "--tools-json" "/var/folders/_w/919x4xzs7zl5kyydzg5nr_v40000gn/T/.tmpcKOpbl" "install" "cmake"' exited with non-zero status code 1

```

closed 2023-03-09 by georgik