esp-rs/rust-build #151: Docs for exporting environment do not work on Intel Mac

kevin-june, 2022-09-27:
In the [installation commands](https://github.com/esp-rs/rust-build/#installation-commands) section of the README, the docs state to run:
```sh
./install-rust-toolchain.sh --toolchain-version 1.64.0.0
. export-esp.sh
```

However, the export command doesn't seem to work on my x86 Mac:
```
% . export-esp.sh
.: no such file or directory: export-esp.sh
```

The file is present in the current working directory:
```
% ls export-esp.sh
export-esp.sh
```

I can solve this by running:
```
. ./export-esp.sh
```
or
```
% source export-esp.sh
```

Thoughts? Am I missing something or is this a typo? Thanks!

SergioGasquez, 2022-09-29:
Hi! Thanks for opening the issue, do you mind detailing your environment (macOS version, shell...). Using `. export-esp.sh` works fine in macOS Monterey 12.5 on M1 with bash 3.2.57 and fish 3.4.1.  I am happy to make the change to `. ./export-esp.sh` as it also works, but I would like to understand what is causing the issue.

kevin-june, 2022-09-29:
Certainly!

I'm using macOS Monterey 12.6 on Intel, with:
* zsh 5.9 (x86_64-apple-darwin21.3.0)
* GNU bash, version 3.2.57(1)-release (x86_64-apple-darwin21)

My terminal is configured to use `zsh`. Perhaps that is the difference?

Let me know if I can provide any other info.

SergioGasquez, 2022-10-05:
Hi! Would you mind trying with no `zsh`?

SergioGasquez, 2022-10-05:
Thanks for all your tests! We will merge #156 to solve the issue for `zsh` users.

kevin-june, 2022-10-05:
Much appreciated!

SergioGasquez, 2022-11-07:
Now that https://github.com/esp-rs/rust-build/pull/156 and https://github.com/esp-rs/rust-build/pull/165 are merged, this issue is now solved!

PR #156 (2022-11-02): Update source command
Closes #151
closed 2022-11-07 by SergioGasquez