esp-rs/rust-build #157: cargo on MacOS should be statically linked to libssl / libcrypto

pebender, 2022-10-05:
The official builds of cargo for MacOS do not have runtime dependencies on libssl and libcrypto. However, the esp build does. This is like due to not building cargo with openssl statically linked. This is controlled in https://github.com/rust-lang/cargo/blob/master/Cargo.toml using vendored-openssl.

SergioGasquez, 2022-11-04:
Hi! Can you try if the issue stills persists with the new release `1.65.0.0` version.

pebender, 2022-11-08:
I still have the same problem with https://github.com/esp-rs/rust-build/releases/tag/v1.65.0.1

tschundler, 2023-01-24:
I just recently ran into this issue and created https://github.com/esp-rs/espup/issues/141 only to learn that wasn't the right place.

Cargo in the esp-rs build for MacOS expects those libraries to live in the place homebrew (http://brew.sh) installs them (eg `brew install openssl`), which is not the case for many people. Incidentally, this is only for the x84_64 builds, the M1 aarch64 builds seem to be static linked.

closed 2023-01-26 by georgik

tschundler, 2023-01-28:
I did `espup update` and can confirm it is statically linked now (not listed in `otool -L`). Thank you.

```
$ otool -L ~/.espressif/dist/rust/rust-nightly-x86_64-apple-darwin/cargo/bin/cargo
/Users/ted/.espressif/dist/rust/rust-nightly-x86_64-apple-darwin/cargo/bin/cargo:
	/usr/lib/libiconv.2.dylib (compatibility version 7.0.0, current version 7.0.0)
	/System/Library/Frameworks/Security.framework/Versions/A/Security (compatibility version 1.0.0, current version 59306.101.1)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1675.129.0)
	/usr/lib/libcurl.4.dylib (compatibility version 7.0.0, current version 9.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
```

PR #188 (2023-01-26): WIP: Update to 1.67.0.0
Closes: #187 
Closes: #157 
Closes: #186