esp-rs/rust-build #159: Could not find specification for target "xtensa-esp32-espidf"

sk, 2022-10-15:
Hello,
i want to get started with my esp32 using rust. Unfortunately i'm unable to build the example project.

What i did so far:

`➜  Development git clone https://github.com/ivmarkov/rust-esp32-std-demo
Cloning into 'rust-esp32-std-demo'...
remote: Enumerating objects: 1009, done.
remote: Counting objects: 100% (1009/1009), done.
remote: Compressing objects: 100% (302/302), done.
remote: Total 1009 (delta 554), reused 978 (delta 538), pack-reused 0
Receiving objects: 100% (1009/1009), 663.16 KiB | 6.50 MiB/s, done.
Resolving deltas: 100% (554/554), done.
➜  Development cd rust-esp32-std-demo 
➜  rust-esp32-std-demo git:(main) curl -LO https://github.com/esp-rs/rust-build/releases/download/v1.64.0.0/install-rust-toolchain.sh
chmod a+x install-rust-toolchain.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 26193  100 26193    0     0  42359      0 --:--:-- --:--:-- --:--:-- 42359
➜  rust-esp32-std-demo git:(main) ✗ curl -LO https://github.com/esp-rs/rust-build/releases/download/v1.64.0.0/install-rust-toolchain.sh
chmod a+x install-rust-toolchain.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 26193  100 26193    0     0  63166      0 --:--:-- --:--:-- --:--:-- 63166
➜  rust-esp32-std-demo git:(main) ✗ ./install-rust-toolchain.sh -i reinstall                                                           
Processing configuration:
--build-target          = esp32,esp32s2,esp32s3
--cargo-home            = /Users/sk/.cargo
--clear-cache           = YES
--esp-idf-version       = 
--export-file           = export-esp.sh
--extra-crates          = ldproxy cargo-espflash
--installation-mode     = reinstall
--llvm-version          = esp-14.0.0-20220415
--minified-esp-idf      = NO
--minified-llvm         = YES
--nightly-version       = nightly
--rustup-home           = /Users/sk/.rustup
--toolchain-version     = 1.64.0.0
--toolchain-destination = /Users/sk/.rustup/toolchains/esp
/Users/sk/.cargo/bin/rustup
nightly-aarch64-apple-darwin
Processing toolchain for aarch64-apple-darwin - operation: reinstall
Removing:
 - /Users/sk/.rustup/toolchains/esp
 - /Users/sk/.espressif/tools/xtensa-esp32-elf-clang/esp-14.0.0-20220415-aarch64-apple-darwin
Removing cached dist files:
 - rust-1.64.0.0-aarch64-apple-darwin
 - rust-1.64.0.0-aarch64-apple-darwin.tar.xz
 - rust-src-1.64.0.0
 - rust-src-1.64.0.0.tar.xz
 - xtensa-esp32-elf-llvm14_0_0-esp-14.0.0-20220415-aarch64-apple-darwin.tar.xz
 - *-elf-gcc*.tar.gz
** downloading: https://github.com/esp-rs/rust-build/releases/download/v1.64.0.0/rust-1.64.0.0-aarch64-apple-darwin.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 77.5M  100 77.5M    0     0  8753k      0  0:00:09  0:00:09 --:--:-- 9430k
install: creating uninstall script at /Users/sk/.rustup/toolchains/esp/lib/rustlib/uninstall.sh
install: installing component 'rustc'
install: installing component 'rust-std-aarch64-apple-darwin'
install: installing component 'cargo'
install: installing component 'rustfmt-preview'
install: installing component 'clippy-preview'

    rust installed.

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
[... 68 lines cut ...]
linker = "ldproxy"

# Future - necessary for the experimental "native build" of esp-idf-sys with ESP32C3
# See also https://github.com/ivmarkov/embuild/issues/16
rustflags = ["-C", "default-linker-libraries"]

[unstable]
build-std = ["std", "panic_abort"]
#build-std-features = ["panic_immediate_abort"] # Only necessary if building against ESP-IDF tag `v4.3.2` (the minimum supported version)

[env]
# Uncomment this and use the esp-idf-sys "native" build feature (`cargo build --features native`) to build against ESP-IDF 5.0 (master)
#ESP_IDF_VERSION = { value = "master" }

# Comment out this when using the PlatformIO build (it only supports `v4.3.2`)
ESP_IDF_VERSION = { value = "release/v4.4" }

# These configurations will pick up your custom "sdkconfig.release", "sdkconfig.debug" or "sdkconfig.defaults[.*]" files
# that you might put in the root of the project
# The easiest way to generate a full "sdkconfig" configuration (as opposed to manually enabling only the necessary flags via "sdkconfig.defaults[.*]"
# is by running "cargo pio espidf menuconfig" (that is, if using the pio builder)
#ESP_IDF_SDKCONFIG = { value = "sdkconfig.release;sdkconfig.debug" }
ESP_IDF_SDKCONFIG_DEFAULTS = { value = "sdkconfig.defaults;sdkconfig.defaults.esp32;sdkconfig.defaults.esp32s2" }

Am i missing something?

SergioGasquez, 2022-10-17:
Can you `echo $LIBCLANG_PATH` after sourcing the export file? It looks like it's finding the Xtensa LLVM

sk, 2022-10-17:
➜  rust-esp32-std-demo git:(main) ✗ echo $LIBCLANG_PATH
/Users/sk/.espressif/tools/xtensa-esp32-elf-clang/esp-14.0.0-20220415-aarch64-apple-darwin/lib/
➜  rust-esp32-std-demo git:(main) ✗ ls $LIBCLANG_PATH
clang          libclang.dylib
➜  rust-esp32-std-demo git:(main) ✗

SergioGasquez, 2022-10-18:
It looks like you are not using the  `esp` toolchain, try with `cargo +esp espflash  /dev/tty.usbserial-0001` or any other[ override method](https://rust-lang.github.io/rustup/overrides.html). 

By the way, I would recommend posting questions like this one in the [`#esp-rs:matrix.org` Matrix room](https://matrix.to/#/#esp-rs:matrix.org).

sk, 2022-10-18:
Hello,
thanks for the hint. Unfortunately it didn't work:
➜  rust-esp32-std-demo git:(main) ✗ cargo +esp espflash /dev/tty.usbserial-0001
error: no such subcommand: `+esp`

	Did you mean `help`?

But due to this error i figured the issue out.

It seems like the rust version installed via homebrew isn't compatible.
After uninstalling the homebrew rust the binary is now just too big but that's a different topic.

Thanks a lot!

closed 2022-10-18 by sk

SergioGasquez, 2022-10-18:
Try using release mode: `cargo +esp espflash --release /dev/tty.usbserial-0001`

sk, 2022-10-18:
➜  rust-esp32-std-demo git:(main) ✗ cargo +esp espflash /dev/tty.usbserial-0001 --release
Serial port: /dev/tty.usbserial-0001
Connecting...

    Finished release [optimized] target(s) in 0.27s
Chip type:         ESP32 (revision 1)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC address:       ac:67:b2:38:94:60
App/part. size:    1422448/1048576 bytes, 135.66%
Error:
  × Supplied elf image is too big and doesn't fit configured app partition

Tried that too, still too large...

SergioGasquez, 2022-10-18:
Is your project open source?

sk, 2022-10-18:
I'm currently building the demo application: https://github.com/ivmarkov/rust-esp32-std-demo

SergioGasquez, 2022-10-18:
Demo application migth be too big for some chips, I would recommend starting with one of the templates:
- [std](https://github.com/esp-rs/esp-idf-template): based on esp-idf
- [no-std ](https://github.com/esp-rs/esp-template): pure rust bare metal

sk, 2022-10-18:
I stripped the example project down until it fitted into the available space.
Now i'm able to use it.

Again: Thanks a lot!