esp-rs/rust-build #160: Builds with nightly are failing - display information about working version on GH page

georgik, 2022-10-17:
Build for C3 is failing with latest nightly.
It would be very useful to be explicit about which version of nightly is working.
The last working version known to me is `+nightly-2022-08-15`

SergioGasquez, 2022-10-17:
The CI is passing successfully with all 3 latest nightly, which means that the esp-template project is building successfully with that nightly version: https://github.com/esp-rs/rust-build/actions/runs/3263489115/jobs/5362430878#step:5:21

How is latest nightly failing for you?

georgik, 2022-10-24:
Simulated problem with Windows 11
```
error: cannot find attribute `default_method_body_is_const` in this scope
   --> C:\Users\User\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\clone.rs:132:7
    |
132 |     #[default_method_body_is_const]
    |       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: could not compile `core` due to 6 previous errors
warning: build failed, waiting for other jobs to finish...
PS C:\projects\esp-hal\esp32c3-hal> rustc --version
rustc 1.66.0-nightly (dcb376115 2022-10-20)
```

SergioGasquez, 2022-10-24:
I assume you were building one of the esp-hal examples, right? Can you try building the following project with the same toolchain? 
```sh
cargo generate --git https://github.com/esp-rs/esp-template --name test --vcs none --silent -d mcu=esp32c3 -d devcontainer=false -d alloc=false
```

georgik, 2022-10-24:
Command: ` cargo build --release`

```
error: cannot find attribute `default_method_body_is_const` in this scope
   --> C:\Users\User\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\clone.rs:132:7
    |
132 |     #[default_method_body_is_const]
    |       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

Command: ` cargo "+nightly-2022-08-15" build --release`

```
   Compiling test v0.1.0 (C:\projects\test)
    Finished release [optimized] target(s) in 35.82s
```

SergioGasquez, 2022-11-02:
I [updated the nightly CI check action on a branch](https://github.com/esp-rs/rust-build/blob/feature/nightly-checks/.github/workflows/nigthly-check.yaml) and I was able to successfully build `esp-template` project for `esp32c3` in Windows, see: https://github.com/esp-rs/rust-build/actions/runs/3337353944/jobs/5523591802

closed 2022-12-20 by georgik