esp-rs/rust-build #163: install script some problems in windows (Environment variables)

pan-xw, 2022-10-25:
1, No process RUSTUP_HOME and CARGO_HOME Environment variables. but in not windows (.sh file) is done.
2, The LIBCLANG_PATH set to file is bug? If set to the file(libclang.dll), then cannot compile, report: Could not find specification for target "xtensa-esp32-espidf" like [https://github.com/ivmarkov/rust-esp32-std-demo/issues/20](url). I set to the path where is the libclang.dll, and then ok.

SergioGasquez, 2022-10-25:
Do you mind using [`espup`](https://github.com/esp-rs/espup) for the installation?  It is a rust version of the installers' scripts, and we are slightly moving to this approach.  Using `espup` allows [modifying the installation path via `RUSTUP_HOME` and `CARGO_HOME` environment variables](https://github.com/esp-rs/espup#install-subcommand).

pan-xw, 2022-10-25:
> Do you mind using [`espup`](https://github.com/esp-rs/espup) for the installation? It is a rust version of the installers' scripts, and we are slightly moving to this approach. Using `espup` allows [...]

Good!  I didn't  find [`espup`](https://github.com/esp-rs/espup) before. I will try it.

SergioGasquez, 2022-10-25:
> Good! I didn't find [`espup`](https://github.com/esp-rs/espup) before. I will try it.

Let me know how it goes!

pan-xw, 2022-10-31:
> > Good! I didn't find [`espup`](https://github.com/esp-rs/espup) before. I will try it.
> 
> Let me know how it goes!

 I've tried. It build esp32 and esp32c3 ok. Thank you for the work!

closed 2022-10-31 by pan-xw