esp-rs/rust-build #164: Under Ubuntu, getting "Could not find specification for target "xtensa-esp32-espidf"" error

mbratch, 2022-10-27:
I followed all of the instructions step by step under https://github.com/esp-rs/rust-build. When I finally got to the demo program and tried to build with:

`cargo build`

or with

`cargo build --target xtensa-esp32-espidf`

I got the following error:

```
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --target xtensa-esp32-espidf --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (exit status: 1)
  --- stderr
  error: Error loading target specification: Could not find specification for target "xtensa-esp32-espidf". Run `rustc --print target-list` for a list of built-in targets

```
So there's a step that's needed to make the new target known to `rustc`, but I'm not sure what it is.

SergioGasquez, 2022-10-28:
Hi! Is your project open source by any chance? If not, do you mind sharing a minimal reproducer of the issue? It doesn't look like an installation problem to me.

mbratch, 2022-10-28:
@SergioGasquez  I don't have any custom code. I only downloaded the demo per the instructions in this repository and was unable to build due to the error I posted.

SergioGasquez, 2022-10-28:
Can you try building a template project?
```sh
cargo install cargo-generate
cargo generate  https://github.com/esp-rs/esp-idf-template cargo --name demo-app --vcs none --silent -d mcu=esp32 -d std=false -d espidfver=v4.4 -d devcontainer=false
cd demo-app
cargo build
```

mbratch, 2022-10-28:
Thanks @SergioGasquez .

When I tried the first step: `cargo install cargo-generate` it gave me an error at the end:

```
error: could not compile `cargo-generate`

Caused by:
  process didn't exit successfully: `rustc --crate-name cargo_generate --edition=2021 /home/mark/.cargo/registry/src/github.com-1ecc6299db9ec823/cargo-generate-0.16.0/src/main.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C opt-level=3 -C lto -C metadata=9a6dfd88802ba74b -C extra-filename=-9a6dfd88802ba74b --out-dir /tmp/cargo-install8OXqxw/release/deps -L dependency=/tmp/cargo-install8OXqxw/release/deps --extern anyhow=/tmp/cargo-install8OXqxw/release/deps/libanyhow-19347f4d271cbe3f.rlib --extern cargo_generate=/tmp/cargo-install8OXqxw/release/deps/libcargo_generate-eb025e12118298a2.rlib --extern clap=/tmp/cargo-install8OXqxw/release/deps/libclap-6662d9b03292cff7.rlib --extern console=/tmp/cargo-install8OXqxw/release/deps/libconsole-13db54020c254e3c.rlib --extern dialoguer=/tmp/cargo-install8OXqxw/release/deps/libdialoguer-7f1547d7070e5a3c.rlib --extern dirs=/tmp/cargo-install8OXqxw/release/deps/libdirs-2af025c743973e7b.rlib --extern git_config=/tmp/cargo-install8OXqxw/release/deps/libgit_config-486c49a5d6226dc7.rlib --extern git2=/tmp/cargo-install8OXqxw/release/deps/libgit2-835963d7bcc7c3b7.rlib --extern heck=/tmp/cargo-install8OXqxw/release/deps/libheck-b232a84a8d490530.rlib --extern home=/tmp/cargo-install8OXqxw/release/deps/libhome-0b4aa797323f5fc5.rlib --extern ignore=/tmp/cargo-install8OXqxw/release/deps/libignore-b544de13a0a8d260.rlib --extern indicatif=/tmp/cargo-install8OXqxw/release/deps/libindicatif-d0d034123acdabbd.rlib --extern liquid=/tmp/cargo-install8OXqxw/release/deps/libliquid-c23a42f41be85c5c.rlib --extern liquid_core=/tmp/cargo-install8OXqxw/release/deps/libliquid_core-ff84de9f890d8ac0.rlib --extern liquid_derive=/tmp/cargo-install8OXqxw/release/deps/libliquid_derive-5277c561ab29e8f4.so --extern liquid_lib=/tmp/cargo-install8OXqxw/release/deps/libliquid_lib-698366a80320f93f.rlib --extern names=/tmp/cargo-install8OXqxw/release/deps/libnames-4cfe2f0e39b8142f.rlib --extern paste=/tmp/cargo-install8OXqxw/release/deps/libpaste-2428e6b05bd704d6.so --extern path_absolutize=/tmp/cargo-install8OXqxw/release/deps/libpath_absolutize-bbad45c80d05a3c1.rlib --extern regex=/tmp/cargo-install8OXqxw/release/deps/libregex-9da9ec6bbadc45a6.rlib --extern remove_dir_all=/tmp/cargo-install8OXqxw/release/deps/libremove_dir_all-0d4e3686e4f16249.rlib --extern rhai=/tmp/cargo-install8OXqxw/release/deps/librhai-d1046045481ebbe1.rlib --extern sanitize_filename=/tmp/cargo-install8OXqxw/release/deps/libsanitize_filename-356fb2bbae07586c.rlib --extern semver=/tmp/cargo-install8OXqxw/release/deps/libsemver-b4899c52fe338db8.rlib --extern serde=/tmp/cargo-install8OXqxw/release/deps/libserde-f6f60b35a9da74b5.rlib --extern tempfile=/tmp/cargo-install8OXqxw/release/deps/libtempfile-eb87032830af92eb.rlib --extern thiserror=/tmp/cargo-install8OXqxw/release/deps/libthiserror-4ffd994849819bb4.rlib --extern toml=/tmp/cargo-install8OXqxw/release/deps/libtoml-1a4dba6520269a6a.rlib --extern walkdir=/tmp/cargo-install8OXqxw/release/deps/libwalkdir-045a5540833b901a.rlib --cap-lints allow -L native=/tmp/cargo-install8OXqxw/release/build/libgit2-sys-fce4149bb0e706f1/out/build -L native=/tmp/cargo-install8OXqxw/release/build/libssh2-sys-c4b5ad2669c48a50/out/build -L native=/tmp/cargo-install8OXqxw/release/build/openssl-sys-94aeaacdff9c4494/out/openssl-build/install/lib` (signal: 8, SIGFPE: erroneous arithmetic operation)
error: failed to compile `cargo-generate v0.16.0`, intermediate artifacts can be found at `/tmp/cargo-install8OXqxw`
```

SergioGasquez, 2022-10-28:
I would suggest trying:
- `cargo install cargo-generate@0.15.2`
- Download and use one of the binary artifacts: https://github.com/cargo-generate/cargo-generate/releases/tag/v0.15.2

mbratch, 2022-10-28:
@SergioGasquez `cargo install cargo-generate@0.15.2` failed as well, same error.

So I tried the tarball and that got me past the generate stage. When I tried to build the application it got nearly all the way through all the compiles and then generated this error:

`error: failed to run custom build command for `esp-idf-sys v0.31.10`
`
That was followed by a huge amount of output.

SergioGasquez, 2022-10-28:
> @SergioGasquez `cargo install cargo-generate@0.15.2` failed as well, same error.
> 
> So I tried the tarball and that got me past the generate stage. When I tried to build the bare application (without changing it) it got nearly all the way through all the compiles and then generated this error:
> 
> ```
> error: failed to run custom build command for `esp-idf-sys v0.31.10`
> ```
> 
> That was followed by a huge amount of output.

Can you share the tail of it?

mbratch, 2022-10-28:
@SergioGasquez sure thing!
Attached.
[ErrorOutput.txt](https://github.com/esp-rs/rust-build/files/9891272/ErrorOutput.txt)

SergioGasquez, 2022-10-28:
Yes, the error is happening when building ESP-IDF stuff.  For some reason, it's picking `xtensa` instead of `xtensa-esp32-espidf` as your target triple. Are you sourcing the export file? Do you mind sharing what instructions did you used to install the environment and to build the project? 

I won't be able to provide more help until Monday, something that you can try is:
- Delete the `esp` toolchain: `rm - rf $/esp`
- [Install the following dependencies](https://github.com/esp-rs/espup#linux)
- Install `espup`: `cargo install espup --git https://github.com/esp-rs/espup`
- Install the Rust on ESPs environment: `espup install --espidf-version "release/v4.4" --targets "esp32"`
- Source the export file generated by espup: . ./export-esp.sh 
- go to the template project and try to build it

SergioGasquez, 2022-10-28:
I just saw in our matrix channel that some user was having the same issue while building `rust-esp32-std-demo` project : https://matrix.to/#/!LdaNPfUfvefOLewEIM:matrix.org/$dSDa19_VX39HJ2MhLeRsFD5bLp5JBfrUou2JoFU9e8A?via=matrix.org&via=tchncs.de&via=envs.net

mbratch, 2022-10-28:
Thanks @SergioGasquez 

The instructions I followed are here: https://github.com/esp-rs/rust-build/blob/main/README.md.

I'll try your suggestions. This isn't urgent. I hope you have a great weekend!

jessiahr, 2022-10-29:
It seems I have also hit this same issue tonight: 
```
error[E0658]: where clauses on associated types are unstable
   --> /home/jessiahr/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:300:5
    |
300 | /     type WriteFuture<'a> = impl Future<Output = Result<usize, Self::Error>> + 'a
301 | |     where
302 | |         Self: 'a;
    | |_________________^
    |
    = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
    = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable

error[E0658]: generic associated types are unstable
   --> /home/jessiahr/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:309:5
    |
309 | /     type FlushFuture<'a> = impl Future<Output = Result<(), Self::Error>> + 'a
310 | |     where
311 | |         Self: 'a;
    | |_________________^
    |
    = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
    = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable

error[E0658]: where clauses on associated types are unstable
   --> /home/jessiahr/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:309:5
    |
309 | /     type FlushFuture<'a> = impl Future<Output = Result<(), Self::Error>> + 'a
310 | |     where
311 | |         Self: 'a;
    | |_________________^
    |
    = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
    = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable

error[E0658]: generic associated types are unstable
   --> /home/jessiahr/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:322:5
    |
322 | /     type SeekFuture<'a> = impl Future<Output = Result<u64, Self::Error>> + 'a
323 | |     where
324 | |         Self: 'a;
    | |_________________^
    |
    = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
    = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable

error[E0658]: where clauses on associated types are unstable
   --> /home/jessiahr/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:322:5
    |
322 | /     type SeekFuture<'a> = impl Future<Output = Result<u64, Self::Error>> + 'a
323 | |     where
324 | |         Self: 'a;
    | |_________________^
    |
    = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
    = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable

error[E0658]: generic associated types are unstable
   --> /home/jessiahr/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:335:5
    |
335 | /     type WriteFuture<'a> = impl Future<Output = Result<usize, Self::Error>> + 'a
336 | |     where
337 | |         Self: 'a;
    | |_________________^
    |
    = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
    = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable

error[E0658]: where clauses on associated types are unstable
   --> /home/jessiahr/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:335:5
    |
335 | /     type WriteFuture<'a> = impl Future<Output = Result<usize, Self::Error>> + 'a
336 | |     where
337 | |         Self: 'a;
    | |_________________^
    |
    = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
    = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable

error[E0658]: generic associated types are unstable
   --> /home/jessiahr/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:347:5
    |
347 | /     type FlushFuture<'a> = impl Future<Output = Result<(), Self::Error>>
348 | |     where
349 | |         Self: 'a;
    | |_________________^
    |
    = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
    = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable

error[E0658]: where clauses on associated types are unstable
   --> /home/jessiahr/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:347:5
    |
347 | /     type FlushFuture<'a> = impl Future<Output = Result<(), Self::Error>>
348 | |     where
349 | |         Self: 'a;
    | |_________________^
    |
    = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
    = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable

   Compiling hash32 v0.2.1
For more information about this error, try `rustc --explain E0658`.
error: could not compile `embedded-io` due to 42 previous errors
warning: build failed, waiting for other jobs to finish...
```

I may try rolling back to past versions

mbratch, 2022-10-29:
After going through the steps suggested by @SergioGasquez on 28/10/2022 4:21P, I now have the following scenario.

I have three different demo apps on my system.

- The demo application given by this repository (`rust-esp32-std-demo`) still fails to compile due to the "where clauses on associated types are unstable" error as I originally reported.
- The `demo-app` which I generated (but did not modify) using the generator steps indicated by @SergioGasquez on 28/10/2022 12:47P now actually compiles successfully!
- A `demo` application (which now I can't remember where I got it from but it's dead simple and I have attached it to this comment) has several compilation errors saying that `Ok(...) is not a tuple struct or a tuple variant`, for example:
```
error[E0532]: expected tuple struct or tuple variant, found function `Ok`
   --> ~/.cargo/registry/src/github.com-1ecc6299db9ec823/embuild-0.24.5/src/utils.rs:123:13
    |
123 |             Ok(result) => {
    |             ^^ not a tuple struct or tuple variant
    |
   ::: ~/.cargo/registry/src/github.com-1ecc6299db9ec823/embuild-0.24.5/src/bindgen.rs:164:5
    |
164 |     cmd!("rustup", "run", "stable", "rustfmt", output_file)?;
    |     ------------------------------------------------------- in this macro invocation
    |
    = note: this error originates in the macro `cmd` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing one of these items instead
   --> |~/.cargo/registry/src/github.com-1ecc6299db9ec823/embuild-0.24.5/src/bindgen.rs:1:1
    |
1   | use core::result::Result::Ok;
   --> |~/.cargo/registry/src/github.com-1ecc6299db9ec823/embuild-0.24.5/src/bindgen.rs:1:1
    |
1   | use serde::__private::Ok;
   --> |~/.cargo/registry/src/github.com-1ecc6299db9ec823/embuild-0.24.5/src/bindgen.rs:1:1
    |
1   | use std::result::Result::Ok;

```
[demo.zip](https://github.com/esp-rs/rust-build/files/9893786/demo.zip)

bsodmike, 2022-10-30:
Running into the exact same thing here

```
error[E0658]: generic associated types are unstable
   --> /usr/local/cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:347:5
    |
347 | /     type FlushFuture<'a> = impl Future<Output = Result<(), Self::Error>>
348 | |     where
349 | |         Self: 'a;
    | |_________________^
    |
    = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
    = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable

error[E0658]: where clauses on associated types are unstable
   --> /usr/local/cargo/registry/src/github.com-1ecc6299db9ec823/embedded-io-0.3.1/src/asynch.rs:347:5
    |
347 | /     type FlushFuture<'a> = impl Future<Output = Result<(), Self::Error>>
348 | |     where
349 | |         Self: 'a;
    | |_________________^
    |
    = note: see issue #44265 <https://github.com/rust-lang/rust/issues/44265> for more information
    = help: add `#![feature(generic_associated_types)]` to the crate attributes to enable
```

```
active toolchain
----------------

esp (directory override for '/root/esp/rust-esp32-std-demo')
rustc 1.64.0-nightly (f53d2361e 2022-09-20)
```

When will the install script get updated to support rust 1.65?

SergioGasquez, 2022-10-31:
> The demo application given in the README.md of this repository (rust-esp32-std-demo) still fails to compile due to the "where clauses on associated types are unstable" error as I originally reported.

This is expected and will be solved once we release `1.65.0.0` toolchain, which stabilizes GATs. You can try some of the temporary solutions suggested on the matrix channel.

> @bsodmike: When will the install script get updated to support rust 1.65?

Acording to [Rust forge](https://forge.rust-lang.org/#current-release-versions), 1.65 will be stable on Nov 3rd. Then, so we will need to rebase our changes, verify that everything works and build the new binaries, this usually takes us from 1 day to a week.  

> The demo-app which I generated (but did not modify) using the generator steps @SergioGasquez kindly provided on 28/10/2022 12:47P now actually compiles successfully!

Happy to read that! 

> A demo application (which now I can't remember where I got it from but it's very simple and I have attached it to this comment) has several compilation errors saying that `Ok(...) is not a tuple struct or a tuple variant`.

This `demo` project is really really old (you can compare the dependencies versions with the project that is building fine and you'll see how old it is). I would assume that it was the template project long ago!

mbratch, 2022-11-01:
> This `demo` project is really really old (you can compare the dependencies versions with the project that is building fine and you'll see how old it is). I would assume that it was the template project long ago!

ikr? Not sure where I found it. The site didn't have an obvious date stamp. Too bad the internet doesn't have some kind of aging policy where it retires technology pages if they get too old. :D

SergioGasquez, 2022-11-04:
Xtensa Rust `1.65.0.0` just go released, do you mind trying to reproduce the GATs issue with the new version of the toolchain? it should now be fixed!

SergioGasquez, 2022-11-04:
Awesome! I'll then close the issue, thanks for verifying it @bsodmike!

closed 2022-11-04 by SergioGasquez

mbratch, 2022-11-04:
I updated to Rust 1.65.0 and I still get the same error. I think it was premature to close this issue.

bsodmike, 2022-11-04:
@mbratch as I recall the script uses nightly, but can you try this please 

```
./install-rust-toolchain.sh --toolchain-version 1.65.0.0
```

SergioGasquez, 2022-11-04:
You can also use `espup`, starting from `v0.2.0` of `espup` you could just do `espup update` and it will update the toolchain to the latest version

bsodmike, 2022-11-04:
@SergioGasquez not sure if my PR suggestion would help too?

SergioGasquez, 2022-11-04:
> @SergioGasquez not sure if my PR suggestion would help too?

Thanks for making the PR and sorry for closing it. We will update the default version once branch `build/1.65.0.0` gets merged to `main`, which should be soon. Just awaiting some pair reviewing in some small stuff.

mbratch, 2022-11-04:
@bsodmike  @SergioGasquez  this is a little strange. It looks like I'm still on `rustc` 1.64.0. When I had run `rustup install` it did say this:

```
info: syncing channel updates for 'stable-x86_64-unknown-linux-gnu'
info: latest update on 2022-11-03, rust version 1.65.0 (897e37553 2022-11-02)
info: downloading component 'rust-std' for 'thumbv6m-none-eabi'
  9.3 MiB /   9.3 MiB (100 %)   2.8 MiB/s in  3s ETA:  0s
info: downloading component 'cargo'
  6.5 MiB /   6.5 MiB (100 %)   2.8 MiB/s in  2s ETA:  0s
info: downloading component 'clippy'
  2.9 MiB /   2.9 MiB (100 %)   2.8 MiB/s in  1s ETA:  0s
```
etc...

So I thought it was updating to 1.65.0. When I ran `espup update` it said:

```
[2022-11-04T18:27:54Z INFO ] 💽  Updating Xtensa Rust toolchain
[2022-11-04T18:27:54Z INFO ] ✅  Toolchain '1.64.0.0' is already up to date
```
I guess I'll try the script. I'm not sure why these tools left me at 1.64.0.

SergioGasquez, 2022-11-04:
Try the following: `espup update -v "1.65.0.0"`
The problem is `espup@0.1.0`, the version that you have installed, does not support updating very well. 

I would recommend uninstalling the environment (`espup uninstall`), updating `espup` (`cargo install espup --force`) and reinstalling the environment (`espup install`) . This way, `espup ` will create a file with your installation details and would be able to update and handle your toolchain properly.

mbratch, 2022-11-04:
Thanks very much @SergioGasquez.

After doing `espup update -v "1.65.0.0"` it installed the correct `rustc` version and the demo compiled!

I did try to correct my `espup` but ran into errors in that process. I may have to just remove everything and start from the top. Should I be installing all this as a normal user or as root? None of the instructions said (or at least I didn't notice). Lately I've been installing as normal user but I might have started out early on with `root`, not knowing for sure.

I think this issue can remain closed as far as I'm concerned. Would it be much trouble to list the steps to uninstall all the rust/esp stuff and to reinstall it?

SergioGasquez, 2022-11-05:
> I did try to correct my `espup` but ran into errors in that process. I may have to just remove everything and start from the top.

If you did not have any other Espressif thing installed, what you can do is:
1. Delete `.espressif` folder: `rm -rf $HOME/.espressif`
2. Delete `esp` toolchain: `rm -rf $RUSTUP_HOME/toolchains/esp`. If no `$RUSTUP_HOME` defined it will be `$HOME/.rustup`
3. Update `espup`: cargo uninstall espup && cargo install espup. Maybe `cargo install `espup --force` works but since it was previously installed with `--git` option I am not sure if it works, so uninstall and install works 100%.
4. Install the environment with `espup`: `espup install` or `espup install --espidf-version "release/v4.4" --targets "esp32"` depending on what projects/targets would you like to build. See [usage section of `espup`](https://github.com/esp-rs/espup#install-subcommand) and [explanation on the book](https://esp-rs.github.io/book/installation/installation.html#espup) for more info.

mbratch, 2022-11-05:
Thanks @SergioGasquez. Those steps are basically what I ended up doing and it seems to be working fine now.

I will explore the documentation more as what I want to do is build on one PC (running Linux) but program the ESP32 from a different PC. I think the basic app from cargo generate uses the default ESP-IDF partitions of bootloader @ 0x1000, partition table @ 0x8000, and application at 0x10000. I just need to gather the correct bin file products after the build.

SergioGasquez, 2022-11-05:
> Thanks @SergioGasquez. Those steps are basically what I ended up doing and it seems to be working fine now.

Awesome! Regarding your application, I suggest you also have a look at [`cargo-espflash`](https://github.com/esp-rs/espflash/tree/main/cargo-espflash#usage) and [`espflash`](https://github.com/esp-rs/espflash/tree/main/espflash#usage)  documentation as commands as `cargo espflash save-image` can be useful for your project.

mbratch, 2022-11-05:
@SergioGasquez although the program runs and I get the expected init message on the console, the `println!` output from `main` is not appearing on the console. I'm connected via USB. Does the `println!` go to the same console port as the init messages? Or is there a configuration item that determines it?

mbratch, 2022-11-05:
@SergioGasquez  thanks for the tip! Yes, the `cargo espflash save-image` saves the application image. I can flash it to the device along with bootloader and partition table and it works. :)

SergioGasquez, 2022-11-05:
By the way, feel free to join our matrix channel, it's the perfect place to ask questions and chat with the community: https://matrix.to/#/#esp-rs:matrix.org