esp-rs/rust-build #172: Version 1.65.0.0 doesn't work on Arch Linux

kyrias, 2022-11-07:
I'm not sure how to actually solve this properly, but as it stands the 1.65.0.0 release doesn't work on Arch Linux because the new Espressif LLVM toolchain has a `libLLVM` that links against `libtinfo.so.5`, whereas the Arch Linux ncurses package has been on ncurses 6 for years.

I guess the question is which OS and dependency versions the project wants to target. It feels a bit strange to me though for a new version to end up requiring an 11 year old dependency.

SergioGasquez, 2022-11-07:
Hi @kyrias! See https://bbs.archlinux.org/viewtopic.php?id=203039, which may solve your issue. Our next LLVM release will have updated libtinfo dependency.

kyrias, 2022-11-07:
Installing AUR packages in our CI would be a lot of effort so we're just going to stay pinned to the previous version for now.  Great to hear that it will be fixed in the next LLVM release!

SergioGasquez, 2022-11-07:
Thanks for your understanding, if you want to further elaborate on this, please open an issue on https://github.com/espressif/llvm-project

closed 2022-11-07 by SergioGasquez

ivmarkov, 2022-12-06:
@SergioGasquez Do we have at least an open issue at the Espressif LLVM (or is it upstream LLVM?) that we can link here? You know, people keep complaining: https://github.com/esp-rs/esp-idf-template/issues/68#issuecomment-1339536428