esp-rs/rust-build #185: Windows installation fails

ardave, 2022-12-23:
I run 

https://github.com/esp-rs/rust-build/releases/download/v1.66.0.0/Install-RustToolchain.ps1

and it fails with

_* installing C:\Users\dave/.espressif/tools/xtensa-esp32-elf-clang/esp-15.0.0-20221201-x86_64-pc-windows-msvc
** downloading: **https://github.com/esp-rs/rust-build/releases/download/llvm-project-14.0-minified/xtensa-esp32-elf-llvm15_0_0-esp-15.0.0-20221201-x86_64-pc-windows-msvc.zip**
Invoke-WebRequest : Not Found
At C:\Users\dave\Downloads\Install-RustToolchain.ps1:146 char:9
+         Invoke-WebRequest "${LlvmUrl}" -OutFile ${LlvmFile}
+         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (System.Net.HttpWebRequest:HttpWebRequest) [Invoke-WebRequest], WebException
    + FullyQualifiedErrorId : WebCmdletWebResponseException,Microsoft.PowerShell.Commands.InvokeWebRequestCommand_

... a "Not Found" which I can verify by using curl.

Apologies that I do not know how to "browse" the release files available on github to troubleshoot.

JonGrasses, 2022-12-26:
I'm also having this problem.

steventrouble, 2022-12-30:
A lot of things broke in the script when esp32 LLVM upstream changed their links. 

- They changed the name of the file, so the script can't find it
- They switched to `tar.xz` archives instead of `zip`, so windows can't extract it from command line

I manually completed the installation by doing the following:

1. Download the latest win64 release (not libs) from [espressif's github](https://github.com/espressif/llvm-project/releases/)
2. Extract the files to the local directory using 7zip
3. Move the the `esp-clang` directory into the `.../tools/xtensa-esp32-elf-clang` directory 

You can edit the installer ps1 script to output the values of various variables if you need to see where certain things are installed.

TBH, it's a lot of work and the toolchain doesn't feel like it's ready for prime time yet. It ought to be a single command. Maybe I'll give it another try some other day.

SergioGasquez, 2022-12-30:
Hi! I'm sorry for not getting back to you sooner. Most of our team is now on vacation. We will probably look at this issue the next week, in the meantime, I can recommend having a look at [`espup`](https://github.com/esp-rs/espup), a new tool written in Rust that aims to replace installation scripts. [See the `espup` section of the book for some additional documentation](https://esp-rs.github.io/book/installation/installation.html#espup). 
The followings commands should install everything required to develop Rust applications on ESP chips:
```ps1
cargo install espup
espup install 
%USERPROFILE%\export-esp.ps1
```

SergioGasquez, 2023-01-23:
Hi @ardave, @JonGrasses

Have you tried using `espup` to workaround this issue? If so, what were the results?

ardave, 2023-02-01:
> Hi @ardave, @JonGrasses
> 
> Have you tried using `espup` to workaround this issue? If so, what were the results?

Yes!  It looks like your espup-related commands followed by this https://esp-rs.github.io/book/installation/troubleshooting.html#using-wrong-rust-toolchain have me back on track.  Thanks a million! 

(I'll leave it up to you to close, just in case anyone else needs to chime in.)

georgik, 2023-03-09:
`espup` is now the official way how for Windows deployment. PowerShell script was removed.

closed 2023-03-09 by georgik

ryankopf, 2024-03-15:
See this PR if you are trying to use this on Windows: https://github.com/esp-rs/espup/pull/420

The espup installer claims the variables are set, even if they don't get set. I am suggesting a change to the message, but you can just run the `C:\Users\yourusername\export-esp.ps1` file and things start working.