esp-rs/rust-build #213: ld: region `drom0_0_seg' overflowed

hbhong, 2023-06-13:
Hello,
My codes have worked until more and more codes are developed. When I run "cargo build", it throws the following errors

`  
= note: [ldproxy] Running ldproxy
          Error: Linker /home/<my_home>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc failed: exit status: 1
          STDERR OUTPUT:
          /home/<my_home>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld: /drn/RustProjects/pomodoro/target/riscv32imc-esp-espidf/debug/deps/Pomodoro-1b9aac50edb587b3 section ".flash.rodata" will not fit in region `drom0_0_seg'
          /home/<my_home>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld: region "drom0_0_seg" overflowed by 380184 bytes
          collect2: error: ld returned 1 exit status
`
here is my env:

- Ubuntu 22 
- ESP-IDF 5.0
- MCU: ESP32 C3

I found some solutions for it, but they are all for Arduino and C language, not rust, how to fix it?
Your help would be greatly appreciated!

MabezDev, 2023-06-14:
Your application is too big to fit into ROM (flash). Make sure you're building with `--release` flag and/or look into reducing the size of your application.

hbhong, 2023-06-15:
@MabezDev Thanks a lot, `--release` flag works, I am concerned that even this flag may not be sustainable for longer periods of time as more features are developed.

closed 2023-06-15 by MabezDev