esp-rs/rust-build #218: cargo build error

cmehmet, 2023-06-24:
This is the first error i get;;;
error: in C:\blink\rust-toolchain.toml: invalid toolchain name: 'esp'

After that i change the toolchain name esp to stable in that path C:\blink\rust-toolchain.toml but i get second error like the below;;;

error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --target xtensa-esp32-espidf --crate-type bin 
--crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=split-debuginfo --print=crate-name --print=cfg` (exit code: 1)
  --- stderr
  error: Error loading target specification: Could not find specification for target "xtensa-esp32-espidf". Run `rustc --print 
target-list` for a list of built-in targets

[image]

[image]

I am grateful for any kind of help

MabezDev, 2023-06-24:
The Xtensa enabled toolchain is not installed, please follow the instructions outlined in the book: https://esp-rs.github.io/book/.

closed 2023-06-24 by MabezDev

cmehmet, 2023-06-24:
Thank you so much for help