esp-rs/rust-build #226: thread 'main' panicked at 'Unable to find libclang:

Horryportier, 2023-07-23:
I've tried to flash my esp32 but i encountered issue

i've installed followed instruction in readme and installed espflash  and generated basic esp-idf-template 
also i've installed libclang with `sudo apt-get install -y libclang-dev`

cargo run  output: [bug.txt](https://github.com/esp-rs/rust-build/files/12138612/bug.txt)

Os:  Pop!_OS 22.04  
Rust: rustc 1.70.0-nightly (f112def22 2023-05-31) (1.70.0.1)

SergioGasquez, 2023-07-24:
Can you check if you have the `LIBCLANG_PATH` environment variable set, that variable must be set by sourcing the environment file, see https://esp-rs.github.io/book/installation/index.html#3-set-up-the-environment-variables

Horryportier, 2023-07-24:
LIBCLANG_PATH is set to `$HOME/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516/esp-clang/lib`

SergioGasquez, 2023-07-24:
> LIBCLANG_PATH is set to `$HOME/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516/esp-clang/lib`

Are you building the project in the same terminal where you source the export file or print the `LIBCLANG_PATH`? Also, check the content of that folder just in case is empty (?)

SergioGasquez, 2023-07-24:
> found out `$HOME/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-16.0.0-20230516` is empty

Try reinstalling the environment with `espup`, do you remember having any issues during the installation?

SergioGasquez, 2023-07-24:
I will be closing the issue, feel free to reopen the issue if you encounter any issue after reinstalling.

closed 2023-07-24 by SergioGasquez

Horryportier, 2023-07-24:
i have no longer issue with `LIBCLANG_PATH` but now i have enqunterd this build error: 
```
error[E0063]: missing field `home_chan_dwell_time` in initializer of `esp_idf_sys::wifi_scan_config_t`
  --> /home/horryportier/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.46.0/src/wifi.rs:93:13
   |
93 |             Self {
   |             ^^^^ missing `home_chan_dwell_time`

For more information about this error, try `rustc --explain E0063`.
error: could not compile `esp-idf-svc` (lib) due to previous error
warning: build failed, waiting for other jobs to finish...
```

SergioGasquez, 2023-07-24:
> i have no longer issue with `LIBCLANG_PATH` but now i have enqunterd this build error:
>
> ```
> error[E0063]: missing field `home_chan_dwell_time` in initializer of `esp_idf_sys::wifi_scan_config_t` [...]

See https://github.com/esp-rs/esp-idf-template/issues/139. It should be fixed by adding a patch for `esp-idf-svc` .