esp-rs/rust-build #228: Target not found issue in mac.

yashwanthsinghm, 2023-07-27:
Hi I'm trying to compile code esp32 in rust on Mac
I have installed espup on machine by following commands, I have used the following commands. 
1. git clone https://github.com/esp-rs/espup
2. cargo install espup
3. espup install
4. chmod a+x $HOME/export-esp.sh
5. $HOME/export-esp.sh
6. espup install -t all

I’m trying to compile example esp32 code I’m getting the following error.

how to add target?

```
 error: failed to run `rustc` to learn about target-specific information
Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names -C link-arg=-Tlinkall.x -C link-arg=-Trom_functions.x --target xtensa-esp32-none-elf --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (exit status: 1)
  --- stderr
  error: Error loading target specification: Could not find specification for target "xtensa-esp32-none-elf". Run `rustc --print target-list` for a list of built-in targets

SergioGasquez, 2023-07-27:
Hi! Do you mind sharing where did you find those instructions and which project are you trying to build? I would suggest [following these instructions](https://esp-rs.github.io/book/installation/index.html#risc-v-and-xtensa-targets) and then [generating a project from a template](https://esp-rs.github.io/book/writing-your-own-application/generate-project/index.html)

yashwanthsinghm, 2023-07-27:
Hi, i got my instructions from here https://github.com/esp-rs/espup and i'm trying to compile this project in mac : https://github.com/esp-rs/esp-hal/tree/main/esp32-hal.

SergioGasquez, 2023-07-27:
Try following my suggestions and have a look at https://esp-rs.github.io/book/misc/troubleshooting.html#using-a-wrong-rust-toolchain

yashwanthsinghm, 2023-07-27:
Thank you i will follow this and see.

SergioGasquez, 2023-07-27:
Let me know how it goes!

yashwanthsinghm, 2023-07-27:
I have added $HOME/export-esp.sh into .bashrc and .profile & .bashrc and sourced them, the restarted the terminal. Still i'm getting the same error this issue i'm facing only in mac, linux its working fine in was able to build the project etc.

I have attached my .profile 

[image]

Also i'm getting these error too.

error: Package `test v0.0.0 (/Users/yashwanthsingh/.rustup/toolchains/esp/lib/rustlib/src/rust/library/test)` does not have the feature `backtrace`

SergioGasquez, 2023-07-27:
It looks like it's something wrong in your Mac environment. Can you share the exact commands that I would need to do to reproduce this?

yashwanthsinghm, 2023-07-27:
cat $HOME/export-esp.sh >> ~/.profile
 vim .profile
 source ~/.profile

same goes for bashrc too

SergioGasquez, 2023-07-31:
> cat $HOME/export-esp.sh >> ~/.profile vim .profile source ~/.profile
> 
> same goes for bashrc too

Please, share all the instructions, including how do you install the environment, and how do you build the project that you are trying to build.

Have you tried overriding the toolchain (`cargo +esp build`)?

SergioGasquez, 2023-10-19:
Hi, I will be closing the issue as it's inactive, feel free to reopen if the issue persists

closed 2023-10-19 by SergioGasquez

Bulalu, 2024-02-29:
i'm getting the same error and I'm also on mac

SergioGasquez, 2024-03-04:
> Please, share all the instructions, including how do you install the environment, and how do you build the project that you are trying to build.

Looks like your [not using the proper toolchain](https://esp-rs.github.io/book/troubleshooting/index.html#using-the-wrong-rust-toolchain). Please, share all the instructions, including how do you install the environment, and how do you build the project that you are trying to build.

Bulalu, 2024-03-04:
suree
here's how i install the environment
`cargo install espup`
`espup install`
`cargo install ldproxy`
`. $HOME/export-esp.sh`

`cargo install cargo-generate`

then i create a project from the templates
`cargo generate esp-rs/esp-template`

i choose the esp32 one, then on doing `cargo-build` i get this error

[image]

SergioGasquez, 2024-03-04:
What happens if you do: `cargo +esp build`?

Bulalu, 2024-03-04:
i get this
[image]

Bulalu, 2024-03-04:
[image]

SergioGasquez, 2024-03-04:
Please, run the following: `rustc +esp --print target-list | grep xtensa` and also, are you sourcing the `. $HOME/export-esp.sh` file in this terminal?

Bulalu, 2024-03-04:
yes i am 
[image]

Bulalu, 2024-03-04:
but with with this i seem to be getting the output
[image]

Bulalu, 2024-03-04:
ahh yeah found the little devil, it was indeed how i installed rust on my machine, i used homebrew before so i had to uninstall that and install from the rust site and seemed to solve everything.

thank you so much man 🫡