esp-rs/rust-build #229: Docker image: `cargo-espflash: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found `

systec-ms, 2023-08-08:
Good day,

i am running into the following errors when i use `espflash` commands inside the provided docker images (`espressif/idf-rust:esp32_1.71.0.1` and `espressif/idf-rust:esp32_latest`):

```bash
$ cargo espflash -h
/home/esp/.cargo/bin/cargo-espflash: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/esp/.cargo/bin/cargo-espflash)
/home/esp/.cargo/bin/cargo-espflash: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/esp/.cargo/bin/cargo-espflash)
/home/esp/.cargo/bin/cargo-espflash: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/esp/.cargo/bin/cargo-espflash)
```

Btw. is it correct that it is required to source the export-esp.sh when using the docker images (`source /home/esp/export-esp.sh`)?

systec-ms, 2023-08-09:
Thanks for the quick response, i just want to use `espflash save-image`  & `espflash partition-table --to-binary` and safe the output as an artifact in my CI.

SergioGasquez, 2023-08-09:
> Thanks for the quick response, i just want to use `espflash save-image` & `espflash partition-table --to-binary` and safe the output as an artifact in my CI.

Great, let me know if the solutions on the espflash issue work for you as a temporary workaround!

georgik, 2023-08-23:
@SergioGasquez  The issue is still present in espressif/idf-rust:all_1.71.0.1
It's not possible to run: 
cargo espflash save-image --chip ${CHIP} --release --merge spooky-maze-${FIRMWARE_DIR}.bin

Problem is that espflash is not built on the same base image is idf-rust. One is on Ubuntu 20 and one on Ubuntu 22.
Builds in CI fails:

/home/esp/.cargo/bin/cargo-espflash: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/esp/.cargo/bin/cargo-espflash)
/home/esp/.cargo/bin/cargo-espflash: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/esp/.cargo/bin/cargo-espflash)
/home/esp/.cargo/bin/cargo-espflash: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/esp/.cargo/bin/cargo-espflash)

SergioGasquez, 2023-09-14:
Just verified that the issue is resolved with #233:
```
git clone https://github.com/esp-rs/rust-build --branch features/update-debian
cd rust-build
docker image build --tag idf-test --file idf-rust.Containerfile --build-arg ESP_BOARD=esp32c3 --build-arg XTENSA_VERSION=1.72.0.0 .
docker run --user esp -it --mount type=bind,source="$(pwd)/build-template-project.sh",target=/workspace/build-template-project.sh,consistency=cached --rm idf-test /bin/bash
# Use any cargo-espflash command inside the container:
cargo espflash --help
```
Once the PR is merged, I will publish new tags to Dockerhub with the new Debian variant and solving the issue

closed 2023-09-14 by SergioGasquez