esp-rs/rust-build #246: `cargo build` fails on arm container on M1 Mac

noahbliss, 2023-10-22:
Hey all. I have an M1 Mac running a template build directory. When using the latest (the only version I have tried) version of the docker on dockerhub, the initial `cargo build` fails with the following error. Any ideas? Any details I need to provide in addition? 

Thanks!

```
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.1.1") }
  Using esp-idf v5.1.1 at '/mnt/hp-srid-1-firmware/.embuild/espressif/esp-idf/v5.1.1'
  Compiler supported targets: xtensa-esp32s3-elf

  gmake[2]: Warning: File 'esp-idf/esp_https_ota/CMakeFiles/__idf_esp_https_ota.dir/compiler_depend.make' has modification time 0.38 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/esp_http_server/CMakeFiles/__idf_esp_http_server.dir/compiler_depend.make' has modification time 0.32 s in the future
  Compiler supported targets: xtensa-esp32s3-elf

  gmake[4]: Warning: File 'CMakeFiles/Makefile2' has modification time 0.48 s in the future
  gmake[5]: Warning: File 'esp-idf/log/CMakeFiles/__idf_log.dir/compiler_depend.make' has modification time 0.34 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/esp_common/CMakeFiles/__idf_esp_common.dir/compiler_depend.make' has modification time 0.73 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/esp_http_client/CMakeFiles/__idf_esp_http_client.dir/compiler_depend.make' has modification time 0.0056 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/compiler_depend.make' has modification time 0.56 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/esp_system/CMakeFiles/__idf_esp_system.dir/compiler_depend.make' has modification time 0.29 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/efuse/CMakeFiles/__idf_efuse.dir/compiler_depend.make' has modification time 0.33 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/tcp_transport/CMakeFiles/__idf_tcp_transport.dir/compiler_depend.make' has modification time 0.43 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/compiler_depend.make' has modification time 0.4 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/esp_gdbstub/CMakeFiles/__idf_esp_gdbstub.dir/compiler_depend.make' has modification time 0.085 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/esp_app_format/CMakeFiles/__idf_esp_app_format.dir/compiler_depend.make' has modification time 0.56 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/compiler_depend.make' has modification time 0.66 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/hal/CMakeFiles/__idf_hal.dir/compiler_depend.make' has modification time 0.41 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/compiler_depend.make' has modification time 0.21 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/micro-ecc/CMakeFiles/__idf_micro-ecc.dir/compiler_depend.make' has modification time 0.67 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/soc/CMakeFiles/__idf_soc.dir/compiler_depend.make' has modification time 0.51 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/compiler_depend.make' has modification time 0.31 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'esp-idf/main/CMakeFiles/__idf_main.dir/compiler_depend.make' has modification time 0.38 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[5]: Warning: File 'CMakeFiles/bootloader.elf.dir/compiler_depend.make' has modification time 0.48 s in the future
  gmake[5]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[4]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/http_parser/CMakeFiles/__idf_http_parser.dir/compiler_depend.make' has modification time 0.55 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/esp_wifi/CMakeFiles/__idf_esp_wifi.dir/compiler_depend.make' has modification time 0.0038 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/esp_coex/CMakeFiles/__idf_esp_coex.dir/compiler_depend.make' has modification time 0.54 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/esp_phy/CMakeFiles/__idf_esp_phy.dir/compiler_depend.make' has modification time 0.52 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/nvs_flash/CMakeFiles/__idf_nvs_flash.dir/compiler_depend.make' has modification time 0.32 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/esp_event/CMakeFiles/__idf_esp_event.dir/compiler_depend.make' has modification time 0.13 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/esp_common/CMakeFiles/__idf_esp_common.dir/compiler_depend.make' has modification time 0.44 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
  gmake[2]: Warning: File 'esp-idf/freertos/CMakeFiles/__idf_freertos.dir/compiler_depend.make' has modification time 0.37 s in the future
  gmake[2]: warning:  Clock skew detected.  Your build may be incomplete.
[... 240 lines cut ...]
  /home/esp/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: cannot find esp-idf/newlib/libnewlib.a: Operation not permitted
  /home/esp/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: cannot find esp-idf/pthread/libpthread.a: Operation not permitted
  /home/esp/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: cannot find esp-idf/cxx/libcxx.a: Operation not permitted
  /home/esp/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: cannot find -lphy: No such file or directory
  /home/esp/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: cannot find -lbtbb: No such file or directory
  /home/esp/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: cannot find esp-idf/esp_phy/libesp_phy.a: Operation not permitted
  /home/esp/.rustup/toolchains/esp/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: cannot find -lphy: No such file or directory
[... cut ...]
  gmake: *** [Makefile:136: all] Error 2
  thread 'main' panicked at /home/esp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 2

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
esp@eb3c2603458d:/mnt/code$ uname -a
Linux eb3c2603458d 6.1.29-0-virt #1-Alpine SMP Wed, 17 May 2023 14:22:15 +0000 aarch64 GNU/Linux
esp@eb3c2603458d:/mnt/code$
```

noahbliss, 2023-10-23:
Hey @SergioGasquez I don't have all of the exact commands, but what you assume I did is more or less on-the-money. 

System context: 
M1 Mac
colima with docker tools installed via brew. 
custom docker invocation script, uses a shared volume with the `esp-idf-template` (for std) as you mentioned.

Nothing else really special. I start the container, cd to the directory, run `cargo build` and get this error.

SergioGasquez, 2023-10-24:
Just did a small test on my M1 mac with docker and had no issue, here is what I did:
```
$ docker pull espressif/idf-rust:esp32s3_latest
$ docker run -it --rm -u esp espressif/idf-rust:esp32s3_latest
docker$  cargo install cargo-generate
docker$ USER=esp cargo generate esp-rs/esp-idf-template cargo --name test -d mcu=esp32s3 -d advanced=false
docker$ cd test
docker$ cargo build
  ....
  Finished dev [optimized + debuginfo] target(s) in 2m 57s
```

noahbliss, 2023-10-24:
@SergioGasquez thanks for testing this. I'll repeat your steps, if I also have no issues, then I suspect this may be stemming from the volume mount and will investigate that deeper. Thanks!

noahbliss, 2023-10-26:
Is there a way to change the path where the .cargo and .embuild directories are created?

SergioGasquez, 2023-10-27:
There are some options, here is the documentation for [`.cargo`](https://doc.rust-lang.org/cargo/reference/config.html#config-relative-paths) and [`.embuild`](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp_idf_tools_install_dir-esp_idf_tools_install_dir)

noahbliss, 2023-10-31:
Hmmm it thickens. 

I attempted to effectively "move" .embuild into the container fs by running
`export ESP_IDF_TOOLS_INSTALL_DIR=global`
prior to running `cargo build`. 

It seems that this _does_ install the IDF tools to the container's home directory, but later in the build process, I still receive an error which points to a path inside the workspace directory which now no longer exists rather than to the file which is present in the ~ directory. Any thoughts?

noahbliss, 2023-10-31:
This is the snippet including the error now: 

```
[esp-idf-sys 0.33.3] CMAKE_xtensa_esp32s3_espidf = None
[esp-idf-sys 0.33.3] TARGET_CMAKE = None
[esp-idf-sys 0.33.3] CMAKE = None
[esp-idf-sys 0.33.3] running: cd "/mnt/code/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-67c3ebe566277a4d/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_COMPONENT_MANAGER="1" IDF_PATH="/home/esp/.espressif/esp-idf/v5.1.1" IDF_TARGET="esp32s3" IDF_TOOLS_PATH="/home/esp/.espressif" PATH="/home/esp/.espressif/tools/cmake/3.24.0/bin:/home/esp/.espressif/python_env/idf5.1_py3.11_env/bin:/home/esp/.espressif/esp-idf/v5.1.1/tools:/home/esp/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/home/esp/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/esp/.rustup/toolchains/esp/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/home/esp/.cargo/bin" PROJECT_DIR="/mnt/code" SDKCONFIG_DEFAULTS="/mnt/code/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-67c3ebe566277a4d/out/gen-sdkconfig.defaults;/mnt/code/sdkconfig.defaults" "cmake" "/mnt/code/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-67c3ebe566277a4d/out" "-G" "Unix Makefiles" "-DCMAKE_TOOLCHAIN_FILE=/home/esp/.espressif/esp-idf/v5.1.1/tools/cmake/toolchain-esp32s3.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/home/esp/.espressif/python_env/idf5.1_py3.11_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/mnt/code/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-67c3ebe566277a4d/out" "-DCMAKE_C_FLAGS= -mlongcalls -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
[esp-idf-sys 0.33.3] CMake Error at build/CMakeFiles/3.24.0/CMakeSystem.cmake:6 (include):
[esp-idf-sys 0.33.3]   include could not find requested file:
[esp-idf-sys 0.33.3]
[esp-idf-sys 0.33.3]     /mnt/code/.embuild/espressif/esp-idf/v5.1.1/tools/cmake/toolchain-esp32s3.cmake
[esp-idf-sys 0.33.3] Call Stack (most recent call first):
[esp-idf-sys 0.33.3]   /home/esp/.espressif/esp-idf/v5.1.1/tools/cmake/project.cmake:448 (__project)
[esp-idf-sys 0.33.3]   CMakeLists.txt:28 (project)
[esp-idf-sys 0.33.3]
[esp-idf-sys 0.33.3]
[esp-idf-sys 0.33.3] -- The C compiler identification is GNU 12.2.0
[esp-idf-sys 0.33.3] -- The CXX compiler identification is GNU 12.2.0
[esp-idf-sys 0.33.3] -- The ASM compiler identification is GNU
[esp-idf-sys 0.33.3] -- Found assembler: /usr/bin/cc
[esp-idf-sys 0.33.3] -- Detecting C compiler ABI info
[esp-idf-sys 0.33.3] CMake Error at /mnt/code/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-67c3ebe566277a4d/out/build/CMakeFiles/3.24.0/CMakeSystem.cmake:6 (include):
[esp-idf-sys 0.33.3]   include could not find requested file:
[esp-idf-sys 0.33.3]
[esp-idf-sys 0.33.3]     /mnt/code/.embuild/espressif/esp-idf/v5.1.1/tools/cmake/toolchain-esp32s3.cmake
[esp-idf-sys 0.33.3] Call Stack (most recent call first):
[esp-idf-sys 0.33.3]   /mnt/code/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-67c3ebe566277a4d/out/build/CMakeFiles/CMakeTmp/CMakeLists.txt:4 (project)
[esp-idf-sys 0.33.3]
[esp-idf-sys 0.33.3]
[esp-idf-sys 0.33.3] CMake Error at /home/esp/.espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/CMakeDetermineCompilerABI.cmake:57 (try_compile):
[esp-idf-sys 0.33.3]   Failed to configure test project build system.
[esp-idf-sys 0.33.3] Call Stack (most recent call first):
[esp-idf-sys 0.33.3]   /home/esp/.espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/CMakeTestCCompiler.cmake:26 (CMAKE_DETERMINE_COMPILER_ABI)
[esp-idf-sys 0.33.3]   /home/esp/.espressif/esp-idf/v5.1.1/tools/cmake/project.cmake:448 (__project)
[esp-idf-sys 0.33.3]   CMakeLists.txt:28 (project)
[esp-idf-sys 0.33.3]
[esp-idf-sys 0.33.3]
[esp-idf-sys 0.33.3] -- Configuring incomplete, errors occurred!
[esp-idf-sys 0.33.3] See also "/mnt/code/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-67c3ebe566277a4d/out/build/CMakeFiles/CMakeOutput.log".
[esp-idf-sys 0.33.3] See also "/mnt/code/target/xtensa-esp32s3-espidf/debug/build/esp-idf-sys-67c3ebe566277a4d/out/build/CMakeFiles/CMakeError.log".
[esp-idf-sys 0.33.3] thread 'main' panicked at /home/esp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:
[esp-idf-sys 0.33.3]
[esp-idf-sys 0.33.3] command did not execute successfully, got: exit status: 1
[esp-idf-sys 0.33.3]
[esp-idf-sys 0.33.3] build script failed, must exit now
[esp-idf-sys 0.33.3] note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
error: failed to run custom build command for `esp-idf-sys v0.33.3`

Caused by:
  process didn't exit successfully: `/mnt/code/target/debug/build/esp-idf-sys-7a89f24be9931daf/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
[... cut ...]
    /home/esp/.espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/CMakeTestCCompiler.cmake:26 (CMAKE_DETERMINE_COMPILER_ABI)
    /home/esp/.espressif/esp-idf/v5.1.1/tools/cmake/project.cmake:448 (__project)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /home/esp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
esp@678ab5bdf6f9:/mnt/code$
```

noahbliss, 2023-10-31:
Update: Issue appears to be resolved. 

I found this issue comment (https://github.com/abiosoft/colima/issues/362#issuecomment-1564068063) which hinted that telling colima to use mac vz instead of qemu would help. I switched and now the build is completing and with far fewer errors. I did not switch to virtiofs as the comment suggests. 

Thanks all, I'll close this now.

closed 2023-10-31 by noahbliss