esp-rs/rust-build #252: Toolchain installed by espup does not work under Ubuntu 20.04, GLIBC version error

jobafr, 2024-01-22:
## Bug description

After using `espup install` under Ubuntu 20.04 (aarch64, NVidia Jetson AGX Xavier), running the cargo binary produces a linker error.

```
$ ~/.rustup/toolchains/esp/bin/cargo
/home/me/.rustup/toolchains/esp/bin/cargo: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/me/.rustup/toolchains/esp/bin/cargo)
/home/me/.rustup/toolchains/esp/bin/cargo: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/me/.rustup/toolchains/esp/bin/cargo)
/home/me/.rustup/toolchains/esp/bin/cargo: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/me/.rustup/toolchains/esp/bin/cargo)
```

- Would you like to work on a fix? [y/n]

n

## To Reproduce

Steps to reproduce the behavior:

1. run `espup install` on Ubuntu 20.04 aarch64
2. try to run cargo

## Expected behavior

Cargo works and compiles my project

## Environment

```
$ cat /etc/issue
Ubuntu 20.04.6 LTS \n \l

$ uname -a
Linux aql57 5.10.104-tegra esp-rs/espup#1 SMP PREEMPT Wed Aug 10 20:17:07 PDT 2022 aarch64 aarch64 aarch64 GNU/Linux

$ espup --version
espup 0.10.0
```

## Additional context

This looks similar to [this issue in espflash](https://github.com/esp-rs/espflash/issues/454) from last year.

I tried to provide the cargo version as well, but `/home/me/.rustup/toolchains/esp/bin/cargo --version` does of course fail with the same linker error.

jobafr, 2024-01-22:
Hi, thank you for the quick response! As a quick fix, is it possible to intentionally install an older rust release that's compatible with 20.04 using espup?

SergioGasquez, 2024-01-22:
> Hi, thank you for the quick response! As a quick fix, is it possible to intentionally install an older rust release that's compatible with 20.04 using espup?

Yes, using `espup install -v 1.74.0.1` should work

georgik, 2024-02-12:
Marked as Third Party limitation, since Ubuntu 20 does not support build of latest toolchains. 
@SergioGasquez : Would it be possible to add warning to espup, that distribution is not supported?

If somebody finds a solution how to handle such build, we might consider adding it. Otherwise, please upgrade to newer Ubuntu or use container.

jesper-adolfsson-001, 2024-08-08:
If you do `espup install -v 1.74.0.1` then the default esp-idf template doesn't compile.... Is there no way around this?

Vollbrecht, 2024-08-08:
> If you do `espup install -v 1.74.0.1` then the default esp-idf template doesn't compile.... Is there no way around this?

how is this related to this issue? Please don't hijack issues if you have a new issue and if so please give a full error report!

jesper-adolfsson-001, 2024-08-08:
> > If you do `espup install -v 1.74.0.1` then the default esp-idf template doesn't compile.... Is there no way around this?
> 
> how is this related to this issue? Please don't hijack issues if you have a new issue and if so please give a full error report!

Sorry for breaking some fancy hijacking etiquette here... I was just trying to help beginners like me who has pent 3 days now trying to get this to work with some additional info. But as usual some technical guru like you has to make me feel like inferior....

SergioGasquez, 2024-08-08:
I assume MSRV of the `esp-idf-template` is greater than 1.74.0.1, which is the latest version using the GLIBC that Ubuntu 20.04 support. 

Is there any reason not to update to a recent Ubuntu version?

jesper-adolfsson-001, 2024-08-08:
> I assume MSRV of the `esp-idf-template` is greater than 1.74.0.1, which is the latest version using the GLIBC that Ubuntu 20.04 support.
> 
> Is there any reason not to update to a recent Ubuntu version?

Thanks for answering and I apologize in advance if I provide non-relevant info now and ask stupid questions but I'm a bit new to this.

I installed WSL Ubuntu in Windows today and it says it is version 20.04 "Focal" which I now understand is an old version. I will try to upgrade it and see what happens.

SergioGasquez, 2024-08-08:
You don't need to apologize, or be defensive about our comments. Both, @Vollbrecht and I were trying to help you, also for questions I would recommend using our matrix channel https://matrix.to/#/#esp-rs:matrix.org

Vollbrecht, 2024-08-08:
i am not using windows that much these days though you can get ubuntu 22 directly for wsl i think if you follow for exmaple [this](https://www.microsoft.com/store/apps/9PN20MSR04DW)

jesper-adolfsson-001, 2024-08-08:
> You don't need to apologize, or be defensive about our comments. Both, @Vollbrecht and I were trying to help you, also for questions I would recommend using our matrix channel https://matrix.to/#/#esp-rs:matrix.org

I appreciate that, I really do :)

I did what you recommended and installed WSL Ubuntu 24.04 and it sort of worked. Still have some issues that I don't understand but I will reach out in the matrix.org chat if I need.

closed 2024-09-17 by MabezDev