esp-rs/rust-build #253: error[E0063]: missing field `stack_alloc_caps` in initializer of `esp_idf_sys::esp_pthread_cfg_t

dmitryr117, 2024-02-08:
When I try to run `cargo build` after setting up environment - I get:

```
   Compiling exp-01 v0.1.0 (/home/dmitryr117/dev/iot/esp32s3/exp-01)
   Compiling embassy-futures v0.1.1
   Compiling embedded-svc v0.27.0
error[E0063]: missing field `stack_alloc_caps` in initializer of `esp_idf_sys::esp_pthread_cfg_t`
   --> /home/dmitryr117/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.43.0/src/task.rs:357:13
    |
357 |             Self {
    |             ^^^^ missing `stack_alloc_caps`

For more information about this error, try `rustc --explain E0063`.
error: could not compile `esp-idf-hal` (lib) due to previous error
warning: build failed, waiting for other jobs to finish...

```

My `Cargo.toml` is as follows:

```
[package]
name = "exp-01"
version = "0.1.0"
authors = ["Dmitry <colirs.developer@gmail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-svc/embassy-time-driver"]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.48", default-features = false }

[build-dependencies]
embuild = "0.31.3"

```

Don't know what is causing the problem.

EthnTuttle, 2024-02-22:
I am experiencing the same issue but it is not resolved after reboot

dmitryr117, 2024-02-23:
I'll reopen the issue. But I think it has something to do with environment variables set. Will probably have to do a more detailed investigation why this error occurs.

EthnTuttle, 2024-02-23:
I believe it does have to do with env vars. I was able to get past it by restarting the Dev env setup. I am unsure at what point it was resolved though.

-------- Original Message --------

aklitzke, 2024-03-11:
Running into this issue also. This is with a fresh sample project following https://github.com/esp-rs/esp-idf-template/blob/master/README-cmake.md#prerequisites

using both 'std' and 'hal'.

Steps to reproduce:
1. start fresh dev env
2. run `get_idf`
3. run `idf.py set-target esp32`
4. run `idf.py build`

```
(base) ➜  is-cat-fed git:(master) ✗ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /Users/andrewklitzke/esp/is-cat-fed/build
Executing "ninja all"...
[0/15] Performing build step for 'project_rust_is_cat_fed'   Compiling esp-idf-hal v0.43.1
error[E0063]: missing field `stack_alloc_caps` in initializer of `esp_idf_sys::esp_pthread_cfg_t`
   --> /Users/andrewklitzke/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-hal-0.43.1/src/task.rs:357:13
    |
357 |             Self {
    |             ^^^^ missing `stack_alloc_caps`

For more information about this error, try `rustc --explain E0063`.
error: could not compile `esp-idf-hal` (lib) due to 1 previous error
[1/1] cd /Users/andrewklitzke/esp/is-cat-fed/build/bootloader/esp-idf/esptool_py && /Users/andrewklitzke/....py --offset 0x8000 bootloader 0x1000 /Users/andrewklitzke/esp/is-cat-fed/build/bootloader/bootloader.bin
Bootloader binary size 0x6870 bytes. 0x790 bytes (7%) free.
[3/15] Completed 'bootloader'
FAILED: esp-idf/rust-is-cat-fed/stamp/project_rust_is_cat_fed-build esp-idf/rust-is-cat-fed/target/xtensa-esp32-espidf/release/librust_is_cat_fed.a /Users/andrewklitzke/esp/is-cat-fed/build/esp-idf/rust-is-cat-fed/stamp/project_rust_is_cat_fed-build /Users/andrewklitzke/esp/is-cat-fed/build/esp-idf/rust-is-cat-fed/target/xtensa-esp32-espidf/release/librust_is_cat_fed.a
cd /Users/andrewklitzke/esp/is-cat-fed/components/rust-is-cat-fed && /opt/homebrew/Cellar/cmake/3.28.3/bin/cmake -E env "CARGO_CMAKE_BUILD_INCLUDES=/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/is-cat-fed/build/esp-idf/rust-is-cat-fed/target;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/newlib/platform_include;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/freertos/config/include;/Users/andrewklitzke/esp/esp-idf/components/freertos/config/include/freertos;/Users/andrewklitzke/esp/esp-idf/components/freertos/config/xtensa/include;/Users/andrewklitzke/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include;/Users/andrewklitzke/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include;/Users/andrewklitzke/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos;/Users/andrewklitzke/esp/esp-idf/components/freertos/esp_additions/include;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/esp_hw_support/include;/Users/andrewklitzke/esp/esp-idf/components/esp_hw_support/include/soc;/Users/andrewklitzke/esp/esp-idf/components/esp_hw_support/include/soc/esp32;/Users/andrewklitzke/esp/esp-idf/components/esp_hw_support/dma/include;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/esp_hw_support/port/esp32/.;/Users/andrewklitzke/esp/esp-idf/components/esp_hw_support/port/esp32/include;/Users/andrewklitzke/esp/esp-idf/components/heap/include;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/log/include;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/soc/include;/Users/andrewklitzke/esp/esp-idf/components/soc/esp32;/Users/andrewklitzke/esp/esp-idf/components/soc/esp32/include;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/hal/platform_port/include;/Users/andrewklitzke/esp/esp-idf/components/hal/esp32/include;/Users/andrewklitzke/esp/esp-idf/components/hal/include;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/esp_rom/include;/Users/andrewklitzke/esp/esp-idf/components/esp_rom/include/esp32;/Users/andrewklitzke/esp/esp-idf/components/esp_rom/esp32;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/esp_common/include;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/esp_system/include;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/esp_system/port/soc;/Users/andrewklitzke/esp/esp-idf/components/esp_system/port/include/private;/Users/andrewklitzke/esp/esp-idf/components/xtensa/esp32/include;/Users/andrewklitzke/esp/esp-idf/components/xtensa/include;/Users/andrewklitzke/esp/esp-idf/components/xtensa/deprecated_include;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/lwip/include;/Users/andrewklitzke/esp/esp-idf/components/lwip/include/apps;/Users/andrewklitzke/esp/esp-idf/components/lwip/include/apps/sntp;/Users/andrewklitzke/esp/esp-idf/components/lwip/lwip/src/include;/Users/andrewklitzke/esp/esp-idf/components/lwip/port/include;/Users/andrewklitzke/esp/esp-idf/components/lwip/port/freertos/include;/Users/andrewklitzke/esp/esp-idf/components/lwip/port/esp32xx/include;/Users/andrewklitzke/esp/esp-idf/components/lwip/port/esp32xx/include/arch;/Users/andrewklitzke/esp/esp-idf/components/lwip/port/esp32xx/include/sys;/Users/andrewklitzke/esp/is-cat-fed/build/config;/Users/andrewklitzke/esp/esp-idf/components/pthread/incl [... cut ...]
OPENOCD_SCRIPTS=/Users/andrewklitzke/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
ESP_ROM_ELF_DIR=/Users/andrewklitzke/.espressif/tools/esp-rom-elfs/20240305/
IDF_PYTHON_ENV_PATH=/Users/andrewklitzke/.espressif/python_env/idf5.3_py3.9_env
ESP_IDF_VERSION=5.3
IDF_DEACTIVATE_FILE_PATH=/var/folders/s2/kx0cfh5n54v25c8xprdtmng00000gn/T/tmpqyznp1pvidf_16887
IDF_TOOLS_EXPORT_CMD=/Users/andrewklitzke/esp/esp-idf/export.sh
IDF_TOOLS_INSTALL_CMD=/Users/andrewklitzke/esp/esp-idf/install.sh
_=/usr/bin/env
```

using an m1 mac, cargo and rustc 1.75.0