esp-rs/rust-build #256: Error building projects on MacOS 10.15 - Catalina

DanXS, 2024-02-23:
I'm having trouble running it on an older Mac which can't be upgraded beyond MacOS 10.15 - Catalina, it's an x86 based macbook pro.  I think it could be because of an unsupported library. When building generated template with cargo build i get:

```
error: process didn't exit successfully: `/Users/danshepherd/.rustup/toolchains/esp/bin/rustc -vV` (signal: 6, SIGABRT: process abort signal)
--- stderr
dyld: Symbol not found: __ZNKSt3__115basic_stringbufIcNS_11char_traitsIcEENS_9allocatorIcEEE3strEv
  Referenced from: /Users/danshepherd/.rustup/toolchains/esp/bin/../lib/librustc_driver-0d65bcf01718bf2f.dylib (which was built for Mac OS X 12.0)
  Expected in: /usr/lib/libc++.1.dylib
```

SergioGasquez, 2024-03-04:
Mind expanding the issue with details? How did you install the toolchain? What are you trying to execute...?

DanXS, 2024-03-05:
Unfortunately Apple are a bit vender lock-in, so they don’t update operating systems that aren’t on latest hardware, for example I can’t build iOS apps without a newer Mac. It could be it’s not an issue other than people who don’t have latest hardware won’t be able to use ESP with rust

SergioGasquez, 2024-03-05:
> Yes it was just the normal process not from source but it says the library was built for macOS version 12

I'm still not sure of what exactly you were executing? `espup install`? Looks like the issue is what you described, your hardware is too old and not supported anymore, but let's confirm this.