esp-rs/rust-build #265: Status of `1.78.0.0`?

dimpolo, 2024-05-08:
Don't mean to hurry anyone, just wondering, what is the status of a `1.78.0.0` release?

SergioGasquez, 2024-05-08:
We are having some issues building Windows artifacts, thats why it is still as prerelease (https://github.com/esp-rs/rust-build/releases/tag/v1.78.0.0). If you are not in Windows, you can `espup update -v 1.78.0.0` and update your toolchain

georgik, 2024-05-10:
Windows MSVC has problem with mixing /MT and /MD when building LLD, still not solved.

Windows GNU is available, problem was with Defender shutting down the build, when job was started under CI user.

SergioGasquez, 2024-07-05:
Update on the `1.78.0.0` and `1.79.0.0` releases:
- `1.78.0.0` won't be promoted as `latest` and will be skipped
- `1.79.0.0` is ready and safe to use for all the targets but `x86_64-pc-windows-msvc`. We will try to build this target artifacts, so the release can be marked as `latest`

Also, upcoming releases should not have this issue anymore as it was solved in https://github.com/rust-lang/rust/pull/126701

MabezDev, 2024-07-08:
Sorry about the delay, MSVC toolchain has been a bit troublesome in the last few releases. I've just promoted the 1.79.0.0 to the latest release.

closed 2024-07-08 by MabezDev