esp-rs/rust-build #270: Permission Denied from rust-analyzer

DaneSlattery, 2024-07-29:
Rust analyzer cannot write to the `target` directory.

I am running docker on Ubuntu 24. 

When using this image `idf-rust:esp32_latest`, with the dev container definition from the [template](https://github.com/esp-rs/esp-idf-template/blob/master/cargo/.devcontainer/devcontainer.json), I see the following error from `rust-analyzer`

```
error: failed to create directory `/home/esp/ppm/target/xtensa-esp32-espidf/debug/.fingerprint/ppm-17e5cca47ed3fd90`

Caused by:
  Permission denied (os error 13)
```

From the command line on the container, the workspace is mounted as root.:
```
esp@9927be9ce188:~/ppm$ ls -l
total 132
-rw-rw-r-- 1 root root 64347 Jul 26 12:56 Cargo.lock
-rw-rw-r-- 1 root root   323 Jul 26 09:32 Cargo.toml
-rw-rw-r-- 1 root root  3223 Jul 26 09:32 README.md
-rw-rw-r-- 1 root root  1568 Jul 26 09:32 azure-pipelines.yml
-rw-rw-r-- 1 root root   413 Jul 26 10:33 components_esp32.lock
drwxrwxr-x 2 root root  4096 Jul 26 09:32 docs
-rw-rw-r-- 1 root root  3072 Jul 26 09:32 partitions.bin
drwxrwxr-x 3 root root  4096 Jul 26 10:33 ppm
drwxrwxr-x 3 root root  4096 Jul 26 10:33 ppm-arduino
drwxrwxr-x 3 root root  4096 Jul 26 13:10 ppm-esp32
drwxrwxr-x 4 root root  4096 Jul 26 09:32 ppm-web
-rw-rw-r-- 1 root root   101 Jul 26 10:32 rust-toolchain.toml
drwxrwxr-x 5 root root  4096 Jul 26 10:33 rust-web
drwxrwxr-x 2 root root  4096 Jul 26 09:32 scripts
-rw-rw-r-- 1 root root   923 Jul 26 12:55 sdkconfig.defaults
drwxrwxr-x 4 root root  4096 Jul 26 10:33 stpm3x
drwxrwxr-x 5 root root  4096 Jul 26 12:56 target
drwxrwxr-x 4 root root  4096 Jul 26 09:32 win_stpm3x
```

, while locally, the files are owned by my user:
```
dane@dane-Latitude-5531  ~/dev/PolarPowerMeter   website ±  ls -l
total 132
-rw-rw-r-- 1 dane dane  1568 Jul 26 11:32 azure-pipelines.yml
-rw-rw-r-- 1 dane dane 64347 Jul 26 14:56 Cargo.lock
-rw-rw-r-- 1 dane dane   323 Jul 26 11:32 Cargo.toml
-rw-rw-r-- 1 dane dane   413 Jul 26 12:33 components_esp32.lock
drwxrwxr-x 2 dane dane  4096 Jul 26 11:32 docs
-rw-rw-r-- 1 dane dane  3072 Jul 26 11:32 partitions.bin
drwxrwxr-x 3 dane dane  4096 Jul 26 12:33 ppm
drwxrwxr-x 3 dane dane  4096 Jul 26 12:33 ppm-arduino
drwxrwxr-x 3 dane dane  4096 Jul 26 15:10 ppm-esp32
drwxrwxr-x 4 dane dane  4096 Jul 26 11:32 ppm-web
-rw-rw-r-- 1 dane dane  3223 Jul 26 11:32 README.md
-rw-rw-r-- 1 dane dane   101 Jul 26 12:32 rust-toolchain.toml
drwxrwxr-x 5 dane dane  4096 Jul 26 12:33 rust-web
drwxrwxr-x 2 dane dane  4096 Jul 26 11:32 scripts
-rw-rw-r-- 1 dane dane   923 Jul 26 14:55 sdkconfig.defaults
drwxrwxr-x 4 dane dane  4096 Jul 26 12:33 stpm3x
drwxrwxr-x 5 dane dane  4096 Jul 26 14:56 target
drwxrwxr-x 4 dane dane  4096 Jul 26 11:32 win_stpm3x
```

The docker user has the following id:
```
uid=1000(esp) gid=1000(esp) groups=1000(esp)
```

My local user has the following id:
```
uid=1000(dane) gid=1000(dane) groups=1000(dane),4(adm),20(dialout),24(cdrom),27(sudo),30(dip),46(plugdev),100(users),114(lpadmin),984(docker)
```

I have tried:
1. Setting `"remoteUser":"esp"` in `devcontainer.json`.
2. Adding this to the Dockerfile after `adduser --disabled-password --gecos "" ${CONTAINER_USER}`:
```
https://code.visualstudio.com/remote/advancedcontainers/add-nonroot-user#_change-the-uidgid-of-an-existing-container-user
```

What seems to have worked is to change the ownership of the mounted files on the container:
```
chown -R esp .
```

I'm wondering if this is something that we can address.

DaneSlattery, 2024-07-29:
Further investigation to this: changing the ownership of the mounted files prevents files from being used by the local user. For example, the built artifacts cannot be deployed using `web-flash`.

DaneSlattery, 2024-07-31:
Even more days down this rabbit hole:
Docker Desktop mounts this with `root:root` ownership. Docker `CE` mounts with the correct ownership!

MabezDev, 2024-10-25:
@SergioGasquez @georgik are these docker images still maintained?

SergioGasquez, 2024-10-25:
> are these docker images still maintained?

Yes, somewhat. I forgot to publish the new [tags](https://hub.docker.com/r/espressif/idf-rust/tags) for 1.81.0.0, but just launched the action now, also I should run it for 1.82.0.0. Currently, the main reason to keep maintaining this tags is that they are used as base images for the wokwi builder images. Maybe we can discuss if we want to keep it this way.

Regarding the issue @DaneSlattery, it comes from mixing the published tags with the devcontainer settings in esp-idf-template. Is there any reason to use the Dockerfile that lives in esp-idf-template?

DaneSlattery, 2024-10-25:
I personally use the Dockerfile in `esp-idf-template` to load additional tools to our developer environment, such as more crates, python etc etc, but this particular issue relates to the docker image from this repo.