esp-rs/rust-build #272: Breakage when building `core`.

shymega, 2024-08-20:
Hi,

I'm creating a Nix Flake for the Rust toolchains for the ESP32.

Whilst `rustc` is working, I get quite a lot of low-level errors when building a simple example from the template. Attached is a log.

I've uploaded the example [here][esp_example], and the Flake can be found [here][esp_flake].

Any ideas?

[esp_example]: https://github.com/shymega/esp-idf-nix-test
[esp_flake]: https://github.com/shymega/esp32-dev.nix
[cargo-build-logs.txt](https://github.com/user-attachments/files/16677450/cargo-build-logs.txt)

MabezDev, 2024-08-21:
I don't know anything about Nix, so I can't really help here. I would recommend making sure you can build a project for your host OS, and then moving on to try and cross compile for esp32.

shymega, 2024-08-21:
Yeah, I'm not certain if it's a Nix issue or not. Is it normal for `std` and `core` to need building?

I'm going to try setting `RUST_SRC_PATH`.

shymega, 2024-08-21:
OK. Well, looking at the `RUST_SRC_PATH` env set, it's not valid.

shymega, 2024-08-21:
By the looks of things, my derivation is a bit wonky. I think we should keep this issue open for now, though.

My end-goal is to add credits and Actions workflows to my Flake,  and add details to the `esp-rs` wiki.

Shall we discuss on the Matrix room later on?