esp-rs/rust-build #275: 1.82 missing Cargo.lock in rust-src

jothan, 2024-10-23:
cargo build errors out on the current 1.82.0.0 pre-release on Xtensa:

```
error: "/home/user/.rustup/toolchains/esp/lib/rustlib/src/rust/Cargo.lock" does not exist, unable to build with the standard library, try:
        rustup component add rust-src --toolchain esp
```

This Cargo.lock is present in rust-src-1.81.0.0.tar.xz, but not in rust-src-1.82.0.0.tar.xz.

MabezDev, 2024-10-24:
Not sure what's going on here, I opened a thread in the zulip instance: https://rust-lang.zulipchat.com/#narrow/channel/122651-general/topic/rust.201.2E82.20does.20not.20install.20rust-src. Even when rust-src is installed, it places the `Cargo.lock` in the wrong place.

MabezDev, 2024-10-25:
This should be resolved, could you try and install the 1.82 pre release again?

SergioGasquez, 2024-10-25:
> This should be resolved, could you try and install the 1.82 pre release again?

FWIW, just tried on my machine, which also experienced this issue yesterday, and it's now working fine! Thanks for the fix!

closed 2024-10-25 by MabezDev