esp-rs/rust-build #282: LoadLibraryExW failes on libclang.dll

FssAy, 2024-11-29:
When trying to run `cargo build` on Windows 10 after performing every step from the [book](https://docs.esp-rs.org/book/installation/riscv-and-xtensa.html) I am getting an error:
```
Unable to find libclang: "the `libclang` shared library at C:\\Users\\Dmitrij\\.rustup\\toolchains\\esp\\xtensa-esp32-elf-clang\\esp-clang\\bin\\libclang.dll could not be opened: LoadLibraryExW failed"
```
The path to the file is correct, the file exists and is not corrupted.

Here is the full log:
```
D:\vats>cargo build
   Compiling bindgen v0.69.5
   Compiling embuild v0.32.0
   Compiling esp-idf-sys v0.35.0
   Compiling esp-idf-hal v0.44.1
   Compiling esp-idf-svc v0.49.1
   Compiling vats v0.1.0 (D:\vats)
The following warnings were emitted during compilation:

warning: esp-idf-sys@0.35.0: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `cc` installed? (see https://docs.rs/cc/latest/cc/#compile-time-requirements for help)
warning: esp-idf-sys@0.35.0: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `c++` installed? (see https://docs.rs/cc/latest/cc/#compile-time-requirements for help)
warning: esp-idf-sys@0.35.0: Compiler family detection failed due to error: ToolNotFound: Failed to find tool. Is `cc` installed? (see https://docs.rs/cc/latest/cc/#compile-time-requirements for help)

error: failed to run custom build command for `esp-idf-sys v0.35.0`

Caused by:
  process didn't exit successfully: `D:\vats\target\debug\build\esp-idf-sys-794f3a5c7d9aa6c9\build-script-build` (exit code: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  Python 3.13.0
  pip 24.3.1 from D:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\Lib\site-packages\pip (python 3.13)
  Requirement already satisfied: pip in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (24.3.1)
  Requirement already satisfied: setuptools in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (71.0.0)
  Collecting setuptools
    Using cached setuptools-75.6.0-py3-none-any.whl.metadata (6.7 kB)
  Using cached setuptools-75.6.0-py3-none-any.whl (1.2 MB)
  Installing collected packages: setuptools
    Attempting uninstall: setuptools
      Found existing installation: setuptools 71.0.0
      Uninstalling setuptools-71.0.0:
        Successfully uninstalled setuptools-71.0.0
  Successfully installed setuptools-75.6.0
  Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
  Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
  Collecting setuptools (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 3))
    Using cached https://dl.espressif.com/pypi/setuptools/setuptools-71.0.0-py3-none-any.whl (908 kB)
  Requirement already satisfied: packaging in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 4)) (24.2)
  Requirement already satisfied: click in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 7)) (8.1.7)
  Requirement already satisfied: pyserial in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 8)) (3.5)
  Requirement already satisfied: cryptography in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 9)) (41.0.7)
  Requirement already satisfied: pyparsing in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 10)) (3.2.0)
  Requirement already satisfied: pyelftools in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 11)) (0.31)
  Requirement already satisfied: idf-component-manager in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 12)) (2.0.4)
  Requirement already satisfied: esp-coredump in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 13)) (1.12.0)
  Requirement already satisfied: esptool in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 14)) (4.8.1)
  Requirement already satisfied: esp-idf-kconfig in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 15)) (1.4.0)
  Collecting esp-idf-kconfig (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 15))
    Using cached https://dl.espressif.com/pypi/esp-idf-kconfig/esp_idf_kconfig-1.4.3-py3-none-any.whl (44 kB)
  Requirement already satisfied: esp-idf-monitor in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 16)) (1.5.0)
  Requirement already satisfied: esp-idf-size in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 17)) (1.6.1)
  Requirement already satisfied: esp-idf-panic-decoder in d:\vats\.embuild\espressif\python_env\idf5.2_py3.13_env\lib\site-packages (from -r D:\vats\.embuild\espressif\esp-idf\v5.2.2\tools\requirements\requirements.core.txt (line 18)) (1.2.1)
[... 134 lines cut ...]
          esp_idf_version: Some(
              Tag(
                  "v5.2.2",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.2.2") }
  Using esp-idf v5.2.2 at 'D:\vats\.embuild\espressif\esp-idf\v5.2.2'
  Built components: xtensa, esp_ringbuf, efuse, esp_mm, driver, esp_pm, mbedtls, esp_bootloader_format, esp_app_format, bootloader_support, bootloader, esptool_py, partition_table, esp_partition, app_update, spi_flash, pthread, esp_system, esp_rom, hal, log, heap, soc, esp_hw_support, freertos, newlib, cxx, esp_common, esp_timer, app_trace, esp_event, nvs_flash, esp_phy, vfs, lwip, esp_netif_stack, esp_netif, wpa_supplicant, esp_coex, esp_wifi, bt, unity, cmock, console, http_parser, esp-tls, esp_adc, esp_eth, esp_gdbstub, esp_hid, tcp_transport, esp_http_client, esp_http_server, esp_https_ota, esp_https_server, esp_psram, esp_lcd, protobuf-c, protocomm, esp_local_ctrl, espcoredump, wear_levelling, sdmmc, fatfs, idf_test, ieee802154, json, mqtt, nvs_sec_provider, openthread, perfmon, spiffs, touch_element, ulp, usb, wifi_provisioning, main
  thread 'main' panicked at C:\Users\Dmitrij\.cargo\registry\src\index.crates.io-6f17d22bba15001f\bindgen-0.69.5\lib.rs:622:31:
  Unable to find libclang: "the `libclang` shared library at C:\\Users\\Dmitrij\\.rustup\\toolchains\\esp\\xtensa-esp32-elf-clang\\esp-clang\\bin\\libclang.dll could not be opened: LoadLibraryExW failed"
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
  ```
  
  I have all dependencies installed and using the unmodified generated template with correct options on the correct board (ESP32-S3).
  
  Does anyone have any ideas? I am stuck.

SergioGasquez, 2024-11-29:
Hi! 

Have you checked that `C:\\Users\\Dmitrij\\.rustup\\toolchains\\esp\\xtensa-esp32-elf-clang\\esp-clang\\bin\\libclang.dll` exists?? It should be there if the `espup` command succeeded, but just to verify.

FssAy, 2024-11-29:
> Hi!
> 
> Have you checked that `C:\\Users\\Dmitrij\\.rustup\\toolchains\\esp\\xtensa-esp32-elf-clang\\esp-clang\\bin\\libclang.dll` exists?? It should be there if the `espup` command succeeded, but just to verify.

Yes, it exists:
> The path to the file is correct, the file exists and is not corrupted.

FssAy, 2024-11-29:
I traced back the issue to the `bindgen` crate. Especially the `ensure_libclang_is_loaded` function in `lib.rs`. It fails on 
```
clang_sys::load().expect("Unable to find libclang");
```
But when I run this function manually in a dummy project it loads without any issues, so it's not a problem with my configuration or setup I guess. At this point I have no clue what is going on.

FssAy, 2024-11-30:
Fixed by downgrading to a lower espup version. 

https://github.com/esp-rs/espup/issues/440

closed 2024-11-30 by FssAy