esp-rs/rust-build #285: "riscv32imc_zicsr_zifencei-esp-espidf" unknown target

mawoka-myblock, 2024-12-20:
I was trying to compile my project https://github.com/mawoka-myblock/td-free, but it's not possible anymore as the target is supposedly not found. Logs can be found in the GitHub action: https://github.com/mawoka-myblock/td-free/actions/runs/12436048066/job/34723097689#step:12:24200

And I've tried that on two different PCs and on in the cloud on GitHub actions. I pinned the nightly to version to the 01.12.2024 one, as it's supposed to work with that, but it doesn't. Latest nightly version produces the same result.

Compiling with "cargo +esp" doesn't help either.  I've also tried the nightly version from November 1st, and same problem.

Vollbrecht, 2024-12-20:
It's not a compiler problem but one of target detection in the `cc-rs` / `cmale-rs` dependencys. You can either pin cc-rs to an old enough version or otherwise you would need to use the git versions of esp-idf-sys-/hal/svc.

Vollbrecht, 2024-12-20:
From memory the behaviour change in cc-rs was introduced in cc-rs v1.1.32. So a older version there should work

Vollbrecht, 2024-12-21:
can you show what actual cc-rs version is in your dependency graph via `cargo tree` ? I think you need to set `cc = "=1.1.20"`. Notice the = otherwise its not a "pinned" [version](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html).

mawoka-myblock, 2024-12-21:
Ah, yes! That did the trick! Thank you! Do I keep this issue open? Or is this not a real issue? IMO it is, as it breaks current setups.

Vollbrecht, 2024-12-21:
It is already fixed in what we can do inside esp-idf-sys gith repo and cmake-rs, but as you might notice that was a semver incompatible "breaking" change in cc-rs. 

There is nothing we can do here besides people themself pinning the cc-rs crate, since its a transitive dependancy. So yeah you can use our git branches without the pinning or using the normal crates release version.

mawoka-myblock, 2024-12-21:
Ah, okay. Thanks!