esp-rs/rust-build #289: Add a nix / devbox distribution option

maelp, 2025-01-27:
It's quite useful to have nix/devbox to setup efficient local dev environment, would it be possible to package rust-build in nix, so we could just `nix shell -p rust-build` and get a fully-functional dev env?

SergioGasquez, 2025-01-27:
Hi! At the moment, we do not plan on supporting a Nix distribution option, but we would welcome anyone from the community who is willing to create it and maintain it.

maelp, 2025-01-27:
I'm not a nix expert, just like how "when it works" it makes stuff easy haha, but willing to help if someone wants to give some infos

maelp, 2025-01-27:
Could you tell me for instance the most straightforward and "best practice" installation process for now to install the rust env for esp32-s3 compilation? I can take it from there

SergioGasquez, 2025-01-27:
> Could you tell me for instance the most straightforward and "best practice" installation process for now to install the rust env for esp32-s3 compilation? I can take it from there

Details are explained in https://docs.esp-rs.org/book/installation/index.html, but assuming you want a rust no_std env for S3 here is what you need to do:

1. Install `espup`: From gh release artifacts or `cargo install espup`
2. Install Xtensa Rust toolchain: `espup install`
3. Update environment variables: `. $HOME/export-esp.sh`, you can also set this environment variables elsewhere.
4. Install tools: espflash, probe-rs, esp-generate. Not sure which ones you want, all of them can be installed from sources or gh artifacts

maelp, 2025-01-27:
okay, this seems to work: https://gist.github.com/maelp/323a9ca57620009486cd834a430de680

maelp, 2025-01-27:
I haven't installed all the other tools, I guess it shouldn't be too hard, but I'm not a Nix expert 😅

SergioGasquez, 2025-01-27:
A few comments: 
- You seem to want an std env (for `esp-idf-*` crates) based on "Flake for Rust ESP-IDF setup with espup", so you need installing `ldproxy` (`cargo install ldproxy`). See https://docs.esp-rs.org/book/installation/std-requirements.html for more details
- The default `name` is already `esp` so you could remove the `--name` flag
- I would suggest that the description says that its only for ESP32S3 or that you install Xtensa Rust for all the targets

maelp, 2025-01-27:
@SergioGasquez thanks for the comments! Indeed it was meant more as a kind of "starter config" so that someone more experienced could finish packaging it, do you want me to do the modifications?

I'm targeting no-std for esp32-s3, but indeed if we integrate this in one of the `esp-rs` package we should definitely make this configurable, however my Nix skills are not that great (and ChatGPT doesn't really get nix either haha) so I'm not sure how to do all this

If you know someone who could finish packaging all of this he might be more efficient than the two of us doing back and forth perhaps? However if you think that just the above changes are sufficient I can implement them and send you a new gist

SergioGasquez, 2025-01-28:
> do you want me to do the modifications?

As you want, as mentioned before, we won't be able to maintain this, so if you want to take care of this and share it, maybe other nix user can build on top of it!

> I'm targeting no-std for esp32-s3

Then,  `description = "Flake for Rust ESP-IDF setup with espup";` is not true, no_std is not related ESP-IDF.

polyfloyd, 2025-07-23:
Hey I just found this thread. I have been working on a Nix flake too today. It bypasses espup entirely and installs the required toolchains directly. This is nice because it leverages Nix's integrity checks and caching.

Find it here: https://gist.github.com/polyfloyd/9e8acf73ec86a6c4e211a980c11b1712

Not expecting this to be merged, but for reference for others looking for this :)

waynevanson, 2025-09-07:
I have a fork of a fork of a fork of a fork over here of the `nixpkgs-esp-idf` implementation.

1. fork^4
    https://github.com/waynevanson/nixpkgs-esp-dev-rust
2. Base repository
    https://github.com/mirrexagon/nixpkgs-esp-dev

So the base repository implemented `esp-idf` as a nix flake as an overlay, which includes packages and dev shells for working with esp-idf without rust.

The forks appended with `-rust` aim to add all the `esp` forks like LLVM, cross-tools and Rust as packages and shells too. They just download the binaries from GitHub releases and nixify them. That's how the `nixpkgs` currently downloads them too.

The primary gap with the rust repositories and maybe the base repositories is that they're not kept up to date with the state of `esp` and `esp-rs` organisations. The latest I found was `1.86.0.0` when current version is `1.88.0.0`.

waynevanson, 2025-09-16:
Hey @mirrexagon, your work seems aligned with this feature.

I'm happy to try put the work you've done and the fork I have into repositories owned by esp.

Would you like to implement your changes in the esp-idf repo, or are you happy for me to do it?
I'll admit I know a bit about nix and rust but really not much about esp-idf.

mirrexagon, 2025-09-18:
@waynevanson I'm happy for you to do it and take whatever code and inspiration you need from my repo. I unfortunately don't have time to work on this kind of thing much at all, but if you have questions about how my repo works, let me know!