esp-rs/rust-build #291: Tool doesn't match supported version

maelp, 2025-02-27:
I'm trying to build a project based on `wamr-rust-sdk` which used to compile, but now when I try to do it, it says this:

```
Using activated esp-idf v5.5.0 environment at '/Users/primet/esp/esp-idf'
  CMake Error at /Users/primet/esp/esp-idf/tools/cmake/tool_version_check.cmake:36 (message):

    Tool doesn't match supported version from list ['esp-14.2.0_20241119']:
    /Users/primet/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
```

I'm not sure how to install the 14.2.0_20241119 toolchain (rather than the 14.2.0_20240906) ?

MabezDev, 2025-02-27:
This isn't to do with the rust compiler. If you're using a local esp-idf version, which you are, run the `install.sh` script from esp-idf, then source the esp-idf environment. After that try building your project in the same terminal.

closed 2025-02-27 by MabezDev

maelp, 2025-02-27:
@MabezDev hmmm weird, that's literally what I did? are you also using MacOS? do you know what these versions correspond to? I'm wondering why it doesn't install the latest version for me (esp-14.2.0_20241119)